espressif/esp-idf #3828: Undefined Constants in Example Code

hassankamran012, 2019-07-25:
Hello, I was reading the example code and some of the variable are not defined that VS Code picks up as errors but are built by idf.py build command by default. Kindly explain why this is. I was specfically looking at the uart echo example.

portTICK_RATE_MS
GPIO_NUM_4
GPIO_NUM_5

Alvin1Zhang, 2019-07-25:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

projectgus, 2019-07-25:
@hassankamran012 ,

VS Code needs to know the paths for header files to be included from, usually it infers this from the build system. If it doesn't know where to find the headers then any symbol which is defined in the header will be marked as unknown by VSCode (because it doesn't know the header paths)..

We don't support using VSCode with ESP-IDF yet. But this feature request includes some discussion of the exact problem you're describing:
https://github.com/espressif/esp-idf/issues/303#issuecomment-508423180

If you'd like to be notified when official VSCode support is available, please subscribe to the linked issue.

closed 2019-07-25 by projectgus