espressif/esp-idf #3829: Update to the latest version, smart config  still be errors, although the number is very small

GitOfLance, 2019-07-25:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-728-g826ff7186
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
I used to configure wifi with smartconfig, there was an error, which is described in this [Issues](https://github.com/espressif/esp-idf/issues/3586).
Two days ago, I saw a patch about `wifi stop` and updated the code version.
In these two days of use, smartconfig no longer has an error every time, but there will still be an error, causing the system to crash, probably 3 times. There were about 3 errors in 100 uses.
[image]

In addition, although the function `esp_wifi_set_storage(WIFI_STORAGE_FLASH)` was successfully called when the wifi was initialized, after the restart, the function `esp_wifi_get_config` was called but no information about wifi was obtained for connecting to wifi.
What if I want to save the wifi information configured via smartconfig and use this information to automatically connect to the next time I start it?

filzek, 2019-07-28:
One thing is clear, WiFi need to be initialized and full set in the main app or it will always faill since 20 days ago.

GitOfLance, 2019-07-29:
Can you make it clear?
The wifi initialization function is called in `app_main()`. What about `WiFi need to be initialized and full set in the main app or it will always faill since 20 days ago.` mean?
In this `ISSUSE`[HERE LINK](https://github.com/espressif/esp-idf/issues/3586), isn't that the problem solved? Why is there an error when using smartconfig?
In addition, in the initialization function I call the function "`esp_wifi_set_storage(WIFI_STORAGE_FLASH)`", why does the function "`esp_wifi_get_config(ESP_IF_WIFI_STA, &wifi_staconfig.sta);`" not get ssid and pwd after starting the program and wifi initialization?

GitOfLance, 2019-08-02:
HI, @filzek 
I synced the `IDF` code on `August 29, 2019` and ran the `example/wifi/smart_conifg` test code you provided. The error is as follows:
[image]

Are you sure that the smartconfig demo is working?

GitOfLance, 2019-08-02:
The chip model I am using is: `ESP32-WROVER-B`, with `8MB PSRAM` and `8MB Flash`. Does this affect?

liuzfesp, 2019-08-02:
HI @GitOfLance , just can't reproduce this issue with smartconfig example in v4.0-dev-728-g826ff7186. Does this issue happen with default menuconfig?

GitOfLance, 2019-08-03:
Hi, @liuzfesp 
Yes, I use the `esp32 kit V3` development board and default menuconfig, the example in IDF will be wrong.

filzek, 2019-08-04:
@GitOfLance there is some real issue with smartconfig, but the example compiles fine and runs okay in some wifi routers (it seens that has problem to found and match sniffer in some wifi routers), there is no problem on build and run in any of our testing boards here, all based on wrover-b modules 4/8 and 8/8mb.

Did you try to redownload the entire master SDK again, maybe there is some broken parts in your sdk that coundnt be fix or upgrade by a simple git pull --rec, so, please advise to uninstall and install it again.

GitOfLance, 2019-08-06:
Hi, @filzek 
I deleted the `esp-idf` yesterday and pull the repository from github by `git clone --recursive https://github.com/espressif/esp-idf.git`. Use the default menuconfig and the provided demo (`exmaple/wifi/smart_config`).
Run this demo on `esp32-wrover-b`. In most cases, I can configure the wifi through smartconfig, but there will be a certain probability that the system will crash, as shown in the following figure:
[image]
I uesd demo and IDF version:
[image]

**##So, I can say with certainty that the demo of smartconfig is not completely correct.##**

In addition, I initialized and connected wifi according to the `example/wifi/scan` method, the test result is normal, but my project needs to allow the wifi to be configured repeatedly, so I need the wifi initialization function to be repeatable. Before each initialization, I will call the some functions to release the wifi resource.
wifi init function:
```
void wifi_init_ilock()
{
    int ret = 0;
    
    do {   
        wifi_base_init();       //底层的初始化函数

        /* 适应重复初始化 */
        if (EventGroup_wifiilock)
        {
            ESP_LOGW("wifi init ilock", "wifi deinit");
            wifi_deinit_ilock();
            vEventGroupDelete(EventGroup_wifiilock);
            EventGroup_wifiilock = NULL;
        }
        EventGroup_wifiilock = xEventGroupCreate();
        if (NULL == EventGroup_wifiilock)
        {
            ret = -1;
            ESP_LOGE(__FUNCTION__, "Create wifiilock EventGroup failed");
            break;
        }

        /* 监测base抛出的事件 */
        xTaskCreate(wifi_base_event_monitor, "wifi_monitor", 1024 * 4, NULL, configMAX_PRIORITIES - 2, &wifievent_monitor);  

        if (wifievent_monitor)
            ESP_LOGI("wifi init", "create wifi task success");
        else
            ESP_LOGI("wifi init", "crate wifi task failed");
    }while(0);

    /* 抛出本层的异常 */
    if (ret < 0)        
    {
        if (EventGroup_wifiilock)
            xEventGroupSetBits(EventGroup_wifiilock, wifiilockevent_unavailable);
    }
}

void wifi_base_init()
{
    int ret = 0;     //返回值
    wifi_init_config_t wificonfig = WIFI_INIT_CONFIG_DEFAULT();

    do
    {
        /* 终止上次wifi，并且释放相关资源 */
        /* 创建事件组，如果之前创建过，说明已经初始化过wifi，则删除 */
        if (EventGroup_wifibase)
        {
            wifi_stop_release_sources();        // 终止wifi,释放资源
            vEventGroupDelete(EventGroup_wifibase);     //必须在stop函数后面删除，因为stop函数会设置这个事件组中的事件
            EventGroup_wifibase = NULL;
        }            
        EventGroup_wifibase = xEventGroupCreate();
        if (NULL == EventGroup_wifibase)
        {
[... 94 lines cut ...]
    esp_event_handler_unregister(SC_EVENT, ESP_EVENT_ANY_ID,  wifi_base_event_loop_handler);   
    esp_event_loop_delete_default();
    esp_wifi_disconnect();
    esp_wifi_scan_stop();
    esp_wifi_stop();
    esp_wifi_deinit();
}

/* wifi资源释放，适应重复初始化 */
static void wifi_deinit_ilock()
{
    if (wifievent_monitor)
    {
        ESP_LOGW("WIFI DEINIT", "Delete task by Handler");
        vTaskDelete(wifievent_monitor);
        wifievent_monitor = NULL;
    }
}
```
But the following error message will appear, like the wifi resource is not cleaned up:
[image]

I want konw:
1. How can I operate to ensure that wifi initialization is re-callable?
2. Is there a rough time to solve the smartconfig problem?

negativekelvin, 2019-08-06:
`esp_event_loop_delete_default();`

You call this too early.  Do it last.

liuzfesp, 2019-08-06:
Hi @GitOfLance, for the smartconfig issue,  the challenge part is that we can't reproduce your issue with same IDF commit id and demo code. Now I suspect it may related to your board, could you change to a different board and check whether the issue still exists?

filzek, 2019-08-07:
@GitOfLance  we do not have this problem, we can build and run, only problem is that some Routers will not even work with smartconfig, it seens that the router send some ack in the same time APP is running in IOs or Android, so, to make it work you need to restart the router and run smartconfig same time it get back online. It is a problem with the sdk for sure, but not sure where.

Your problem related doesnt happen in wrover-b rev1 3/4 code scheme fuse.

GitOfLance, 2019-08-07:
Hi, @negativekelvin 
I put this function ( `esp_event_loop_delete_default()` ) at the end of the release resource function, and the problem is solved, although I don't know why, as shown below:
```
void wifi_stop_release_sources()
{
    /* 抛出stop事件，不进行断网重连 */
    xEventGroupSetBits(EventGroup_wifibase, wifieventbits_stop);
    
    esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID,  wifi_base_event_loop_handler);
    esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP,  wifi_base_event_loop_handler);
    esp_event_handler_unregister(SC_EVENT, ESP_EVENT_ANY_ID,  wifi_base_event_loop_handler);   
    esp_wifi_disconnect();
    esp_wifi_scan_stop();
    esp_wifi_stop();
    esp_wifi_deinit();
    esp_event_loop_delete_default();        //放最后，测试
}
```
Thanks for your advice.

GitOfLance, 2019-08-07:
Hi， @liuzfesp @filzek 
The process for making this error appear is as follows:
1. Compile and run the smartconfig demo in the default configuration.
2. When the message appears(`smartconfig: SC version: V2.9.0`), operate the APP to use smartconfig;
3. After the wifi connection is successful, restart the ESP32 through the `restart button` on the board.
4. Then reconfigure the wifi via smartconfig...

Repeat the operation like above and an error message will appear.
You can try to follow this process to see if an error message appears.

[image]

The top five boards from left to right use the `ESP32-WROVER-B` module, which has `4MB` and `8MB` Flash or PSRAM.
According to the above test method, **ten times a group, calculate the number of errors**.
**The test results are as follows:**
- The first board, test a group, the error is `5 times`, that is, the error probability is `5/10`;
- The second board, testing two groups, once `4/10`, once `3/13`;
- The third board, testing two groups, each time is `4/10`;
- The fourth board, test a group, `5/10`;
- The fifth board, test a group, `3/10`;

**There is no single board to ensure that every smartconfig can be successful, so I think the smartconfig demo should be a hidden bug.**

GitOfLance, 2019-08-07:
Another problem is that when I get `ssid` and `pwd`, I will directly call the `esp_wifi_connect()` function to connect to wifi. If the account and password are correct, I can connect to wifi. However, if the password account is wrong, the return of the `esp_wifi_connect()` function is still **`ESP_OK`**. The program cannot exit.
The codes:
```
static void wifi_base_connect_ssid_pwd(wifi_config_t wificonfig, bool def)
{  
    int ret = 0;
    
    do {
        /* 参数断言 */
        if ('\0' == wificonfig.sta.ssid[0] || '\0' == wificonfig.sta.password[0])
        {
            ESP_LOGE(__FUNCTION__, "WiFi Config SSID or PWD is NULL");
            ret = -1;
            break;
        }

        /* 使用默认配置 */
        if (true == def)
        {
            wificonfig.sta.scan_method = WIFI_FAST_SCAN;
            wificonfig.sta.bssid_set = 0;
            wificonfig.sta.channel = 0;
            wificonfig.sta.listen_interval = 0;
            wificonfig.sta.sort_method = WIFI_CONNECT_AP_BY_SIGNAL;
            wificonfig.sta.threshold.rssi = -127;
            wificonfig.sta.threshold.authmode = WIFI_AUTH_OPEN;
        }

        ESP_LOGW("WIFI CONNECT", "SSID : %s", wificonfig.sta.ssid);
        ESP_LOGW("WIFI CONNECT", "PWD : %s", wificonfig.sta.password);

        /* 连接网络 */
        esp_wifi_set_config(ESP_IF_WIFI_STA, &wificonfig);
        if (ESP_OK != esp_wifi_connect())
        {
            ESP_LOGE(__FUNCTION__, "Call esp_wifi_connect failed");
            ret = -1;
            break;
        } 
    }while(0);

    /* 配网失败，设置事件 */
    if (ret < 0)
    {
        xEventGroupSetBits(EventGroup_wifibase, wifieventbits_connectfailed);
    }   
}
```
The error message is as follows:
[image]

GitOfLance, 2019-08-13:
Hi, @liuzfesp @filzek 
Do you have reproduce this issue?

liuzfesp, 2019-08-13:
@GitOfLance just can't find the time slot to debug this issue these days. Will do more investigation this week.

espxiehang, 2019-11-06:
hi @GitOfLance  Fix version : 
```
 3638 commit 53d57dd736e4a56c07bf7f763f776058bb801348
 3639 Author: zhangyanjiao <zhangyanjiao@espressif.com>
 3640 Date:   Sun Sep 29 10:16:38 2019 +0800
 3641 
 3642     esp_wifi:
 3643     1. Fixed smartconfig ipc crash
 3644     2. Fix the WiFi init sequence bug
 3645     3. Fixed the bug that WiFi stop leads to task watchdog
 3646 

```

Please help verify that this version will resolve your issue.

Alvin1Zhang, 2020-01-19:
@GitOfLance Thanks for reporting. And very sorry for the slow turnaround, do you have time to try the details shared by @espxiehang and help share if any updates for the issue? Thanks.

closed 2022-01-05 by jack0c