espressif/esp-idf #3830: Request: offer to install in `C:\esp`, on Windows, by default

freedaun, 2019-07-25:
I propose the installer offer to install in the root folder `C:\esp` on Windows. 

This is not a recommended practice on Windows for normal programs, but for those programs the user does not interact much with the install. Developers, however, do interact with the install, and languages like Lazarus, Go and Flutter do use that system. 

This install results in shorter pathnames, which are more important than one might think, if only to prevent lines in the output from breaking up. But shortness of paths is not the main issue, and not enough reason for making this a default: 

- when the default install uses C:\esp, all communication of both users and the docs (in their examples) can assume absolute paths, simplifying them greatly, and scripts and instructions work without customization.

- it would be possible (or much easier) to make a portable install, and pass the project to someone, without needing an installation or instructions. Tools like Eclipse could be configured simply by importing a file. 

These things make a BIG difference in ergonomics for the user and for Espressif. Reliability (of install and communications) is also highly favored by simplicity.

Doing it this way of course means only one install per machine. Which isn't standard practice, except for development, as exemplified by the examples I gave.

Also:

- anything that depends on instructions is bad, as humans are bad at following instructions. Inevitably they introduce small differences that later cause breakage, without anyone being able to pinpoint why. 

It is therefore important to minimize such needless variations. The insertion of a username in a path is such a thing. I think it is a mistake.

georgik, 2021-04-13:
@freedaun Thank you for the suggestion and our apology for the late response. 

Let me add a few notes and updates. The default configuration of the Windows installer is to deploy the application to userdesktop\esp-idf and cache with data to ~\.espressif. 
This configuration does not require the user to elevate privileges to perform the installation.
Using C:\esp might cause trouble to users without write permission to C:\.

Meanwhile, we updated the installer and the requested behavior can be achieved by adding the following parameter on the command line:
```
.\esp-idf-tools-setup-online-2.7.exe /IDFDIR=C:\esp
```

We also published the rest of the available parameters in README of installer project:
https://github.com/espressif/idf-installer#command-line-parameters

Since version 2.6 the Windows Installer contains also support to install pre-configured Eclipse.

You can download the installer from: https://dl.espressif.com/dl/esp-idf/

We welcome your suggestions to improve the tooling and simplify the onboarding experience.

freedaun, 2021-04-13:
Thank you for getting back to me, and thanks in the name of those who get here in the future, in search of this feature. It still is a much wanted feature, I'm sure to try this out soon. Good work.

closed 2021-04-16 by tom-borcin