espressif/esp-idf #3831: macos python

erdemirbehiye, 2019-07-25:
If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2019-07-25:
Please make sure your PATH variable includes not only the toolchain directory, but also the standard `/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin` directories.

erdemirbehiye, 2019-07-25:
@igrr Hi,
When I typed on terminal  echo $PATH :
/Users/aa/esp/esp-idf/components/esptool_py/esptool:/Users/aa/esp/esp-idf/components/espcoredump:/Users/aa/esp/esp-idf/components/partition_table/:/Users/aa/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:/Users/aa/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/Users/aa/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/Users/aa/.espressif/python_env/idf4.0_py3.7_env/bin:/Users/aa/esp/esp-idf/tools:/Users/aa/anaconda3/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/opt/X11/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/aa/esp/xtensa-esp32-elf/bin

I tried that and this: 
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/aa/esp/xtensa-esp32-elf/bin

[image]

[image]

igrr, 2019-07-25:
Since you are on not using Windows, it may be okay to use `make` command instead of `python .../tools/windows/eclips_make.py flash`?

erdemirbehiye, 2019-07-26:
@igrr  when i used the make, i faced another problem :

[image]

igrr, 2019-07-26:
The text in front of `gcc` (should be `xtensa-esp32-elf-`) is set by CONFIG_SDK_TOOLPREFIX menuconfig option. Is there a chance that you have accidentally set it to your serial port (`/dev/tty.usbserial1420`) instead? You can attach your sdkconfig file if you are not sure.

erdemirbehiye, 2019-07-26:
@igrr you're right. I changed it. It works. Thank you.

closed 2019-07-26 by igrr