espressif/esp-idf #3834: Compile errors in components/esp_gdbstub/src/gdbstub.c with latest dev/master branch

kvbulusu, 2019-07-26:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-1338-g842432fb6
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Linux
Power Supply: USB

Problem Description:
Error while compiling.
esp-idf/components/esp_gdbstub/src/gdbstub.c: In function 'esp_gdbstub_handle_command':
esp-idf/components/esp_gdbstub/src/gdbstub.c:213:13: error: 'handle' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             esp_gdbstub_tcb_to_regfile(handle, &s_scratch.regfile);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

igrr, 2019-07-26:
Hi @kvbulusu, could you please attach your sdkconfig file?

kvbulusu, 2019-07-27:
> Hi @kvbulusu, could you please attach your sdkconfig file?
Here you go..
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3437725/sdkconfig.txt)

michaeltintiuc, 2019-08-10:
Confirming to have the same issue, [sdkconfig](https://hatebin.com/mjahupcqcc)

michaeltintiuc, 2019-08-10:
This seems to work nicely as a temporary workaround:
```
-            TaskHandle_t handle;
+            TaskHandle_t handle = NULL;
```

mahavirj, 2019-08-21:
@michaeltintiuc @kvbulusu Fixed with https://github.com/espressif/esp-idf/commit/5203a1543996e5ad9aacd33672dda9103c0c8aad

closed 2019-08-21 by mahavirj