espressif/esp-idf #3835: "esp_a2d_sink_deinit()" in an endless loop

xiruilin, 2019-07-26:
## Problem Description

In "esp_a2d_sink_deinit()" function,
it's call "osi_thread_post()",
this lead to "osi_thread_run()" activation, 
and the process is below:

->"osi_thread_run()"
->"btc_thread_handler()"
->"btc_a2dp_call_handler()"
->"btc_a2d_sink_deinit()"
->"clean_up()"
->"btc_a2dp_sink_shutdown()"

In "btc_a2dp_sink_shutdown()" function, first call
"btc_a2dp_sink_ctrl_post()", then call "future_await()".

In "btc_a2dp_sink_ctrl_post()" function,
it's call "osi_thread_post()" again.
But now is in "osi_thread_run()" process and
"future_await()" will be never get signal.

Alvin1Zhang, 2019-11-04:
@xiruilin Thanks for reporting, we already have the fix and now under internal reviewing, will be available on GitHub soon. Thanks.

xiruilin, 2019-11-21:
@Alvin1Zhang 
This problem has been solved in commit, thanks.

closed 2019-11-21 by xiruilin