espressif/esp-idf #3838: Error when no longer using legacy event loop

avanbremen, 2019-07-26:
## Environment

- Development Kit: **none**
- Module or chip used: **ESP32-WROOM-32D**
- IDF version: **v4.0-dev-1136-g28f1cdf5e**
- Build System: **CMake**
- Compiler version: **xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0**
- Operating System: **Windows**
- Power Supply: **USB**

## Problem Description

From the latest Event Handling [API Guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/event-handling.html) I understand that the `esp_event` library is designed to supersede the legacy event loop for the purposes of event handling in ESP-IDF.

Therefore I replaced `esp_event_loop_init` with `esp_event_loop_create_default` and `esp_event_handler_register`, as in example [station_example_main.c](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c).

Everything works, however the following error is printed:
`event: system event loop not initialized via esp_event_loop_init`

This is printed by function [esp_event_send_legacy](https://github.com/espressif/esp-idf/blob/3bb7dba9957ee71a236058322ae5e70f96f9a104/components/esp_event/event_loop_legacy.c#L49), which is called by function [esp_event_send](https://github.com/espressif/esp-idf/blob/138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b/components/esp_event/event_send.c#L39).

Function `esp_event_send` immediately returns `if (err != ESP_OK)`, which is always true when the (now deprecated) legacy event loop is not initialized via `esp_event_loop_init`.

This was also the case in the previous version of [event_send.c](https://github.com/espressif/esp-idf/blob/b1d1e37f8723de0cf083a3a268bab421178c60bc/components/esp_event/event_send.c#L42), which is included in esp-idf v4.0-dev-1136-g28f1cdf5e. However in this case the faulty `return err;` was followed by [esp_event_mesh_hook](https://github.com/espressif/esp-idf/blob/b1d1e37f8723de0cf083a3a268bab421178c60bc/components/esp_event/event_send.c#L46), which was never called when the legacy event loop was no longer used.

### Expected Behavior

- The error should not be printed in case the legacy event loop is no longer used.
- Function `esp_event_send_legacy` should not even be called in case the legacy event loop is not used.
- Function `esp_event_send` should not return immediately in case sending an event to the legacy event loop fails.
- Perhaps including the legacy event loop code should even be made configurable via menuconfig.

I noticed the `esp_event_mesh_hook` call in `esp_event_send` is missing from the latest commit, but perhaps it would still be good design to address how the legacy loop is handled.

### Actual Behavior

Errors are printed and `esp_event_send` returns immediately.

igrr, 2019-07-26:
esp_event_send_legacy is not supposed to be linked into the application if the legacy event loop is not used in the app. Instead a weak dummy function from event_send.c should be included. Is it possible that your app is still calling any of the APIs of the legacy interface? If you are unsure you can attach your application's map file, and we will check.

avanbremen, 2019-07-26:
@igrr Thank you very much for your reply. I created a new component to replace the one that used `esp_event_loop_init`. I did not yet delete the old, now deprecated component from my project. Even though esp_event_loop_init is no longer called by the application, apparently the legacy event loop code is still included.

Long story short, I just removed the old component and the error is no longer showing, just like you said. Thanks.

closed 2019-07-26 by avanbremen

q1902505163, 2021-07-08:
up ,I got the same error as you.Can you tell me how to create a new component.

F99, 2021-07-28:
> up ,I got the same error as you.Can you tell me how to create a new component.

this is really weird stuff, but you must remove or uncomment all ocurrences of "esp_event_loop_init" from your code, even if never called. 
otherwise it's okay to use it, but then it has to be called first.