espressif/esp-idf #3840: Insufficient TCP connections in ESP32-WROVER-B with latest SDK

surfskidude, 2019-07-26:
## Problem Description

I get the following error when I run a fairly heavy duty network application that is simultaneously acting as an HTTP server and HTTP client when connecting a browser.

W (338406) wifi: mem fail : 
W (338466) wifi: alloc eb len=24 type=3 fail, heap:4140676

After the above failure, I tried several configurations and they all failed.

I also expected the the ESP32-WROVER-B to be able to do many TCP connections, but I noticed that I can only configure up to 16 connections using make menuconfig. 

Do you have plans on supporting more connections? I would like to run up to 100 full duplex TCP connections.

## Environment

- Development Kit:   ESP32-WROVER-B
- Module or chip used:  WINGONEER T-Koala ESP32 
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         CMake
- Compiler version:  crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

negativekelvin, 2019-07-26:
https://github.com/espressif/esp-idf/issues/3592

freakyxue, 2019-11-13:
@surfskidude This problem has been fixed and synchronized to the hub, please test

surfskidude, 2019-11-15:
I noticed that max TCP connections are still 16 when using make menuconfig. Maybe the tool has not been updated?

I have not looked into manually crafting the config file so can you please send me an sdkconfig file configured for high end TCP use?

BTW, just to make sure, I did a new checkout as follows:
 git clone --recursive https://github.com/espressif/esp-idf.git

freakyxue, 2019-11-15:
@surfskidude 
sorry ，I didn't see what you need。
You want to configure 100 TCP sockets。
But IDF can only configure 16 maximum values through menuconfig. If too many settings are configured, RAM space bu'gou will be generated。

config LWIP_MAX_SOCKETS
        int "Max number of open sockets"
        range 1 16
        default 10

if  you  want to config 100 sockets, you can modify the value,"range 1 100"

jack0c, 2022-05-20:
@surfskidude are still working on this? I will close this issue, feel free to reopen it if necessary.

closed 2022-05-20 by jack0c