espressif/esp-idf #3841: UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 137: ordinal not in range(128)

franzhoepfinger, 2019-07-27:
## Environment

- IDF version (run ``git describe --tags`` to find it): 
        v4.0-dev-1342-g84982ae2e
- Build System:         [CMake]

- Operating System:     [Linux]

## Problem Description

when doing 
. $HOME/esp/esp-idf/export.sh

i get following error message:

franz@franz-Lenovo-V130-14IKB:~$ . $HOME/esp/esp-idf/export.sh
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/home/franz/esp/esp-idf/tools/idf_tools.py", line 1256, in <module>
    main(sys.argv[1:])
  File "/home/franz/esp/esp-idf/tools/idf_tools.py", line 1252, in main
    action_func(args)
  File "/home/franz/esp/esp-idf/tools/idf_tools.py", line 874, in action_export
    tool.find_installed_versions()
  File "/home/franz/esp/esp-idf/tools/idf_tools.py", line 491, in find_installed_versions
    ver_str = self.check_version(self.get_export_paths(version))
  File "/home/franz/esp/esp-idf/tools/idf_tools.py", line 435, in check_version
    in_str = version_cmd_result.decode()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 137: ordinal not in range(128)

### Expected Behavior

no Error Message

### Actual Behavior

Error Message

franzhoepfinger, 2019-07-27:
[image]

franzhoepfinger, 2019-07-27:
[image]

franzhoepfinger, 2019-07-27:
English Ubuntu: no Problem.
German Ububtu: Problem

AnsgarSchmidt, 2019-07-28:
[image]

English Ubuntu 19.10 has Problems too but decoding 0xd0

Iskenof, 2019-07-29:
+1 same problem with german ubuntu

4.15.0-55-generic

NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"

wibsea, 2019-07-31:
Russian Ubuntu 18.04.2
 $HOME/esp/esp-idf/export.sh
.
.
.
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 4: ordinal not in range(128)

But:
LANG=en_US . $HOME/esp/esp-idf/export.sh
no problem

aarizaq, 2019-07-31:
In the file idf_tools.py if you replace the line:
in_str = version_cmd_result.decode()

by:
in_str = version_cmd_result.decode("utf-8")

will solve the problems

Iskenof, 2019-08-01:
Hello @aarizaq 

after this change it work.

. ./export.sh
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements from /home/andre/Downloads/esp/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/andre/Downloads/esp/esp-idf/components/esptool_py/esptool
  /home/andre/Downloads/esp/esp-idf/components/espcoredump
  /home/andre/Downloads/esp/esp-idf/components/partition_table/
  /home/andre/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin
  /home/andre/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin
  /home/andre/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin
  /home/andre/.espressif/python_env/idf4.0_py2.7_env/bin
  /home/andre/Downloads/esp/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

suda-morris, 2019-08-01:
Sorry for the delay, we will fix the bug soon.

closed 2019-08-21 by igrr