espressif/esp-idf #3842: assertion "netconn marked closed" failed

PerMalmberg, 2019-07-27:
v4.0-dev-1338-g842432fb6 (with @igrr 's modified libc.a from [here](https://github.com/espressif/esp-idf/issues/3624#issuecomment-515170464))
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0

I got the following assertion today. What exactly does it mean? Based on the stacktrace it looks as if it is all outside my code, but can I have caused it somehow? It happened when I had several active sockets all sending data. I've been unable to reproduce the assertion.

Fwiw, the system had previously logged the following message while `accept()`-ing incoming sockets:

> Too many open files in system: Too many open files in system (23)

Assertion with stacktrace:
```
assertion "netconn marked closed" failed: file "/home/permal/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c", line 860, function: netconn_drain
abort() was called at PC 0x4020ceff on core 0
0x4020ceff: __assert_func at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: cc3a727c7b571b062530e969af4ca4858f703abea5e0f7a2fd9071d72e7d8c6c

Backtrace: 0x400936e9:0x3ffd0a50 0x40093b5c:0x3ffd0a80 0x4020ceff:0x3ffd0ab0 0x4017a71d:0x3ffd0ae0 0x4017a699:0x3ffd0b10 0x4017a8c9:0x3ffd0b40 0x40167669:0x3ffd0b70 0x401682b1:0x3ffd0bb0 0x4016fe46:0x3ffd0bf0 0x40177042:0x3ffd0c20 0x4015fd09:0x3ffd0c50 0x4015fdc4:0x3ffd0c80 0x400983ea:0x3ffd0cb0
0x400936e9: invoke_abort at /home/permal/esp/esp-idf/components/esp32/panic.c:155

0x40093b5c: abort at /home/permal/esp/esp-idf/components/esp32/panic.c:172

0x4020ceff: __assert_func at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x4017a71d: netconn_drain at /home/permal/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:860 (discriminator 1)

0x4017a699: netconn_free at /home/permal/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:826

0x4017a8c9: accept_function at /home/permal/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c:603

0x40167669: tcp_process at /home/permal/esp/esp-idf/components/lwip/lwip/src/core/tcp_in.c:941 (discriminator 1)

0x401682b1: tcp_input at /home/permal/esp/esp-idf/components/lwip/lwip/src/core/tcp_in.c:438

0x4016fe46: ip4_input at /home/permal/esp/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:760

0x40177042: ethernet_input at /home/permal/esp/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186

0x4015fd09: tcpip_thread_handle_msg at /home/permal/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:180

0x4015fdc4: tcpip_thread at /home/permal/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:154

0x400983ea: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.
```

freakyxue, 2019-10-22:
hi @PerMalmberg   How can I reproduce this problem

PerMalmberg, 2019-10-22:
@freakyxue I have no idea - I've not seen it since I filed this issue. This was a request for information on what the assert actually means.

 I have no proof, but I suspect it might be related to the PSRAM cache issue #2892

freakyxue, 2019-11-13:
hi @PerMalmberg   There is fix Mr already. Please wait for the code to be synchronized to GitHub.

PerMalmberg, 2019-11-13:
@freakyxue A fix to the netconn-issue reported here, or do you mean the referenced PSRAM issue? 

If the former, what was the actual problem?

freakyxue, 2019-11-14:
hi @PerMalmberg  I found that the root cause is that when the server listens to the connection information of the client, it sends data to the accept box. If it fails, it will delete the Conn, but it is not set as invalid at this time. When deleting the Conn, an assertion will appear

accept_function(void *arg, struct tcp_pcb *newpcb, err_t err)
{
  struct netconn *newconn;
  struct netconn *conn = (struct netconn *)arg;

  if (conn == NULL) {
    return ERR_VAL;
  }
  if (!NETCONN_MBOX_VALID(conn, &conn->acceptmbox)) {
    LWIP_DEBUGF(API_MSG_DEBUG, ("accept_function: acceptmbox already deleted\n"));
    return ERR_VAL;
  }

  if (newpcb == NULL) {
    /* out-of-pcbs during connect: pass on this error to the application */
    if (sys_mbox_trypost(&conn->acceptmbox, lwip_netconn_err_to_msg(ERR_ABRT)) == ERR_OK) {
      /* Register event with callback */
      API_EVENT(conn, NETCONN_EVT_RCVPLUS, 0);
    }
    return ERR_VAL;
  }
  LWIP_ASSERT("expect newpcb == NULL or err == ERR_OK", err == ERR_OK);
  LWIP_UNUSED_ARG(err); /* for LWIP_NOASSERT */

  LWIP_DEBUGF(API_MSG_DEBUG, ("accept_function: newpcb->state: %s\n", tcp_debug_state_str(newpcb->state)));

  /* We have to set the callback here even though
   * the new socket is unknown. newconn->socket is marked as -1. */
  newconn = netconn_alloc(conn->type, conn->callback);
  if (newconn == NULL) {
    /* outof netconns: pass on this error to the application */
    if (sys_mbox_trypost(&conn->acceptmbox, lwip_netconn_err_to_msg(ERR_ABRT)) == ERR_OK) {
      /* Register event with callback */
      API_EVENT(conn, NETCONN_EVT_RCVPLUS, 0);
    }
    return ERR_MEM;
  }
  newconn->pcb.tcp = newpcb;
  setup_tcp(newconn);

  /* handle backlog counter */
  tcp_backlog_delayed(newpcb);

  if (sys_mbox_trypost(&conn->acceptmbox, newconn) != ERR_OK) {
    /* When returning != ERR_OK, the pcb is aborted in tcp_process(),
       so do nothing here! */
    /* remove all references to this netconn from the pcb */
    struct tcp_pcb *pcb = newconn->pcb.tcp;
    tcp_arg(pcb, NULL);
    tcp_recv(pcb, NULL);
    tcp_sent(pcb, NULL);
    tcp_poll(pcb, NULL, 0);
    tcp_err(pcb, NULL);
    /* remove reference from to the pcb from this netconn */
    newconn->pcb.tcp = NULL;
#if ESP_LWIP    
    newconn->flags |= NETCONN_FLAG_MBOXINVALID;
#endif /* ESP_LWIP */

PerMalmberg, 2019-11-14:
@freakyxue Which server do you mean? The read-made http-server in IDF or the underlying lwip layer?

freakyxue, 2019-11-14:
hi @PerMalmberg 

When I navigate to the error, I start with the accept function,

If (sys ﹐ Mbox ﹐ trypost (& conn - > acceptmbox, newconn)! = err ﹐ OK), if it fails, you need to set conn to be invalid, otherwise an assertion will appear when deleting this conn

PerMalmberg, 2019-11-14:
Ok, so it is in the lwip-layer. Glad you found it, looking forward for it to be merged. :+1:

freakyxue, 2020-03-06:
hi @PerMalmberg 
this bug has been fixed .thanks

PerMalmberg, 2020-03-06:
@freakyxue And merged? What commit?

freakyxue, 2020-03-09:
@PerMalmberg 
f7d2aa96bd7b8f6516d7ad0a4a81dd42c1132027
fix commit

freakyxue, 2020-03-09:
@PerMalmberg 
lwip fix commit:
076270e7324c3d2f36404601038618317f8f0e74

Alvin1Zhang, 2020-09-23:
Thanks for reporting and sorry for slow turnaround, the fix has been available at https://github.com/espressif/esp-idf/commit/f7d2aa96bd, feel free to reopen if the issue still happens. Thanks.

closed 2020-09-23 by Alvin1Zhang