espressif/esp-idf #3843: Auth does not work correctly in the HTTP Client

ducalex, 2019-07-28:
## Environment

    Development Kit: Lolin32
    Kit version
    Module or chip used: ESP32-WROOM-32D
    IDF version: v3.3-beta1-849-g630ffde6b
    Build System: Make
    Compiler version: crosstool-ng-1.22.0-80-g6c4433a5
    Operating System: Windows
    Power Supply: USB

## Problem Description

A call to `esp_http_client_set_url` discards any username previously set through esp_http_client_init(esp_http_client_config_t). This is especially a problem because `esp_http_client_init` implicitly calls `esp_http_client_set_url` so the only way to use auth with esp-idf's HTTP Client is currently to add them to the URL (http://user:pass@url).

### Expected Behavior
HTTP auth basic/digest to work.

### Actual Behavior
It doesn't

### Steps to repropduce
Commenting those lines in `esp_http_client_set_url` resolves the issue. It does not cause a memory leak as they are freed elsewhere, but it may have other unintended consequences.
````
        free(client->connection_info.username);
        free(client->connection_info.password);
        client->connection_info.username = NULL;
        client->connection_info.password = NULL;
````

### Code to reproduce this issue

```cpp
    esp_http_client_config_t http_config = {};

    http_config.url = "http://pagethatneedsauth.com";
    http_config.username = "admin"; // Will never be used
    http_config.password = "admin";; // Will never be used

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
````
E (5960) HTTP_CLIENT: Error, reach max_redirection_count count=10
E (5960) HTTP_CLIENT: Error response
[E][main.cpp:471] httpRequest(): HTTP: Request failed: ESP_ERR_HTTP_MAX_REDIRECT
````

## Other items if possible

I have found this forum post regarding this issue: https://esp32.com/viewtopic.php?t=6935

negativekelvin, 2019-07-28:
https://github.com/espressif/esp-idf/commit/4a6c5032f4a84a73e0878669201509cc2de3f68a#diff-c7c711c57aa3482655dd39165cf29087

ducalex, 2019-07-28:
@negativekelvin I guess it hasn't made its way to the 3.3 branch then?

mahavirj, 2019-07-29:
@ducalex Thanks for notifying, we will add it to the `release/v3.3`

ducalex, 2019-07-29:
@mahavirj Thank you!

Alvin1Zhang, 2019-10-11:
@ducalex The fix has been merged. Feel free to reopen if you still have the issue. Thanks.

closed 2019-10-11 by Alvin1Zhang