espressif/esp-idf #3844: HTTP client doesn't discard response buffer before a redirection

ducalex, 2019-07-28:
## Environment

Development Kit: Lolin32
Kit version
Module or chip used: ESP32-WROOM-32D
IDF version: v3.3-beta1-849-g630ffde6b
Build System: Make
Compiler version: crosstool-ng-1.22.0-80-g6c4433a5
Operating System: Windows
Power Supply: USB

## Problem Description

When a redirection occurs during an http request the response buffer isn't cleaned, resulting in garbage when calling `esp_http_client_read()`

### Expected Behavior
Only see the content of the last request.

### Actual Behavior
See mangled data, usually content from previous request

### Steps to repropduce

1. Setup an HTTP request to a website that returns 401 or 30X but also outputs some content.
2. Call esp_http_client_perform()
3. Client does two requests as expected.
4. Call esp_http_client_read()

### Code to reproduce this issue

```cpp
Hard to provide, it requires a test server
```
## Debug Logs

```
```

## Other items if possible

boarchuz, 2019-07-28:
It's not zeroed out or null-terminated, if that's what you're expecting. 
The value returned from esp_http_client_read() tells you how many bytes have just been read. There may well be garbage (or old data) in the buffer beyond that range.

ducalex, 2019-07-28:
I do zero terminate, that is not the problem.
The problem is:
I call esp_http_client_perform, which does:
>> Request http://page.com
<< Receive 401, body is "Need auth"
>> Request http://page.com again with authorization header set
<< Receive 200 with empty body

After which I call esp_http_client_read and it returns 9, "Need auth".

I expect it to return 0, as there was no buffer to read from the final request

pdemianczuk07, 2019-12-11:
> After which I call esp_http_client_read and it returns 9, "Need auth".
How you call this function? What were parameteres?

shubhamkulkarni97, 2019-12-27:
@ducalex Can you please share sample code and sdkconfig?

Alvin1Zhang, 2020-01-15:
@ducalex Thanks for reporting. Would you please help share sample code and sdkconfig? Thanks.

mahavirj, 2020-01-22:
Closing due to lack of response, please feel free to re-open if you have any updates.

closed 2020-01-22 by mahavirj