espressif/esp-idf #3845: std::weak_ptr crashes during stress test.

PerMalmberg, 2019-07-28:
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
v4.0-dev-1338-g842432fb6

For various reasons I've found myself using `std::weak_ptr` extensively today. For the same reasons I've also written a stress test to test the functionality I'm using std::weak_ptr for. This code has been running in a Linux system with ASAN enabled for hours and hours without any issues.

When I move the same code over to the ESP32, it sometimes runs a few minutes and then crashes with one of the two stack traces below. Longest run on the ESP32 is ~30 minutes.

There are up to ~400 weak_ptr with a similar number of std::shared_ptr from which they are created at any one time during the test. The test itself is about creating and destroying event queues at run time such that the weak_ptr expires. Somewhat simplified, the test uses two threads, one that creates and destroy X number of receivers, each one owning (shared_ptr) an event queue. Each receiver starts a timer (owned by the second thread) which, via std::weak_ptr, if not expired, pushes items into the event queue for consumption by the first thread.

As seen in the logs below, there are lots of memory free so I don't see this being a out-of-memory issue. And, for the record, I am aware that the smart pointers do not protect what they refer to, only their internal ref-count etc.

The reason I even post this here is the fact that hours of ASAN testing on Linux hasn't shown any issues, while it on ESP32 sometimes crashes after a few minutes. If I remember correctly,  @igrr wrote that Espressif did their own implementation of std::shared_ptr and std::unique_ptr, correct? If so, I assume that std::weak_ptr also has a custom implementation?

What kind of tests are run on shared_ptr/weak_ptr internally at Espressif, if any?

```
I (84376) Worker: User count 300, expire count: 27556
I (92659) MemStat: [INTERNAL]
I (92666) MemStat: 8-bit F:45976 LB:32716 M:35068 | 32-bit: F:67668 LB:32716 M:56740
I (92666) MemStat: [INTERNAL | DMA]
I (92672) MemStat: 8-bit F:45976 LB:32716 M:35068 | 32-bit: F:45976 LB:32716 M:35068
I (92676) MemStat: [SPIRAM]
I (92684) MemStat: 8-bit F:4106876 LB:4099040 M:4093320 | 32-bit: F:4106876 LB:4099040 M:4093320
I (92689) MemStat: Name	Stack	Min free stack
I (92696) MemStat: TimerService	3072	468
I (92700) MemStat: Worker	9000	4612
I (92704) MemStat: MainTask	25600	22764
I (99743) Worker: User count 350, expire count: 29094
I (100353) Worker: User count 300, expire count: 30943
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4019da24  PS      : 0x00060330  A0      : 0x800dd6c0  A1      : 0x3ffcf400  
0x4019da24: __gnu_cxx::__atomic_add(int volatile*, int) at TimerService.cpp:?

A2      : 0xcececed2  A3      : 0x00000001  A4      : 0xcececed2  A5      : 0x3ffb4cb8  
A6      : 0x000000aa  A7      : 0x3ffcf400  A8      : 0x00000001  A9      : 0x3ffb5ef8  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
A14     : 0x0000002a  A15     : 0x3ffcf530  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xcececed2  LBEG    : 0x40090124  LEND    : 0x40090152  LCOUNT  : 0xffffffff  
0x40090124: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x40090152: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 546ff1d91eff865afbd9a822332a1c8d11a87978e00b807d8872922251e11c15

Backtrace: 0x4019da21:0x3ffcf400 0x400dd6bd:0x3ffcf430 0x400de056:0x3ffcf460 0x400dda85:0x3ffcf490 0x400dd7c5:0x3ffcf4c0 0x400dd7e5:0x3ffcf4f0 0x400def2b:0x3ffcf520 0x400de55f:0x3ffcf550 0x400dde3b:0x3ffcf580 0x400dd1d9:0x3ffcf5b0 0x400e56dd:0x3ffcf650 0x400e5185:0x3ffcf760 0x400e5b81:0x3ffcf790 0x400e5a0e:0x3ffcf7c0 0x400e5d5d:0x3ffcf7f0 0x400e5d2d:0x3ffcf820 0x400e5d15:0x3ffcf850 0x40161b25:0x3ffcf880 0x4011ad80:0x3ffcf8a0 0x4009653a:0x3ffcf8d0
0x4019da21: __gnu_cxx::__atomic_add(int volatile*, int) at TimerService.cpp:?

0x400dd6bd: __gnu_cxx::__atomic_add_dispatch(int*, int) at TimerService.cpp:?

0x400de056: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy() at ??:?

0x400dda85: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&) at ??:?

0x400dd7c5: std::__shared_ptr<smooth::core::timer::Timer, (__gnu_cxx::_Lock_policy)2>::__shared_ptr(std::__shared_ptr<smooth::core::timer::Timer, (__gnu_cxx::_Lock_policy)2> const&) at ??:?

0x400dd7e5: std::shared_ptr<smooth::core::timer::Timer>::shared_ptr(std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400def2b: void __gnu_cxx::new_allocator<std::shared_ptr<smooth::core::timer::Timer> >::construct<std::shared_ptr<smooth::core::timer::Timer>, std::shared_ptr<smooth::core::timer::Timer> const&>(std::shared_ptr<smooth::core::timer::Timer>*, std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400de55f: void std::allocator_traits<std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::construct<std::shared_ptr<smooth::core::timer::Timer>, std::shared_ptr<smooth::core::timer::Timer> const&>(std::allocator<std::shared_ptr<smooth::core::timer::Timer> >&, std::shared_ptr<smooth::core::timer::Timer>*, std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400dde3b: std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::push_back(std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400dd1d9: smooth::core::timer::TimerService::tick() at ??:?

0x400e56dd: smooth::core::Task::exec() at ??:?

0x400e5185: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

[... 68 lines cut ...]
0x400dd1c5: smooth::core::timer::TimerService::tick() at ??:?

0x400e5705: smooth::core::Task::exec() at ??:?

0x400e51ad: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

0x400e5ba9: void std::__invoke_impl<void, smooth::core::Task::start()::{lambda()#1}>(std::__invoke_other, smooth::core::Task::start()::{lambda()#1}&&) at Task.cpp:?

0x400e5a36: std::__invoke_result<smooth::core::Task::start()::{lambda()#1}>::type std::__invoke<smooth::core::Task::start()::{lambda()#1}>(std::__invoke_result&&, (smooth::core::Task::start()::{lambda()#1}&&)...) at Task.cpp:?

0x400e5d85: decltype (__invoke((_S_declval<0u>)())) std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) at Task.cpp:?

0x400e5d55: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e5d3d: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x40161b51: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4011adac: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x4009653a: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.

```

igrr, 2019-07-29:
> If I remember correctly, @igrr wrote that Espressif did their own implementation of std::shared_ptr and std::unique_ptr, correct? If so, I assume that std::weak_ptr also has a custom implementation?

No, not really, all these are implemented in libstdc++. We do provide a pthread implementation in IDF, which may or may not be a factor here.

> What kind of tests are run on shared_ptr/weak_ptr internally at Espressif, if any?

We don't have such tests, so if you can share the workload which leads to these crashes we will investigate. Both crashes look like use-after-free and use-uninitialized, which would indicate a race condition of some sort.

By the way, are you using PSRAM? Does the same happen if only internal memory is used?

PerMalmberg, 2019-07-29:
I am using PSRAM, but with the adjusted libc.a your provided in #3624, I forgot to mention that in the OP. I'll disable it and re-run the tests to see if it makes a difference.

The load [test is this](https://github.com/PerMalmberg/Smooth/blob/feature/52-allow-queue-destruction/test/destructing_event_queues/destructing_event_queues.cpp). You can clone that repo and build the tests if you want to, but I think you'd prefer a slimmer version without all the fluff that comes with the rest of my framework? I should be able to put that together this evening.

PerMalmberg, 2019-07-29:
I disabled PSRAM, and then get this trace instead of those in the OP. Just to make sure, I lowered the number of max shared/weak_ptr to 10 just in case it ran out of memory, but the result is the same. Still using the special libc.a.

```
I (211671) MemStat: [INTERNAL]
I (211676) MemStat: 8-bit F:15184 LB:4032 M:12476 | 32-bit: F:64804 LB:49620 M:62076
I (211677) MemStat: [INTERNAL | DMA]
I (211683) MemStat: 8-bit F:15184 LB:4032 M:12476 | 32-bit: F:15184 LB:4032 M:12476
I (211688) MemStat: [SPIRAM]
I (211694) MemStat: 8-bit F:0 LB:0 M:0 | 32-bit: F:0 LB:0 M:0
I (211698) MemStat: Name	Stack	Min free stack
I (211704) MemStat: TimerService	3072	468
I (211708) MemStat: Worker	9000	4620
I (211713) MemStat: MainTask	25600	22748
I (211913) Worker: User count: 11, expire count: 8316
I (212473) Worker: User count: 0, expire count: 8379
I (212978) Worker: User count: 0, expire count: 8379
I (213481) Worker: User count: 0, expire count: 8379
I (213983) Worker: User count: 0, expire count: 8379
I (214485) Worker: User count: 0, expire count: 8379
I (214988) Worker: User count: 0, expire count: 8379
abort() was called at PC 0x40151acf on core 1
0x40151acf: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47

ELF file SHA256: d5436b288e17727fde9ff1e874337f6185f70985fdf0a8d610a537e41185139e

Backtrace: 0x4008e634:0x3ffcc4f0 0x4008ea50:0x3ffcc520 0x40151acf:0x3ffcc550 0x40151b16:0x3ffcc570 0x40151ca7:0x3ffcc590 0x40151b6e:0x3ffcc5b0 0x400dd79e:0x3ffcc5d0 0x400dd48a:0x3ffcc600 0x400dcfe6:0x3ffcc630 0x400dc6ff:0x3ffcc660 0x400dc09f:0x3ffcc6b0 0x400db5e9:0x3ffcc6e0 0x400e256d:0x3ffcc780 0x400e2059:0x3ffcc890 0x400e29a9:0x3ffcc8c0 0x400e286e:0x3ffcc8f0 0x400e2b09:0x3ffcc920 0x400e2ae7:0x3ffcc950 0x400e2ad5:0x3ffcc980 0x401531fd:0x3ffcc9b0 0x4010df44:0x3ffcc9d0 0x4009002a:0x3ffcca00
0x4008e634: invoke_abort at /home/permal/esp/esp-idf/components/esp32/panic.c:155

0x4008ea50: abort at /home/permal/esp/esp-idf/components/esp32/panic.c:172

0x40151acf: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47

0x40151b16: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57

0x40151ca7: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95

0x40151b6e: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54

0x400dd79e: __gnu_cxx::new_allocator<std::shared_ptr<smooth::core::timer::Timer> >::allocate(unsigned int, void const*) at ??:?

0x400dd48a: std::allocator_traits<std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::allocate(std::allocator<std::shared_ptr<smooth::core::timer::Timer> >&, unsigned int) at ??:?

0x400dcfe6: std::_Vector_base<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::_M_allocate(unsigned int) at ??:?

0x400dc6ff: void std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::_M_realloc_insert<std::shared_ptr<smooth::core::timer::Timer> const&>(__gnu_cxx::__normal_iterator<std::shared_ptr<smooth::core::timer::Timer>*, std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > > >, std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400dc09f: std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::push_back(std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400db5e9: smooth::core::timer::TimerService::tick() at ??:?

0x400e256d: smooth::core::Task::exec() at ??:?

0x400e2059: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

0x400e29a9: void std::__invoke_impl<void, smooth::core::Task::start()::{lambda()#1}>(std::__invoke_other, smooth::core::Task::start()::{lambda()#1}&&) at Task.cpp:?

0x400e286e: std::__invoke_result<smooth::core::Task::start()::{lambda()#1}>::type std::__invoke<smooth::core::Task::start()::{lambda()#1}>(std::__invoke_result&&, (smooth::core::Task::start()::{lambda()#1}&&)...) at Task.cpp:?

0x400e2b09: decltype (__invoke((_S_declval<0u>)())) std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) at Task.cpp:?

0x400e2ae7: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e2ad5: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x401531fd: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4010df44: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x4009002a: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.

```

igrr, 2019-07-29:
__cxa_throw inside `operator new` indeed is a failed allocation. One way to confirm this would be to enable C++ exception handling in menuconfig and add a try/catch block to the task function, catching out of memory exceptions.

PerMalmberg, 2019-07-30:
Did a quick test before heading of to work. Noticed too late that SPIRAM was reeneabled again, but its still a bit interesting. I added a `try {} catch(std::exception& ex)` in the base class for my Task classes so I really expected this to be caught since std::bad_alloc inherits from it, but it didn't catch it. The emergency buffer for exception is 0, but as PSRAM is enabled there are 4126956 bytes free on the heap.

[Timer::expired()](https://github.com/PerMalmberg/Smooth/blob/feature/52-allow-queue-destruction/lib/smooth/core/timer/Timer.cpp#L76) does two things - lock a weak_ptr and push the resulting shard_ptr onto a queue, so it is still relating to the same classes.

Not sure what to make of this....

```
I (183247) MemStat: [INTERNAL]
I (183254) MemStat: 8-bit F:48228 LB:32716 M:34260 | 32-bit: F:69944 LB:32716 M:55956
I (183255) MemStat: [INTERNAL | DMA]
I (183260) MemStat: 8-bit F:48228 LB:32716 M:34260 | 32-bit: F:48324 LB:32716 M:34260
I (183265) MemStat: [SPIRAM]
I (183278) MemStat: 8-bit F:4126956 LB:4102388 M:4099192 | 32-bit: F:4126956 LB:4102388 M:4099192
I (183280) MemStat: Name	Stack	Min free stack
I (183287) MemStat: TimerService	3072	828
I (183292) MemStat: Worker	9000	4604
I (183297) MemStat: MainTask	25600	22728
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400dbcd8  PS      : 0x00060830  A0      : 0x800dd315  A1      : 0x3ffd4b60  
0x400dbcd8: smooth::core::timer::Timer::expired() at ??:?

A2      : 0x3f815960  A3      : 0xfefeff12  A4      : 0x3ffb4d48  A5      : 0x3ffb4cb8  
A6      : 0x00000000  A7      : 0x3ffd4b60  A8      : 0x800dbcd1  A9      : 0x3ffd4b30  
A10     : 0x3f815960  A11     : 0x00000001  A12     : 0x3f815960  A13     : 0x0002c46d  
A14     : 0x00000003  A15     : 0x3ffd4b30  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefeff12  LBEG    : 0x4008f810  LEND    : 0x4008f83e  LCOUNT  : 0xffffffff  
0x4008f810: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x4008f83e: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 9856662bf448515681cc2afcbc79c9d046f5144c996a1d0f574e353c6e977770

Backtrace: 0x400dbcd5:0x3ffd4b60 0x400dd312:0x3ffd4be0 0x400e58c9:0x3ffd4cc0 0x400e5359:0x3ffd4e00 0x400e5e31:0x3ffd4e30 0x400e5cbe:0x3ffd4e60 0x400e600d:0x3ffd4e90 0x400e5fdd:0x3ffd4ec0 0x400e5fc5:0x3ffd4ef0 0x40161df1:0x3ffd4f20 0x4011b04c:0x3ffd4f40 0x40096522:0x3ffd4f70
0x400dbcd5: smooth::core::timer::Timer::expired() at ??:?

0x400dd312: smooth::core::timer::TimerService::tick() at ??:?

0x400e58c9: smooth::core::Task::exec() at ??:?

0x400e5359: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

0x400e5e31: void std::__invoke_impl<void, smooth::core::Task::start()::{lambda()#1}>(std::__invoke_other, smooth::core::Task::start()::{lambda()#1}&&) at Task.cpp:?

0x400e5cbe: std::__invoke_result<smooth::core::Task::start()::{lambda()#1}>::type std::__invoke<smooth::core::Task::start()::{lambda()#1}>(std::__invoke_result&&, (smooth::core::Task::start()::{lambda()#1}&&)...) at Task.cpp:?

0x400e600d: decltype (__invoke((_S_declval<0u>)())) std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) at Task.cpp:?

0x400e5fdd: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e5fc5: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x40161df1: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4011b04c: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x40096522: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.

```

PerMalmberg, 2019-08-01:
I've been attempting to write a very scaled down version of my load test, but so far I've not been able to reproduce the issue. Either I'm not properly replicating the load, or there is some error in the original load test that cause the crashes.

Anyway, this is the scaled down test that hasn't crashed yet.

```C++
/* This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <thread>
#include <memory>
#include <esp_pthread.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_log.h>
#include <vector>
#include <mutex>
#include <condition_variable>
#include <esp_log.h>

using namespace std::chrono;

class Item
{
public:
    void f() const noexcept
    {
        ESP_LOGI("Item", "f!");
    }
};

template<typename T>
class Carrier
{
public:
    explicit Carrier(std::weak_ptr<T> p)
            : p(std::move(p))
    {}

    Carrier() = default;

    Carrier(const Carrier&) = default;

    Carrier(Carrier&&) noexcept = default;

    Carrier& operator=(const Carrier&) = default;

    Carrier& operator=(Carrier&&) noexcept = default;

    std::weak_ptr<T> get() const
    {
        return p;
    }

private:
    std::weak_ptr<T> p{};
};

template<typename T>
class Queue
{
public:
    void enqueue(const T& item)
    {
        std::unique_lock<std::mutex> lock{guard};
        q.emplace_back(item);
        cv.notify_all();
    }

    void dequeue(T& p)
    {
        std::unique_lock<std::mutex> lock{guard};
        cv.wait(lock, [&] { return !q.empty(); });

        p = q.front();
        q.erase(q.begin());
    }

    std::size_t size() const
    {
        std::unique_lock<std::mutex> lock{guard};
        return q.size();
    }

private:
    mutable std::mutex guard{};
    std::condition_variable cv;
    std::vector<T> q{};
};

esp_pthread_cfg_t create_config(const char* name, int core_id, int stack, int prio)
{
    auto cfg = esp_pthread_get_default_config();
    cfg.thread_name = name;
    cfg.pin_to_core = core_id;
    cfg.stack_size = stack;
    cfg.prio = prio;
    return cfg;
}

static Queue<Carrier<Item>> queue{};

std::vector<std::shared_ptr<Item>> owner{};

void consumer()
{
    while (true)
    {
        Carrier<Item> c;
        queue.dequeue(c);

        auto item = c.get().lock();
        if (item)
        {
            item->f();
        }
        else
        {
            ESP_LOGI("Expired", "%u", queue.size());
        }
    }
}

void producer()
{
    bool toggle = false;

    while (true)
    {
        if (queue.size() <= 500)
        {
            for (int i = 0;
                 i < 3000;
                 ++i)
            {
                auto item = std::make_shared<Item>();
                owner.emplace_back(item);
                Carrier<Item> c{std::weak_ptr<Item>{item}};
                queue.enqueue(c);
            }
        }
        else if (queue.size() >= 1500)
        {
            // Make every second item expired.
            auto curr = owner.begin();
            while (curr != owner.end())
            {
                auto next = owner.erase(curr);
                if (next != owner.end())
                {
                    curr = ++next;
                }
                std::this_thread::yield();
            }

        }

        std::this_thread::sleep_for(seconds{1});
    }
}

extern "C" void app_main()
{
    auto cfg = create_config("Consumer", 0, 8 * 1024, 5);
    esp_pthread_set_cfg(&cfg);
    std::thread cons(consumer);

    cfg = create_config("Producer", 0, 8 * 1024, 10);
    esp_pthread_set_cfg(&cfg);
    std::thread prod(producer);

    while (true)
    {
        std::this_thread::sleep_for(seconds{1});
    }

}
```

PerMalmberg, 2019-08-02:
I found the cause of the out of memory issue, that one is all on me; fix pushed to the above linked repo.

As to the original shared_ptr/weak_ptr issue, I've found no reason for them to happen. I have however a new stack trace which shows the receiver of the events, as opposed to the previous traces which show the producer side. So the problem appears to happen in both threads.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d85d3  PS      : 0x00060230  A0      : 0x800e588c  A1      : 0x3ffd3950  
0x400d85d3: smooth::core::ipc::TaskEventQueue<smooth::core::timer::TimerExpiredEvent>::forward_to_event_listener() at ??:?

A2      : 0xfefeff06  A3      : 0x3f807af4  A4      : 0x3f807b24  A5      : 0x00000000  
A6      : 0x3ffd38f8  A7      : 0x3ffd3950  A8      : 0x800d85c1  A9      : 0x3ffd3910  
A10     : 0x00000001  A11     : 0x3ffd391c  A12     : 0x3ffd3914  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffd3910  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefeff06  LBEG    : 0x4008fd0c  LEND    : 0x4008fd17  LCOUNT  : 0xffffffff  
0x4008fd0c: memset at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:142

0x4008fd17: memset at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:152

ELF file SHA256: 6af67c91eba7e936b962d650a97e1e92ae916987d7fb42256337a80df9082fa1

Backtrace: 0x400d85d0:0x3ffd3950 0x400e5889:0x3ffd3990 0x400e52e1:0x3ffd3aa0 0x400e5cdd:0x3ffd3ad0 0x400e5b6a:0x3ffd3b00 0x400e5eb9:0x3ffd3b30 0x400e5e89:0x3ffd3b60 0x400e5e71:0x3ffd3b90 0x40161c7d:0x3ffd3bc0 0x4011aed8:0x3ffd3be0 0x40096522:0x3ffd3c10
0x400d85d0: smooth::core::ipc::TaskEventQueue<smooth::core::timer::TimerExpiredEvent>::forward_to_event_listener() at ??:?

0x400e5889: smooth::core::Task::exec() at ??:?

0x400e52e1: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

0x400e5cdd: void std::__invoke_impl<void, smooth::core::Task::start()::{lambda()#1}>(std::__invoke_other, smooth::core::Task::start()::{lambda()#1}&&) at Task.cpp:?

0x400e5b6a: std::__invoke_result<smooth::core::Task::start()::{lambda()#1}>::type std::__invoke<smooth::core::Task::start()::{lambda()#1}>(std::__invoke_result&&, (smooth::core::Task::start()::{lambda()#1}&&)...) at Task.cpp:?

0x400e5eb9: decltype (__invoke((_S_declval<0u>)())) std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) at Task.cpp:?

0x400e5e89: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e5e71: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x40161c7d: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4011aed8: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x40096522: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.

```

PerMalmberg, 2019-08-03:
Did some further testing this morning and made an interesting discovery - even if I disable the destruction of my objects ([L84-88](https://github.com/PerMalmberg/Smooth/blob/feature/52-allow-queue-destruction/test/destructing_event_queues/destructing_event_queues.cpp#L84)) and keep a constant 400 items alive once created, construction of weak_ptr still crashes pretty consistently as per the below trace.

In other words, this means that all the shared_ptr that previously were destructed are now kept intact and that the application is essentially only creating new weak_ptr from these shared_ptr.

What's your take on this, @igrr ?

```
I (102727) MemStat: [INTERNAL]
I (102734) MemStat: 8-bit F:54704 LB:32716 M:54040 | 32-bit: F:76396 LB:32716 M:75712
I (102734) MemStat: [INTERNAL | DMA]
I (102741) MemStat: 8-bit F:54704 LB:32716 M:54040 | 32-bit: F:54704 LB:32716 M:54040
I (102744) MemStat: [SPIRAM]
I (102751) MemStat: 8-bit F:4160568 LB:4139028 M:4129904 | 32-bit: F:4160568 LB:4139028 M:4129904
I (102758) MemStat: Name	Stack	Min free stack
I (102764) MemStat: TimerService	3072	968
I (102769) MemStat: Worker	9000	4664
I (102773) MemStat: MainTask	25600	22764
I (102996) Worker: User count: 401, expire count: 93933
....
I (108552) Worker: User count: 401, expire count: 98938
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4019c550  PS      : 0x00060d30  A0      : 0x800d2210  A1      : 0x3ffcf2f0  
0x4019c550: __gnu_cxx::__atomic_add(int volatile*, int) at destructing_event_queues.cpp:?

A2      : 0xcececed6  A3      : 0x00000001  A4      : 0xcececed6  A5      : 0x3ffcf458  
A6      : 0x000000aa  A7      : 0x3ffcf2f0  A8      : 0x00000001  A9      : 0x3ffb5ef8  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
A14     : 0x0000002a  A15     : 0x3ffcf420  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xcececed6  LBEG    : 0x40090124  LEND    : 0x40090152  LCOUNT  : 0xffffffff  
0x40090124: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x40090152: memcpy at /builds/c7bbafe1/0/idf/newlib-cygwin/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 9d9df888f211fe502b133c17706367b5c2faef8efb2d2cc23680adc1f09c230f

Backtrace: 0x4019c54d:0x3ffcf2f0 0x400d220d:0x3ffcf320 0x400d65e6:0x3ffcf350 0x400da981:0x3ffcf380 0x400da159:0x3ffcf3b0 0x400da179:0x3ffcf3e0 0x400da1b3:0x3ffcf410 0x400d9a93:0x3ffcf440 0x400d95f5:0x3ffcf470 0x400d8ccd:0x3ffcf4a0 0x400d82bd:0x3ffcf4e0 0x400d80e7:0x3ffcf520 0x400dba47:0x3ffcf560 0x400dd22d:0x3ffcf5a0 0x400e5621:0x3ffcf640 0x400e50c9:0x3ffcf750 0x400e5ac5:0x3ffcf780 0x400e5952:0x3ffcf7b0 0x400e5ca1:0x3ffcf7e0 0x400e5c71:0x3ffcf810 0x400e5c59:0x3ffcf840 0x401618ad:0x3ffcf870 0x4011ab08:0x3ffcf890 0x4009653a:0x3ffcf8c0
0x4019c54d: __gnu_cxx::__atomic_add(int volatile*, int) at destructing_event_queues.cpp:?

0x400d220d: __gnu_cxx::__atomic_add_dispatch(int*, int) at destructing_event_queues.cpp:?

0x400d65e6: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_weak_add_ref() at ??:?

0x400da981: std::__weak_count<(__gnu_cxx::_Lock_policy)2>::__weak_count(std::__weak_count<(__gnu_cxx::_Lock_policy)2> const&) at ??:?

0x400da159: std::__weak_ptr<smooth::core::ipc::ITaskEventQueue, (__gnu_cxx::_Lock_policy)2>::__weak_ptr(std::__weak_ptr<smooth::core::ipc::ITaskEventQueue, (__gnu_cxx::_Lock_policy)2> const&) at ??:?

0x400da179: std::weak_ptr<smooth::core::ipc::ITaskEventQueue>::weak_ptr(std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400da1b3: void __gnu_cxx::new_allocator<std::weak_ptr<smooth::core::ipc::ITaskEventQueue> >::construct<std::weak_ptr<smooth::core::ipc::ITaskEventQueue>, std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&>(std::weak_ptr<smooth::core::ipc::ITaskEventQueue>*, std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400d9a93: void std::allocator_traits<std::allocator<std::weak_ptr<smooth::core::ipc::ITaskEventQueue> > >::construct<std::weak_ptr<smooth::core::ipc::ITaskEventQueue>, std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&>(std::allocator<std::weak_ptr<smooth::core::ipc::ITaskEventQueue> >&, std::weak_ptr<smooth::core::ipc::ITaskEventQueue>*, std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400d95f5: std::weak_ptr<smooth::core::ipc::ITaskEventQueue>& std::deque<std::weak_ptr<smooth::core::ipc::ITaskEventQueue>, std::allocator<std::weak_ptr<smooth::core::ipc::ITaskEventQueue> > >::emplace_back<std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&>(std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400d8ccd: smooth::core::ipc::QueueNotification::notify(std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400d82bd: smooth::core::ipc::TaskEventQueue<smooth::core::timer::TimerExpiredEvent>::push_internal(smooth::core::timer::TimerExpiredEvent const&, std::weak_ptr<smooth::core::ipc::ITaskEventQueue> const&) at ??:?

0x400d80e7: smooth::core::ipc::TaskEventQueue<smooth::core::timer::TimerExpiredEvent>::push(smooth::core::timer::TimerExpiredEvent const&) at ??:?

0x400dba47: smooth::core::timer::Timer::expired() at ??:?

0x400dd22d: smooth::core::timer::TimerService::tick() at ??:?

0x400e5621: smooth::core::Task::exec() at ??:?

[... cut ...]
0x400e5c71: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e5c59: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x401618ad: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4011ab08: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x4009653a: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.
```

PerMalmberg, 2019-08-08:
@igrr I appreciative you have lots of other things to do, but have you had a chance to give this some thought?

igrr, 2019-08-08:
@permalmberg sorry for the late reply. Are you able to reproduce the issue when using internal memory only, no PSRAM?

PerMalmberg, 2019-08-08:
@igrr - I wanted to re-confirm before I answered so I've spent the evening running the same code on the same device with the same compiler and IDF and I can't reproduce the issue at all, regardless of PSRAM being enabled or not. Considering the ease of repeatability earlier, this is not what I expected.

Either the (possible) race condition is affected by something else (star alignment?), or something has changed, but I have no idea what if so. I've triple checked that I'm using the same code etc.

I'll let the device run during the night without PSRAM, perhaps that will happen during the night.

(As I've merged some things into the branch I previously referred to I made a new branch off the correct commit and labeled it debugging-shared-weak-ptr.)

PerMalmberg, 2019-08-09:
7h and still no weak/shared_ptr-related crash (PSRAM disabled). The only crash I can create is an out-of-memory when using the full 400 items as per the original code. So I guess the answer is no, I can't reproduce it without PSRAM. Then again, I couldn't reproduce it last night even with PSRAM.

I'll let it run with PSRAM enabled during the day to see if that results in anything.

PerMalmberg, 2019-08-09:
Apparently i crashed 20 minutes after I left with the following. As the same code runs fine with ASAN, I think we can attribute this to PSRAM issues?
```
I (1249373) MemStat: [INTERNAL]
I (1249384) MemStat: 8-bit F:46348 LB:32716 M:33932 | 32-bit: F:68708 LB:32716 M:56356
I (1249387) MemStat: [INTERNAL | DMA]
I (1249396) MemStat: 8-bit F:45552 LB:32716 M:33932 | 32-bit: F:46080 LB:32716 M:33932
I (1249398) MemStat: [SPIRAM]
I (1249414) MemStat: 8-bit F:4175748 LB:4155344 M:4149948 | 32-bit: F:4175748 LB:4155344 M:4149948
I (1249417) MemStat: Name	Stack	Min free stack
I (1249425) MemStat: TimerService	3072	860
I (1249432) MemStat: Worker	9000	4556
I (1249437) MemStat: MainTask	25600	22568
I (1250649) Worker: User count: 300, expire count: 1124914
I (1251978) Worker: User count: 350, expire count: 1126644
I (1253600) Worker: User count: 400, expire count: 1128327
CORRUPT HEAP: Invalid data at 0x3f80a2f8. Expected 0xfefefefe got 0xfefefefd
assertion "ret" failed: file "/home/permal/esp/esp-idf/components/heap/multi_heap_poisoning.c", line 198, function: multi_heap_malloc
abort() was called at PC 0x4017ed5b on core 0
0x4017ed5b: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: dfb5536ef1a7c518ba5663821a3392d203b03a22a49520e336d8b95e2f453dc1

Backtrace: 0x400921ed:0x3ffd49d0 0x40092660:0x3ffd4a00 0x4017ed5b:0x3ffd4a30 0x40090cab:0x3ffd4a60 0x400869b5:0x3ffd4a90 0x40086a3a:0x3ffd4ac0 0x4009591d:0x3ffd4af0 0x401605de:0x3ffd4b20 0x400dfd5a:0x3ffd4b40 0x400df9c1:0x3ffd4b70 0x400df49e:0x3ffd4ba0 0x400deab3:0x3ffd4bd0 0x400de34f:0x3ffd4c20 0x400dd78d:0x3ffd4c50 0x400e5b45:0x3ffd4cf0 0x400e55ed:0x3ffd4e00 0x400e5fe9:0x3ffd4e30 0x400e5e76:0x3ffd4e60 0x400e61c5:0x3ffd4e90 0x400e6195:0x3ffd4ec0 0x400e617d:0x3ffd4ef0 0x40161e01:0x3ffd4f20 0x4011b05c:0x3ffd4f40 0x4009624a:0x3ffd4f70
0x400921ed: invoke_abort at /home/permal/esp/esp-idf/components/esp32/panic.c:155

0x40092660: abort at /home/permal/esp/esp-idf/components/esp32/panic.c:172

0x4017ed5b: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x40090cab: multi_heap_malloc at /home/permal/esp/esp-idf/components/heap/multi_heap_poisoning.c:198 (discriminator 1)

0x400869b5: heap_caps_malloc at /home/permal/esp/esp-idf/components/heap/heap_caps.c:111

0x40086a3a: heap_caps_malloc_default at /home/permal/esp/esp-idf/components/heap/heap_caps.c:150

0x4009591d: malloc at /home/permal/esp/esp-idf/components/newlib/heap.c:32

0x401605de: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:50

0x400dfd5a: __gnu_cxx::new_allocator<std::shared_ptr<smooth::core::timer::Timer> >::allocate(unsigned int, void const*) at ??:?

0x400df9c1: std::allocator_traits<std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::allocate(std::allocator<std::shared_ptr<smooth::core::timer::Timer> >&, unsigned int) at ??:?

0x400df49e: std::_Vector_base<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::_M_allocate(unsigned int) at ??:?

0x400deab3: void std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::_M_realloc_insert<std::shared_ptr<smooth::core::timer::Timer> const&>(__gnu_cxx::__normal_iterator<std::shared_ptr<smooth::core::timer::Timer>*, std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > > >, std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400de34f: std::vector<std::shared_ptr<smooth::core::timer::Timer>, std::allocator<std::shared_ptr<smooth::core::timer::Timer> > >::push_back(std::shared_ptr<smooth::core::timer::Timer> const&) at ??:?

0x400dd78d: smooth::core::timer::TimerService::tick() at ??:?

0x400e5b45: smooth::core::Task::exec() at ??:?

0x400e55ed: smooth::core::Task::start()::{lambda()#1}::operator()() const at Task.cpp:?

0x400e5fe9: void std::__invoke_impl<void, smooth::core::Task::start()::{lambda()#1}>(std::__invoke_other, smooth::core::Task::start()::{lambda()#1}&&) at Task.cpp:?

0x400e5e76: std::__invoke_result<smooth::core::Task::start()::{lambda()#1}>::type std::__invoke<smooth::core::Task::start()::{lambda()#1}>(std::__invoke_result&&, (smooth::core::Task::start()::{lambda()#1}&&)...) at Task.cpp:?

0x400e61c5: decltype (__invoke((_S_declval<0u>)())) std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) at Task.cpp:?

0x400e6195: std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> >::operator()() at Task.cpp:?

0x400e617d: std::thread::_State_impl<std::thread::_Invoker<std::tuple<smooth::core::Task::start()::{lambda()#1}> > >::_M_run() at Task.cpp:?

0x40161e01: execute_native_thread_routine at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/thread.cc:80

0x4011b05c: pthread_task_func at /home/permal/esp/esp-idf/components/pthread/pthread.c:209

0x4009624a: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143

CPU halted.
```

igrr, 2019-08-09:
Yes, the pattern in which all data is corrupted suggests that the PSRAM cache bug to be the likely cause. There is another open issue (linked somewhere above) related to the PSRAM bug, which tracks updating the compiler to use a more reliable workaround. Please subscribe to it if you haven't yet, so that you are notified when a new toolchain version with the fix is available.

PerMalmberg, 2019-08-09:
Thanks for confirming. Do you mean #3624? Yes, I filed that bug too :)

PerMalmberg, 2019-08-21:
I'm closing this as it is no longer reproducable with the new libc.a

closed 2019-08-21 by PerMalmberg