espressif/esp-idf #3846: OTA from github not working

gearsmotion789, 2019-07-28:
I am trying to use github to host my new .bin file for OTA. I am using the Simple Ota example from https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example. I've updated the Server Cert to github (see my readme @ https://github.com/gearsmotion789/OTA/blob/master/README.md). My ota images is located @ https://github.com/gearsmotion789/OTA/blob/master/esp32/thing_shadow.bin.

When I run the example, I get an error. How can I get this to work?

negativekelvin, 2019-07-28:
That is not a direct link to the file, you need to add `?raw=true`

mahavirj, 2019-08-01:
@gearsmotion789 Above mentioned url is not correct file, you may validate same by downloading using `curl/wget` on your host machine.

We have offering https://github.com/espressif/esp-jumpstart/, which has OTA example that uses file hosted on github server, you might want to refer to that: https://docs.espressif.com/projects/esp-jumpstart/en/latest/firmwareupgrade.html

closed 2019-08-01 by mahavirj