espressif/esp-idf #3848: Failing to install esp-idf-tools-setup-2.0.exe

lachlanleslie1995, 2019-07-29:
Hello,

I am trying to install the new command tools that use cmake with the link found on this page:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html

Cloning the repo and such seems to work fine but the installer later fails on this:

2019-07-29 15:14:29.256   Running command: C:\Python27\\python.exe C:\esp-idf\tools\idf_tools.py install
                          Traceback (most recent call last):
                            File "C:\esp-idf\tools\idf_tools.py", line 1256, in <module>
                              main(sys.argv[1:])
                            File "C:\esp-idf\tools\idf_tools.py", line 1252, in main
                              action_func(args)
                            File "C:\esp-idf\tools\idf_tools.py", line 1017, in action_install
                              tools_info = load_tools_info()
                            File "C:\esp-idf\tools\idf_tools.py", line 757, in load_tools_info
                              with open(tool_versions_file_name, 'r') as f:
                          IOError: [Errno 2] No such file or directory: 'lachlan\\ESP32\\esp-idf\\tools/tools.json'

Could anyone suggest a fix for this? It seems to be using the wrong idf_path or something to try access the tools.json file

igrr, 2019-07-29:
When the installer fails, it offers to view the log file. Could you please attach that log here?

lachlanleslie1995, 2019-07-30:
Please see attached log below

[Setup Log 2019-07-29 #009.txt](https://github.com/espressif/esp-idf/files/3444730/Setup.Log.2019-07-29.009.txt)

negativekelvin, 2019-08-02:
https://github.com/espressif/esp-idf/issues/3806

igrr, 2019-08-02:
@negativekelvin I don't think it's related, #3806 was about not being able to change IDF_TOOLS_PATH when reinstalling. This one seems to be related to the IDF path...

@lachlanleslie1995 one thing to try, can you please edit your user PATH environment variable to remove the last entry, which looks suspicious?

```
C:\Users\lachlan\AppData\Local\Continuum\anaconda3\;C:\Users\lachlan\AppData\Local\Programs\Git\cmd\;C:\Program Files (x86)\Google\Chrome\Application;C:\Python27\;C:\Python27\Scripts;C:\Program Files\Espressif\ESP-IDF Tools\mconf-idf;C:\Program Files\Espressif\ESP-IDF Tools\tools\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\doxygen\bin;C:\Windows\System32;C:\Program Files\CMake\bin;C:\Python27\;C:\Python27\Scripts;C:\Program Files\Espressif\ESP-IDF Tools\mconf-idf;C:\Program Files\Espressif\ESP-IDF Tools\tools\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\doxygen\bin;C:\Windows\System32;C:\Program Files\CMake\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\doxygen\bin;C:\Windows\System32;C:\MinGW\msys\1.0\bin;C:\Program Files\doxygen\bin;C:\Windows\System32;C:\Users\lachlan\AppData\Local\Continuum\anaconda3;C:\Users\lachlan\AppData\Local\Continuum\anaconda3\Library\mingw-w64\bin;C:\Users\lachlan\AppData\Local\Continuum\anaconda3\Library\usr\bin;C:\Users\lachlan\AppData\Local\Continuum\anaconda3\Library\bin;C:\Users\lachlan\AppData\Local\Continuum\anaconda3\Scripts;C:\Users\lachlan\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\lachlan\ESP32\esp-idf\Tools\;C:\Users\lachlan\ESP32\esp-idf\Tools\;C:\Users\lachlan\AppData\Local\Programs\Git\cmd; %C:\Users\lachlan\ESP32\esp-idf%\Tools\
```

(specifically, ` %C:\Users\lachlan\ESP32\esp-idf%\Tools\` with space in front of it and odd percent signs inside)

lachlanleslie1995, 2019-08-14:
Resolved thanks @igrr

closed 2019-08-14 by lachlanleslie1995