espressif/esp-idf #3849: Smartconfig stop working

filzek, 2019-07-29:
Smartconfig is broken in DEV, cant find channel and connect to it, it happen from last monday to now. Not sure where it is broken.

This is the example wifi/smartconfig just build and upload to a wrover-b

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6472
load:0x40078000,len:11224
load:0x40080400,len:6648
entry 0x4008076c
I (28) boot: ESP-IDF v4.0-dev-1319-ge36a76d94-dirty 2nd stage bootloader
I (28) boot: compile time 11:11:34
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x159b0 ( 88496) map
I (125) esp_image: segment 1: paddr=0x000259d8 vaddr=0x3ffb0000 size=0x03180 ( 12672) load
I (131) esp_image: segment 2: paddr=0x00028b60 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/dev/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1778

I (132) esp_image: segment 3: paddr=0x00028f68 vaddr=0x40080400 size=0x070a8 ( 28840) load
I (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x6f5f0 (456176) map
0x400d0018: _stext at ??:?

����I (316) esp_image: segment 5: paddr=0x0009f610 vaddr=0x400874a8 size=0x08d88 ( 36232) load
0x400874a8: ppSearchTxframe at ??:?

I (342) boot: Loaded app from partition at offset 0x10000
I (342) boot: Disabling RNG early entropy source...
I (343) cpu_start: Pro cpu up.
I (346) cpu_start: Application information:
I (351) cpu_start: Project name:     smart_config
I (356) cpu_start: App version:      v4.0-dev-1319-ge36a76d94-dirty
I (363) cpu_start: Compile time:     Jul 29 2019 11:38:20
I (369) cpu_start: ELF file SHA256:  1d66a48d3e3f7f8e...
I (375) cpu_start: ESP-IDF:          v4.0-dev-1319-ge36a76d94-dirty
I (382) cpu_start: Starting app cpu, entry point is 0x400810c8
0x400810c8: call_start_cpu1 at C:/Users/dev/Desktop/esp-idf/components/esp32/cpu_start.c:283

I (0) cpu_start: App cpu up.
I (393) heap_init: Initializing. RAM available for dynamic allocation:
I (400) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (406) heap_init: At 3FFB9158 len 00026EA8 (155 KiB): DRAM
I (412) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (418) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (425) heap_init: At 40090230 len 0000FDD0 (63 KiB): IRAM
I (431) cpu_start: Pro cpu start user code
I (449) spi_flash: detected chip: generic
I (450) spi_flash: flash io: dio
W (450) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (460) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (566) wifi: wifi driver task: 3ffc0a2c, prio:23, stack:3584, core=0
I (566) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (566) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (596) wifi: wifi firmware version: af21fdf
I (596) wifi: config NVS flash: enabled
I (596) wifi: config nano formating: disabled
I (596) wifi: Init dynamic tx buffer num: 32
I (596) wifi: Init data frame dynamic rx buffer num: 32
I (606) wifi: Init management frame dynamic rx buffer num: 32
I (606) wifi: Init management short buffer num: 32
I (616) wifi: Init static rx buffer size: 1600
I (616) wifi: Init static rx buffer num: 10
I (626) wifi: Init dynamic rx buffer num: 32
I (716) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (716) wifi: mode : sta (30:ae:a4:c9:d9:60)
I (776) smartconfig: SC version: V2.9.0
I (4876) wifi: ic_enable_sniffer
I (4876) smartconfig: Start to find channel...
I (4876) smartconfig_example: Scan done

filzek, 2019-07-29:
Problem is if you have two AP with same SSID and not same IPs, it will not work!

liuzfesp, 2019-07-30:
HI @filzek, "it happen from last monday to now, Not sure where it is broken.", could you let me know the newest IDF commit that smartconfig has no problem?

filzek, 2019-08-04:
> HI @filzek, "it happen from last monday to now, Not sure where it is broken.", could you let me know the newest IDF commit that smartconfig has no problem?

The problem still but only with some wifi routers, not sure exact why this is happen.

Tested with some dlink, tplink and Mitrastar GPT-2541GNAC,  tplink ac and n models also dlink bg, bgn models.

What is happen is that smartconfig simple cant find anything when attached to some routers.

But, if we start app in other router it works. Seens something related to abstract hw layer for smartconfig sniffer action on wifi.