espressif/esp-idf #3850: esp_https_ota fails with ESP_ERR_INVALID_ARG

IsaackRasmussen, 2019-07-29:
## Environment

- Development Kit:      None/ESP32D0WDQ5
- Kit version: ?
- Module or chip used:  ESP32-SOLO-1
- IDF version v4.0-dev-1287-gd7e659df2-dirty
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

I'm calling esp_https_ota and it fails with ESP_ERR_INVALID_ARG

This is what I see in the console just before the error:

I (4125) IL_Firmware: HTTP_EVENT_ON_CONNECTED
I (4125) IL_Firmware: HTTP_EVENT_HEADER_SENT
I (4145) esp_https_ota: Starting OTA...
I (4145) esp_https_ota: Writing to partition subtype 16 at offset 0x10000
I (7045) esp_https_ota: Connection closed, all data received
I (7045) IL_Firmware: HTTP_EVENT_DISCONNECTED
I (7045) IL_Firmware: HTTP_EVENT_DISCONNECTED
E (7055) IL_Firmware: Upgrading firmware: Failed: ESP_ERR_INVALID_ARG

And I can't figure out what's causing this error.
I enabled HTTP for OTA in the flash config.
And the url is pointing to the latest build firmware. I assume that i can flash with a version that is not newer?

This is my flash partitions:

# Name,   Type, SubType, Offset,  Size
nvs, data, nvs, , 0x4000
otadata, data, ota, , 0x2000
phy_init, data, phy, , 0x1000
ota_0, 0, ota_0, , 0x130000
ota_1, 0, ota_1, , 0x130000
factory,  0,    0,  , 0x130000

And I've executed "make erash_flash"

chegewara, 2019-07-29:
Is it not too fast to download whole binary:
```
I (4145) esp_https_ota: Starting OTA...
I (4145) esp_https_ota: Writing to partition subtype 16 at offset 0x10000
I (7045) esp_https_ota: Connection closed, all data received
```
Also im wondering how is it possible that  your first OTA partition is beginning at 0x10000:
`Writing to partition subtype 16 at offset 0x10000`
because we have partition table at `0x8000`, then according to your partition table you should have 0x8000 + 0x4000 + 0x2000 + 0x1000 = 0x15000

negativekelvin, 2019-07-30:
@chegewara don't forget the length of the partition table itself

negativekelvin, 2019-07-30:
0x8000 + 0x1000 + 0x4000 + 0x2000 + 0x1000 = 0x10000 ( hex numbers go to 15)

chegewara, 2019-07-30:
> 0x8000 + 0x1000 + 0x4000 + 0x2000 + 0x1000 = 0x10000 ( hex numbers go to 15)

You are correct, looks like issue i have with my app exhausted me so much so im having problem with simple math. Sorry for confusion.

IsaackRasmussen, 2019-07-30:
Yes, it turned out to be way too fast a download. I got it working now and it seems like it's downloading about 300bytes at a time. taking about 10-15secs.
So for others with the same error, this could be a clue.

My mistake was silly, I used non standard HTTP/HTTPS ports and forgot to switch port when I went back to HTTP for testing as I couldn't get the certificate working.

Thanks for the help.

closed 2019-07-30 by IsaackRasmussen