espressif/esp-idf #3852: ESP32 simple OTA example failed

huazhenjiang, 2019-07-30:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1287-gd7e659d
- Build System:    CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Windows 10
- Power Supply:         USB

## Problem Description

我下載了esp-idf最新版本(https://github.com/espressif/esp-idf.git)，並使用ESP32-DevKitC進行simple_ota_example的驗證測試(esp32_https_ota)，但是無法成功，只有單純跑example

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. 我編譯了一個example中的hello-world.bin並將其CMakefile中添加set(PROJECT_VER 2)，將其作為ota更新目標的bin，擺放到aws的server上(已確認wget 可以下載)
2. 回到simple_ota_example中，在menuconfig內填上可連接的SSID與其PASSWORD + 預期下載bin的URL
3. 確認menuconfig中config component>mbedTLS>  (16384) TLS maximum message content length >save
4. idf.py clean build erase_flash flash monitor

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
none.

## Debug Logs

I (29) boot: ESP-IDF v4.0-dev-1287-gd7e659d-dirty 2nd stage bootloader
I (29) boot: compile time 16:36:46
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  3 factory          factory app      00 00 00010000 00100000
I (88) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (96) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1dd34 (122164) map
I (165) esp_image: segment 1: paddr=0x0002dd5c vaddr=0x3ffb0000 size=0x022b4 (  8884) load
I (169) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x8e164 (581988) map
0x400d0018: _stext at ??:?

I (379) esp_image: segment 3: paddr=0x000be184 vaddr=0x3ffb22b4 size=0x00f04 (  3844) load
I (381) esp_image: segment 4: paddr=0x000bf090 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/huazhen/Work/ESP32/esp-idf/components/freertos/xtensa_vectors.S:1778

I (387) esp_image: segment 5: paddr=0x000bf498 vaddr=0x40080400 size=0x0ff84 ( 65412) load
I (433) boot: Loaded app from partition at offset 0x10000
I (433) boot: Disabling RNG early entropy source...
I (433) cpu_start: Pro cpu up.
I (437) cpu_start: Application information:
I (442) cpu_start: Project name:     simple_ota
I (447) cpu_start: App version:      1
I (451) cpu_start: Compile time:     Jul 30 2019 09:41:10
I (457) cpu_start: ELF file SHA256:  cace8701225aa792...
I (463) cpu_start: ESP-IDF:          v4.0-dev-1287-gd7e659d-dirty
I (470) cpu_start: Starting app cpu, entry point is 0x400810cc
0x400810cc: call_start_cpu1 at /home/huazhen/Work/ESP32/esp-idf/components/esp32/cpu_start.c:283

I (0) cpu_start: App cpu up.
I (481) heap_init: Initializing. RAM available for dynamic allocation:
I (488) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (494) heap_init: At 3FFB9370 len 00026C90 (155 KiB): DRAM
I (500) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (506) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (513) heap_init: At 40090384 len 0000FC7C (63 KiB): IRAM
I (519) cpu_start: Pro cpu start user code
I (537) spi_flash: detected chip: generic
I (538) spi_flash: flash io: dio
I (538) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (601) wifi: wifi driver task: 3ffc0af4, prio:23, stack:3584, core=0
I (601) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (601) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (801) wifi: wifi firmware version: 3c39e82
I (801) wifi: config NVS flash: enabled
I (801) wifi: config nano formating: disabled
I (801) wifi: Init dynamic tx buffer num: 32
I (801) wifi: Init data frame dynamic rx buffer num: 32
I (811) wifi: Init management frame dynamic rx buffer num: 32
I (811) wifi: Init management short buffer num: 32
I (821) wifi: Init static rx buffer size: 1600
I (821) wifi: Init static rx buffer num: 10
I (821) wifi: Init dynamic rx buffer num: 32
I (831) example_connect: Connecting to test...
W (831) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1081) phy: phy_version: 4101, 100880e, Jul 12 2019, 19:32:06, 0, 2
I (1141) wifi: mode : sta (30:ae:a4:08:93:b8)
I (2481) wifi: new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (2481) wifi: state: init -> auth (b0)
I (2481) wifi: state: auth -> assoc (0)
I (2491) wifi: state: assoc -> run (10)
I (2501) wifi: connected with test, channel 11, 40D, bssid = 78:44:76:f8:b4:44
I (2501) wifi: pm start, type: 1

I (3581) tcpip_adapter: sta ip: 192.168.200.9, mask: 255.255.255.0, gw: 192.168.200.1
I (3581) example_connect: Connected to test
I (3581) example_connect: IPv4 address: 192.168.200.9
I (3591) example_connect: IPv6 address: fe80:0000:0000:0000:32ae:a4ff:fe08:93b8
I (3601) simple_ota_example: Starting OTA example

I (9911) esp_https_ota: Starting OTA...
I (9911) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
E (15271) esp-tls: read error :-76:
E (15271) TRANS_SSL: esp_tls_conn_read error, errno=No more processes
I (15271) esp_https_ota: Connection closed, all data received
E (15281) simple_ota_example: Firmware upgrade failed

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3444957/sdkconfig.zip)
[simple_ota.zip](https://github.com/espressif/esp-idf/files/3444958/simple_ota.zip)

xuhongv, 2019-07-31:
pem证书不正确把

huazhenjiang, 2019-07-31:
> pem证书不正确把
@xuhongv 謝謝您的回覆。

如果是PEM證書不正確，應該是會出現與下方圖片一樣的log吧?
[image]

確實我不曉得如何取得正確的證書，上圖也是我嘗試過錯誤的證書而得到的結果。
但我原始問題中的證書，是藉由openssl s_client -showcerts -connect <file_host_url>去得到的，還是說有其他辦法確認證書的正確與否呢? 謝謝

0anubis0, 2019-08-05:
I have the something similar after updating to master branch. The update was succeeded, but it stucked at the error a few sec. What the error mean? 
I (12641) native_ota_example: esp_ota_begin succeeded
E (27031) esp-tls: read error :-76:
E (27031) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
I (27031) native_ota_example: Connection closed,all data received

huazhenjiang, 2019-08-06:
確實是pem問題

closed 2019-08-06 by huazhenjiang