espressif/esp-idf #3853: NiBLE no longer builds, breaks make

morganrallen, 2019-07-30:
If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta1-470-gd8e4ec91e feature/nimble-preview branch
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
No `make` targets work in latest checkout of `preview/feature-nimble` branch

All make actions are met with `No rule to make target` errors.
```
Toolchain path: /home/morgan/devel/esp-idf/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
make[1]: Entering directory '/home/morgan/devel/esp-idf/tools/kconfig'
make[1]: *** No rule to make target '/home/morgan/devel/esp-idf/tools/kconfig/expand_env.h', needed by 'zconf.tab.o'.  Stop.
make[1]: Leaving directory '/home/morgan/devel/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/morgan/devel/esp-idf/tools/kconfig/conf-idf', needed by '/home/morgan/devel/cacoffirmware/build/include/config/auto.conf'.  Stop.
```

Neither ensuring submodules were all updated, manually deleting build directory helped.

### Expected Behavior
Build proceeds as normal.

### Actual Behavior
No rule errors no matter the target called.

### Steps to repropduce
```
cd $IDF_PATH
git submodule update --init
cd - # or path to project
make
```

dhrishi, 2019-07-30:
Hi @morganrallen , we no longer support the `feature/nimble-preview` branch as NimBLE has been integrated in the ESP-IDF master branch. Please give it a try, it should work fine.

morganrallen, 2019-07-30:
Oh sorry, I got a different build error related to NimBLE in main branch that made me think it was unmerged. Great to hear and I'll check it out this evening.

closed 2019-07-30 by morganrallen