espressif/esp-idf #3854: MCPWM

INTKILOW, 2019-07-30:
mcpwm_gpio_init(MCPWM_UNIT_0, MCPWM0A, GPIO_PWM0A_OUT);

then

initializing mcpwm gpio...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4011a048: ffffffff ffffffff ffffffff
0x4011a048: mcpwm_gpio_init at C:/Users/lenovo/Desktop/esp-idf/components/driver/mcpwm.c:70 (discriminator 8)

Core 0 register dump:
PC      : 0x4011a04e  PS      : 0x00060030  A0      : 0x800d2188  A1      : 0x3ffbb000
0x4011a04e: mcpwm_gpio_init at C:/Users/lenovo/Desktop/esp-idf/components/driver/mcpwm.c:71

A2      : 0x00000021  A3      : 0x00000000  A4      : 0x00000006  A5      : 0x3ff49060
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00002b00  A9      : 0x00000b00
A10     : 0x0000000a  A11     : 0x3ff000c4  A12     : 0x3ffb1968  A13     : 0x06fd1ff8
A14     : 0x3ffbacd0  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9

Backtrace: 0x4011a04e:0x3ffbb000 0x400d2185:0x3ffbb030 0x400d246a:0x3ffbb050 0x400d0912:0x3ffbb070 0x400869b9:0x3ffbb090
0x4011a04e: mcpwm_gpio_init at C:/Users/lenovo/Desktop/esp-idf/components/driver/mcpwm.c:71

0x400d2185: mcpwm_example_gpio_initialize at C:\Users\lenovo\Desktop\mini-fan\build/../main/mini_fan_main.c:240

0x400d246a: app_main at C:\Users\lenovo\Desktop\mini-fan\build/../main/mini_fan_main.c:246

0x400d0912: main_task at C:/Users/lenovo/Desktop/esp-idf/components/esp32/cpu_start.c:497

0x400869b9: vPortTaskWrapper at C:/Users/lenovo/Desktop/esp-idf/components/freertos/port.c:403

negativekelvin, 2019-07-30:
What is the value of GPIO_PWM0A_OUT

INTKILOW, 2019-07-30:
19,16,……

INTKILOW, 2019-07-30:
mcpwm_basic_config demo

negativekelvin, 2019-07-30:
Which board are you using?

INTKILOW, 2019-07-30:
ESP-32-S

koobest, 2019-07-31:
Hi, @INTKILOW 
I don't quite understand why I encountered this problem.  Can you provide your code?

thanks !!

INTKILOW, 2019-07-31:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_basic_config

INTKILOW, 2019-07-31:
LED Control can used

koobest, 2019-07-31:
Don't need any configuration? 
```
#define MCPWM_EN_CARRIER 0   //Make this 1 to test carrier submodule of mcpwm, set high frequency carrier parameters
#define MCPWM_EN_DEADTIME 0  //Make this 1 to test deadtime submodule of mcpwm, set deadtime value and deadtime mode
#define MCPWM_EN_FAULT 0     //Make this 1 to test fault submodule of mcpwm, set action on MCPWM signal on fault occurence like overcurrent, overvoltage, etc
#define MCPWM_EN_SYNC 0      //Make this 1 to test sync submodule of mcpwm, sync timer signals
#define MCPWM_EN_CAPTURE 0   //Make this 1 to test capture submodule of mcpwm, measure time between rising/falling edge of captured signal
#define MCPWM_GPIO_INIT 0    //select which function to use to initialize gpio signals
```

INTKILOW, 2019-07-31:
maybe not，build success

costaud, 2019-10-28:
Please try using the original example code and see whether you would still get this issue. 

From the log, seems a NULL pointer is called.  I guess you copied wrong code from #if/#else branch. Please provide the specific code and sdkconfig you used to test this issue, so that it can be solved sooner.

Alvin1Zhang, 2020-01-19:
@INTKILOW Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-05-28:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-05-28 by Alvin1Zhang