espressif/esp-idf #3856: Sample device name not changing in SPP_SERVER demo

dhavalhparikh, 2019-07-31:
Although I changed the `#define SAMPLE_DEVICE_NAME` from `ESP_SPP_SERVER` to `MY_DEMO` in the spp server demo, it always shows up as `ESP_SPP_SERVER` when I scan for ble devices.

In the example, the device name is being set using the `esp_ble_gap_set_device_name(SAMPLE_DEVICE_NAME)` function and the return code is always `0`.

Any idea why is this happening?
I could not find any examples anywhere to deal with this issue.

Any help is appreciated.
Thank you.

dhrishi, 2019-07-31:
@dhavalhparikh If you see here, the name "ESP_SPP_SERVER" is also being added in the advertisement data (Type 0x09 - complete local name):
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_spp_server/main/ble_spp_server_demo.c#L49

`0x45 0x53 0x50 0x5f 0x53 0x50 0x50 0x5f 0x53 0x45 0x52 0x56 0x45 0x52 = ESP_SPP_SERVER`

So, you need to change this to the name you want

Having said that, when you connect to the server and read the Device Name characteristic in GAP service then you should be able to see the value you have set using `esp_ble_gap_set_device_name`.

dhavalhparikh, 2019-07-31:
@dhrishi Thanks a ton! It totally makes sense!
I really appreciate this.
I'll give this a try but I'm sure it's gonna the way you suggested.

dhavalhparikh, 2019-07-31:
@dhrishi 
I tried changing the advertising data from:
```
static const uint8_t spp_adv_data[23] = {
    0x02,0x01,0x06,
    0x03,0x03,0xF0,0xAB,
    0x0F,0x09,0x45,0x53,0x50,0x5f,0x53,0x50,0x50,0x5f,0x53,0x45,0x52,0x56,0x45,0x52
};
```
to
```
static const uint8_t spp_adv_data[] = {
    0x02,0x01,0x06,
    0x03,0x03,0xF0,0xAB,
    0x0F,0x09,0x54, 0x49, 0x4d, 0x5f, 0x31, 0x32, 0x33, 0x34 // --> TIM_1234
};
```
and I couldn't see the device in the scan list anymore for some reason.
But you were right about the Device name characteristic in GAP service.

What could be the issue here?

dhrishi, 2019-07-31:
@dhavalhparikh you have not updated the length of the advertisement entry as per the new name. The below should work (0x0F -> 0x09)

```
static const uint8_t spp_adv_data[] = {
    0x02,0x01,0x06,
    0x03,0x03,0xF0,0xAB,
    0x09,0x09,0x54, 0x49, 0x4d, 0x5f, 0x31, 0x32, 0x33, 0x34 // --> TIM_1234
};
```

dhrishi, 2019-08-01:
@dhavalhparikh Please close this issue, if it works

dhavalhparikh, 2019-08-01:
@dhrishi I'll close it by tonight. Still need to try and see if it works :)

dhavalhparikh, 2019-08-02:
@dhrishi confirmed. it works!

closed 2019-08-02 by dhavalhparikh