espressif/esp-idf #3857: CoreDump to flash fails in master but works in v3.2.2

dries007, 2019-07-31:
While trying to debug stability issues I've come across an issue with the Core Dump functionality of IDF. Possibly related to #1394 and #1656 but I think I might have a different issue since I have verified it does not happen with IDF release 3.2.2.

## Environment

- Development Kit: I have this problem with all my `ESP-WROVER-KIT v1.4` and `ESP32-Azure-Kit` boards.
- Module: `ESP32-WROVER-B`
- IDF versions: master (`v4.0-dev-1342-g84982ae2e`) and `v3.2.2`
- OS: Windows & Linux
- Power supply: External (1.5A) & USB

## Problem Description

Triggering a core dump via abort causes a watchdog timeout which triggers a reset.
The core dump is not actually written to flash.

Here is a pseudo-stacktrace:

+ `esp_core_dump_flash_write_prepare`
+ `esp_flash_erase_region`
+ `CHECK_WRITE_ADDRESS` -> `memspi_region_protected`
+ `esp_partition_main_flash_region_safe`
+ `esp_ota_get_running_partition`
+ `spi_flash_cache2phys`
+ `spi_flash_protected_read_mmu_entry`
+ `spi_flash_disable_interrupts_caches_and_other_cpu`
+ `esp_ipc_call` -> `esp_ipc_call_and_wait` -> `xSemaphoreTake(s_ipc_ack, portMAX_DELAY);`
+ `xQueueGenericReceive`
+ `vTaskPlaceOnEventList`
+ `vListInsert`
+ Stuck in the `for` loop with the big note `If you find your application is crashing here then likely causes are listed below......`

The problem seems to be that the flash driver is dependent on the OS running, which does not seem to be the case in the abort handler.

### How to reproduce

In both master (`v4.0-dev-1342-g84982ae2e`) and `v3.2.2`:

1. Load the Hello World example.
2. Edit `esp_restart();` to `abort();`
3. Enable core dump to flash in `menuconfig`
4. Optionally: Set debug output to verbose
5. Flash and monitor

## Debug Logs

**Working:**

```
I (4220) esp_core_dump: Save core dump to flash...
D (4225) esp_core_dump: Current task EXIT/PC/PS/A0/SP 32303230 40087768 60530 800879fc 3ffb54c0
D (4234) esp_core_dump: Stack len = 500 (3ffb5400 3ffb55f4)
D (4239) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 400e26aa 60630 800d1ee6 3ffb63e0
D (4248) esp_core_dump: Stack len = 428 (3ffb6320 3ffb64cc)
D (4253) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 400e26aa 60330 800d1ee6 3ffb5c80
D (4261) esp_core_dump: Stack len = 416 (3ffb5bc0 3ffb5d60)
D (4267) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 4008687c 60a30 8008696f 3ffb6e40
D (4275) esp_core_dump: Stack len = 428 (3ffb6d80 3ffb6f2c)
D (4281) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 40084e5d 60030 800d0cd3 3ffafa10
D (4289) esp_core_dump: Stack len = 416 (3ffaf950 3ffafaf0)
D (4294) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 40084e5d 60230 80081e3f 3ffb3ee0
D (4302) esp_core_dump: Stack len = 420 (3ffb3e20 3ffb3fc4)
D (4308) esp_core_dump: Task EXIT/PC/PS/A0/SP 400828bc 40084e5d 60e30 80081e3f 3ffb3970
D (4316) esp_core_dump: Stack len = 424 (3ffb38b0 3ffb3a58)
D (4322) esp_core_dump: Core dump len = 5620 (7 0)
D (4416) esp_core_dump: Dump task 3ffb55fc
D (4417) esp_core_dump: Dump task 3ffb64d4
D (4419) esp_core_dump: Dump task 3ffb5d68
D (4420) esp_core_dump: Dump task 3ffb6f34
D (4422) esp_core_dump: Dump task 3ffafaf8
D (4427) esp_core_dump: Dump task 3ffb3fcc
D (4431) esp_core_dump: Dump task 3ffb3a60
D (4435) esp_core_dump: Data from flash:
D (4438) esp_core_dump: e32c04ed
D (4441) esp_core_dump: 15fc
D (4444) esp_core_dump: 7
D (4446) esp_core_dump: 164
I (4449) esp_core_dump: Core dump has been saved to flash.
Rebooting...
```

**Broken:**
```
I (11748) esp_core_dump_flash: Save core dump to flash...
I (11754) esp_core_dump_common: Found tasks: (7)!
D (11759) esp_core_dump_port: Current task 3ffb58dc EXIT/PC/PS/A0/SP 33396130 40084974 60530 80084d48 3ffb57c0
D (11769) esp_core_dump_port: Stack len = 468 (3ffb5700 3ffb58d4)
D (11775) esp_core_dump_port: Task 3ffb67b4 EXIT/PC/PS/A0/SP 4008231c 400e30f6 60630 800d1d3a 3ffb66c0
D (11784) esp_core_dump_port: Stack len = 428 (3ffb6600 3ffb67ac)
D (11791) esp_core_dump_port: Task 3ffb6048 EXIT/PC/PS/A0/SP 4008231c 400e30f6 60330 800d1d3a 3ffb5f60
D (11800) esp_core_dump_port: Stack len = 416 (3ffb5ea0 3ffb6040)
D (11806) esp_core_dump_port: Task 3ffb7214 EXIT/PC/PS/A0/SP 4008231c 4008945c 60630 8008954b 3ffb7120
D (11816) esp_core_dump_port: Stack len = 428 (3ffb7060 3ffb720c)
D (11822) esp_core_dump_port: Task 3ffafab4 EXIT/PC/PS/A0/SP 4008231c 40087a5d 60030 800d1c77 3ffaf9c0
D (11831) esp_core_dump_port: Stack len = 428 (3ffaf900 3ffafaac)
D (11837) esp_core_dump_port: Task 3ffb415c EXIT/PC/PS/A0/SP 4008231c 40087a5d 60230 80081c13 3ffb4070
D (11847) esp_core_dump_port: Stack len = 420 (3ffb3fb0 3ffb4154)
D (11853) esp_core_dump_port: Task 3ffafd98 EXIT/PC/PS/A0/SP 4008231c 40087a5d 60e30 80081c13 3ffb3c70
D (11862) esp_core_dump_port: Stack len = 416 (3ffb3bb0 3ffb3d50)
D (11868) esp_core_dump_common: Core dump len = 5596 (7 0)
ets Jun  8 2016 00:22:57

rst:0x9 (RTCWDT_SYS_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6512
ho 0 tail 12 room 4
load:0x40078000,len:10980
load:0x40080400,len:6676
entry 0x40080768
W (61) boot: PRO CPU has been reset by WDT.
```

Alvin1Zhang, 2019-08-23:
@dries007 Thanks for reporting this and sorry for getting back to you so late. We are working on this issue, and will share updates later. Thanks.

jstroem, 2019-11-20:
@Alvin1Zhang  any updates on this?

alisitsyn, 2019-11-20:
Hi @jstroem, 
Let me answer to your question. The issue happened because spi_flash component has been updated and functions used by core dump to save data to flash were not placed into IRAM. The spi_flash fix has been prepared but merged later. Could you rebase your code to latest master where the core dump works correctly and is able to save data.

Thanks.

jstroem, 2019-11-22:
@alisitsyn thanks for the update, it seems to be working with the latest rebase on master. 

Thanks!

Alvin1Zhang, 2020-01-15:
@dries007 Thanks for reporting. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-15 by Alvin1Zhang

jakkra, 2020-10-15:
Is this fix backported to release/v4.0? We have avery similar problem. @Alvin1Zhang Can you link to the commit fixing this bug in master branch?

alisitsyn, 2020-10-19:
@jakkra,

As I know it was backported to release/v4.0. The commit ID is: b0bd4d0f149652749e8eb5a9401559bc28090237

AxelLin, 2020-10-20:
The commit id in release/4.0: 399477cd10aa7effde2c3f6bffe5ebc80860a114

alisitsyn, 2020-10-20:
@jakkra,

Feel free to open a new issue if you found something else.

doothez, 2021-01-24:
Hello,
I'm using esp-idf v3.3.2 and having same problem.
So I tried to patch below one but that was for esp-idf v4.0 or higher, so I got below error and could not figure it out.
https://gitlab.unizar.es/761347/esp-idf/commit/e5ff431b35b21fffe486cf6bbaa835d74e66e22a

`[ 83%] Built target idf_component_openssl
[ 83%] Built target idf_component_openssl_sections_info
[ 83%] Built target idf_component_protobuf-c
[ 83%] Built target idf_component_protobuf-c_sections_info
[ 84%] Built target idf_component_spiffs
[ 84%] Built target idf_component_spiffs_sections_info
[ 84%] Built target ldgen_section_infos
[ 85%] Generating esp32.project.ld
linker script generation failed for /Users/sung.lee/Code/esp/bov950_esp/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: Parse error in linker fragment /Users/sung.lee/Code/esp/bov950_esp/amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/linker.lf: error at line 9 col 1 (char 185)
make[2]: *** [amazon-freertos/esp-idf/esp32/esp32.project.ld] Error 1
make[1]: *** [amazon-freertos/esp-idf/esp32/CMakeFiles/ldgen_esp32.project.ld_script.dir/all] Error 2
make: *** [all] Error 2`

Seem like esp-idf v3.3.2 linker setting is not good for the patch.
Are there any backport for v3.3.2 ?

doothez, 2021-01-26:
I figured out why coredump to flash was freezed, since I was using External memory for my task, spi_flash API freezed with external memory stack.
Once I changed to internal memory for all my tasks, it worked well.