espressif/esp-idf #3858: Modbus Master Example can't compile

bonastreyair, 2019-08-01:
## Environment
- Development Kit: ESP32-DevKitC
- Kit version (DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.0-dev-1381-g6fe853a2c
- Build System: Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: macOS
- Power Supply: USB

## Problem Description
Compiling Modbus Master example from master branch does not work. 
I tried with Modbus Slave example and there is no problem on compiling. So it is not a problem on configuration or installation.

### Expected Behavior
Good compiling of modbus_master example

### Actual Behavior
Bad compiling of modbus_master example

### Steps to repropduce
1.Download master branch
2.Compile modbus master example

### Code to reproduce this issue

```
/esp-idf/examples/protocols/modbus_master 
```

## Debug Logs

```
Yairs-MacBook-Pro:modbus_master yair$ make
Toolchain path: /Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /Users/yair/esp/esp-idf/requirements.txt are satisfied.

App "modbus_master_sense" version: v4.0-dev-1381-g6fe853a2c
CXX build/asio/asio/asio/src/asio.o
In file included from /Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /Users/yair/esp/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from /Users/yair/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:2,
                 from /Users/yair/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
make[1]: *** [asio/asio/src/asio.o] Error 1
make: *** [component-asio-build] Error 2
```

negativekelvin, 2019-08-01:
I've seen this error before but cannot reproduce with the example.  I am not on macos though.

alisitsyn, 2019-08-01:
Hi @bonastreyair,

Thank you for this found issue. Are you able to compile it using cmake command:  idf.py build ?
I think it is not related to modbus_master example itself. I will try to find the reason for this but first of all I need to reproduce the issue.

--
Alex

alisitsyn, 2019-08-01:
Hi @bonastreyair,
I was able to reproduce your issue but it is not easy to reproduce. Make sure you updated submodules recursively. Could you execute the commands below and send me the output:
```echo $PATH```
Please check that $PATH includes path to ```$IDF_PATH/tools``` directory, if not please use the command to set it: ```export PATH=$IDF_PATH/tools:$PATH```
Then please type the command: ```idf_tools.py check``` and send me the output.
Also use ```idf.py build``` to compile modbus_master_sense example and then ```make all```. After this your issue should gone. This issue is related to build system but not example. After this we can find the root cause of the issue.
Thanks. 

--
Alex

bonastreyair, 2019-08-02:
Thanks for your quick response! 

After running `./install.sh`  and `. ./export.sh` I have this PATH running `echo $PATH`:
```
/Users/yair/esp/esp-idf/components/esptool_py/esptool:/Users/yair/esp/esp-idf/components/espcoredump:/Users/yair/esp/esp-idf/components/partition_table/:/Users/yair/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin:/Users/yair/.espressif/tools/esp32ulp-elf/2.28.51.20170517/esp32ulp-elf-binutils/bin:/Users/yair/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190313/openocd-esp32/bin:/Users/yair/.espressif/python_env/idf4.0_py2.7_env/bin:/Users/yair/esp/esp-idf/tools:/Library/Frameworks/Python.framework/Versions/3.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/Wireshark.app/Contents/MacOS
```

Executing `idf_tools.py check` these are my results:
```
Checking for installed tools...
Checking tool xtensa-esp32-elf
    version found in PATH: esp32-2019r1-8.2.0
    version installed in tools directory: esp32-2019r1-8.2.0
Checking tool esp32ulp-elf
    version found in PATH: 2.28.51.20170517
    version installed in tools directory: 2.28.51.20170517
Checking tool cmake
    version found in PATH: 3.15.1
Checking tool openocd-esp32
    version found in PATH: v0.10.0-esp32-20190313
    version installed in tools directory: v0.10.0-esp32-20190313
Checking tool ninja
    version found in PATH: 1.9.0
```
I ran `idf.py build` and then `make all` and no errors show.

It seems it compiled fine! Thank you! By the way I also tried to compile with `make` using old compiler 5.2 and it was compiled OK surprisingly!

For your information, I also have the ESP-MDF library (in another directory) installed and whenever I change from one to the other I change my `.bash_profile`
```
# For ESP-MDF
#export IDF_PATH=~/esp/esp-mdf/esp-idf
#export MDF_PATH=~/esp/esp-mdf
#export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
# For ESP-IDF
export IDF_PATH=~/esp/esp-idf
```

I will close the issue thanks!

closed 2019-08-02 by bonastreyair

alisitsyn, 2019-08-02:
Hi @bonastreyair,

It is good that this workaround helps. This is related to update of toolchain and  I will open internal issue to get root reason for this. 
May be it is better to create the symbolic link to esp-idf folder from esp-mdf in your case?

--
Alex

igrr, 2019-08-02:
@bonastreyair The issue happens when you use the new toolchain (GCC 8 based) to compile IDF v3.2 or v3.2. In your case, IDF_PATH was still set to the copy of IDF which was bundled into MDF, which is v3.2. At the same time, example was part of the other copy of IDF (v4.0-dev as indicated in the log).
Now that you have set IDF_PATH to point to the v4.0-dev copy of IDF, compilation is successful.
By the way, when building IDF v4.0-dev with older toolchain, you should receive warnings about incompatible toolchain version. Compilation happens to be successful for now, but this will change in the future.