espressif/esp-idf #3861: Unprovision node

jkockx, 2019-08-01:
I'm trying to unprovison a node which I achieved to provision earlier. To do so I looked in the configuration model api but I can't find exactly how to do so.  Namely, I can't find the correct input for the function esp_ble_mesh_config_client_set_state to reset the node. It's clear to me I have to use the ESP_BLE_MESH_MODEL_OP_NODE_RESET opcode but I can't find the corresponding esp_ble_mesh_cfg_client_set_state_t union member to pass to esp_ble_mesh_config_client_set_state.
Can someone help me out?

Alvin1Zhang, 2019-08-01:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @jkockx

jkockx, 2019-08-01:
OK, I think I have provided as much info as possible. As I think an appropriate esp_ble_mesh_config_client_set_state union member is missing I don't even have some debug log to share.

Alvin1Zhang, 2019-08-01:
@jkockx Thanks for sharing the details. We will look into this soon. Thanks.

WCCWCC, 2019-08-01:
Hi jkockx，

    Currently you need to reboot the device.

    `ESP_BLE_MESH_MODEL_OP_NODE_RESET ` : We will provide this API later.

Campou, 2019-10-11:
Hi @jkockx 

For sending **Config Node Reset** message, because this message has no parameters, hence we only need to set the proper opcode and other information in the parameter `esp_ble_mesh_client_common_param_t`, and for the parameter `esp_ble_mesh_cfg_client_set_state_t`, just define this struct and no parameters need to be changed in it.

Thanks.

Campou, 2020-05-28:
Hi @jkockx 

Sorry for the late reply. Currently when sending Config Node Reset, and `NULL` can be used as the parameter `esp_ble_mesh_cfg_client_set_state_t`, since the parameter of the Config Node Reset is empty.

Thanks.

Alvin1Zhang, 2020-07-27:
Thanks for reporting, will close now, feel free to reopen if the issue still happens. Thanks.

closed 2020-07-27 by Alvin1Zhang