espressif/esp-idf #3862: PROVIDE ( GPIO = 0x60004000 ); in tools/sdk/ld/esp32.peripherals.ld

marcmerlin, 2019-08-01:
I was told to move https://github.com/espressif/arduino-esp32/issues/2571 to this queue.

As per that bug, 
I had to reapply this patch:
tools/sdk/ld/esp32.peripherals.ld
-PROVIDE ( GPIO = 0x3ff44000 );
+PROVIDE ( GPIO = 0x60004000 );

I got it from there https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/master/README.md 'You should patch the tools/sdk/ld/esp32.peripherals.ld to avoid random pixel noise on the display as following
Please refer section 3.3. in https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf for more details."

also present here http://gitlab.studiopresent.rs/ESP32-dotlab/framework/blob/master/tools/sdk/ld/esp32.peripherals.ld

and also found it here https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA

https://qiita.com/h_nari/items/d52734440a7256072e8e when translated by google, says " I encountered GPIO 's unintelligible behavior. After examining the cause, it turned out to be a CPU bug (Eratta)"

I'm not sure if any of those people opened a bug or tried to get it changed, but in case none of them did, I'm filing this bug.
I am not knowledgeable about why this fix is needed, although the first and last link does give details that should help

negativekelvin, 2019-08-01:
> Details: 
> Some ESP32 peripherals are mapped to two internal memory buses (AHB & DPORT). 
> When written via DPORT, consecutive writes to the same address may be lost. 
> Workarounds: 
> This issue is automatically worked around in the drivers of ESP-IDF V1.0 and newer. 
> Workaround Details: 
> When writing the same register address (i.e., FIFO-like addresses) in sequential instructions, 
> use the equivalent AHB address not the DPORT address. 
> (For other kinds of register writes, using DPORT registers will give better write 
> performance.)
> Fixes: 
> This issue is fixed in silicon revision 1.

So this should only affect old rev0 chips and only consecutive writes to the same address. Because of this and the fact there is a performance trade-off, it is better to selectively work around this issue instead of applying a blanket fix.  This is what esp-idf drivers do.

Can you explain what problem this resolved for you?

marcmerlin, 2019-08-02:
@negativekelvin people using ESP32 to run RGBPanels via I2S DMA seem to have timing problems without it, I gave the links in my original message. For instance

Patching GPIOhttps://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/master/README.md to avoid eratta of ESP32
You should patch the tools/sdk/ld/esp32.peripherals.ld to avoid random pixel noise on the display as following:

/* PROVIDE ( GPIO = 0x3ff44000 ); */
PROVIDE ( GPIO = 0x60004000 );

negativekelvin, 2019-08-02:
Ok but have you confirmed this yourself and what do GPIO registers have to do with I2S DMA?

For instance https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/blob/master/README.md mostly uses REG_WRITE so changing the GPIO struct address would have no effect.

marcmerlin, 2019-08-02:
@negativekelvin  no, I did not. I just used multiple projects that told me I had to change this, so I did.
It might be possible they are all wrong and have been passing around an obsolete fix. If you think that's the case, feel free to close this.

projectgus, 2019-10-25:
Hi @marcmerlin ,

@negativekelvin is right that this is only a problem on ESP32 rev0, and only a problem with very specific workloads - writing to the exact same DPORT register twice in consecutive instructions.

I don't see anywhere in the code where this would happen, but it's not inconceivable that it could have happened at some point. Also, as @negativekelvin  points out, the library also uses register macros and for these the suggested "fix" would have no effect.

This advice looks like it was added in the initial commit 2 years ago, so my guess is that it was copied over from some other codebase where it may have been necessary on the rev0 chip:
https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/commit/6143785bc7c0e88439e0ef58a590d3a8f6d25677#diff-04c6e90faac2675aa89e2176d2eec7d8R14

If you do find a situation where changing this address fixes a problem, please let us know and we will reopen.

closed 2019-10-25 by projectgus

marcmerlin, 2019-10-25:
thanks for the look/review @projectgus , this wouldn't be the first time that people have been copying over an unnecessary workaround.