espressif/esp-idf #3863: Heap corruption with mbedtls mbedtls_pk_decrypt

nikojarvinen, 2019-08-01:
## Environment

- Development Kit:      [ESP32-Wrover-Kit}
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     I'm on windows 10 and running ubuntu on WSL. ESP-IDF is on 4.0-dev-1287
- Power Supply:         USB

I'm trying to build a simple script that would encrypt and then decrypt a string using private/public keys. It's built on the hello world example and only the main c file is modified. The encryption part seems to work ok but the decryption part crashes esp32. I've been trying anything I can think of but can't get it to work for some reason. 

The full code to replicate this is below in GIST. Just flash it, no other steps needed.
The line where it seems to be crashing is: 
if( ( ret = mbedtls_pk_decrypt( &pk2, buf, olen, result, &olen2, sizeof(result), mbedtls_ctr_drbg_random, &ctr_drbg ) ) != 0 )   {

https://gist.github.com/nikojarvinen/923f0d5c892015824c20cf94ff5b84cb

Errors/dump below:

```
--- idf_monitor on /dev/ttyS9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
RVets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6440
load:0x40078000,len:11224
load:0x40080400,len:6648
entry 0x4008076c
I (29) boot: ESP-IDF v4.0-dev-1287-gd7e659df2 2nd stage bootloader
I (29) boot: compile time 17:55:01
I (33) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x08d54 ( 36180) map
I (107) esp_image: segment 1: paddr=0x00018d7c vaddr=0x3ffb0000 size=0x021a8 (  8616) load
I (110) esp_image: segment 2: paddr=0x0001af2c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/testing/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (114) esp_image: segment 3: paddr=0x0001b334 vaddr=0x40080400 size=0x04cdc ( 19676) load
I (131) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x23e84 (147076) map
0x400d0018: _stext at ??:?

I (184) esp_image: segment 5: paddr=0x00043ea4 vaddr=0x400850dc size=0x0481c ( 18460) load
0x400850dc: memspi_host_read_id_hs at /home/testing/esp/esp-idf/components/spi_flash/memspi_host_driver.c:61

I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
I (199) cpu_start: Pro cpu up.
I (202) cpu_start: Application information:
I (207) cpu_start: Project name:     hello-world
I (213) cpu_start: App version:      1
I (217) cpu_start: Compile time:     Aug  1 2019 17:55:02
I (223) cpu_start: ELF file SHA256:  05fa92f95111edc2...
I (229) cpu_start: ESP-IDF:          v4.0-dev-1287-gd7e659df2
[... 108 lines cut ...]
0x400d757d: mbedtls_calloc at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/platform.c:66

0x400dc87e: mbedtls_mpi_grow at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:132

0x400dd997: mbedtls_mpi_exp_mod at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1897 (discriminator 16)

0x400d7781: rsa_prepare_blinding at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:798

0x400d7dca: mbedtls_rsa_private at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:926

0x400d8495: mbedtls_rsa_rsaes_pkcs1_v15_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:1597 (discriminator 1)

0x400d861d: mbedtls_rsa_pkcs1_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:1735

0x400d651a: rsa_decrypt_wrap at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:132

0x400d63ef: mbedtls_pk_decrypt at /home/testing/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:439

0x400d3ffc: app_main at /home/testing/esp/hello_world/build/../main/hello_world_main.c:137

0x400d0fca: main_task at /home/testing/esp/esp-idf/components/esp32/cpu_start.c:549

0x40086fd9: vPortTaskWrapper at /home/testing/esp/esp-idf/components/freertos/port.c:143

Rebooting...```

nikojarvinen, 2019-08-01:
Also tried with SP-IDF v4.0-dev-1381-g6fe853a2c and it results in the same issue.

negativekelvin, 2019-08-01:
Try increasing your main task stack size

nikojarvinen, 2019-08-02:
Thanks for the pointer. That did solve the issue. Need to learn more about that concept.

closed 2019-08-02 by nikojarvinen