espressif/esp-idf #3864: CMake compilation fails when using CONFIG_SECURE_SIGNED_ON_UPDATE

skandalfo, 2019-08-01:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):  v4.0-dev-1381-g6fe853a2c
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp32-2019r1-rc2
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

When trying to enable signature verification on OTA updates, my project fails to link when built with CMake. The Makefile variant works.

This is the error I obtain:

```
/home/skandalfo/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures.c.obj):(.literal.esp_secure_boot_verify_signature_block+0x0): undefined reference to `_binary_signature_verification_key_bin_end'
/home/skandalfo/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures.c.obj):(.literal.esp_secure_boot_verify_signature_block+0x4): undefined reference to `_binary_signature_verification_key_bin_start'
```

### Expected Behavior

It should just work as expected (build and execute properly).

### Actual Behavior

Linking fails because the binary data for the signature verification key doesn't get linked into the application binary.

My understanding is that, in the `bootloader_support` module, `esp_image_format.c` in the application binary needs to call into the signature verification code only if CONFIG_SECURE_SIGNED_ON_UPDATE is set. At that point the functions and references in `idf/secure_boot_signatures.c` end up linked into the binary.

The CMakeLists.txt for `bootloader_support` only enables the generation of the signature and its linking if it's building the bootloader, though, via [this conditional](https://github.com/espressif/esp-idf/blob/6fe853a2c73437f74c0e6e79f9b15db68b231d32/components/bootloader_support/CMakeLists.txt#L39):

```
if(BOOTLOADER_BUILD AND CONFIG_SECURE_SIGNED_APPS)
```

Replacing this with this extended conditional:

```
if(BOOTLOADER_BUILD AND CONFIG_SECURE_SIGNED_APPS OR CONFIG_SECURE_SIGNED_ON_UPDATE)
```

Is enough to get the example working with the specified configuration.

### Steps to repropduce

1. Just try to compile the advanced_https_ota example with CMake with CONFIG_SECURE_SIGNED_ON_UPDATE selected in menuconfig.

hex55, 2019-10-07:
The same!

mahavirj, 2019-11-14:
Should be fixed with https://github.com/espressif/esp-idf/commit/75488f18060553d7853e3c93cb6264d112b8eb07, can you please try on latest `master` branch once?

skandalfo, 2019-11-15:
Confirmed it's now working with v4.1-dev-997-gd57890cdf

Thanks for the fix!

closed 2019-11-15 by skandalfo

TonyTrupp, 2019-11-29:
@mahavirj, is there any way a similar fix could be included within a 3.3.1 patch?  I'm not able to update beyond 3.3, since I'm using Arduino-ESP32, which doesn't support 4.0 yet, and unfortunately that fix didn't merge cleanly into 3.3. 

Since this thread is already closed, I created a new ticket here with that request: 
https://github.com/espressif/esp-idf/issues/4419 

Thank you.