espressif/esp-idf #3865: Exception in psram_gpio_config

jeremyherbert, 2019-08-02:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         External 3.3V

## Problem Description

Crash occurs on startup in psram_gpio_config. Unfortunately the application is quite complicated and proprietary, and so I'm still trying to work out how to provide some sort of reproduceable code.

## Debug Logs

```
I (1182) psram: This chip is ESP32-D0WD
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40082de3  PS      : 0x00060b30  A0      : 0x80084759  A1      : 0x3ffe3b60  
0x40082de3: psram_gpio_config at /home/jeremy/esp/esp-idf/components/esp32/spiram_psram.c:549 (discriminator 6)

A2      : 0x3ffe3b94  A3      : 0x4aa52ec2  A4      : 0x8a952ec2  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0xfffffffa  A8      : 0x00013ffc  A9      : 0x3ffe3b40  
A10     : 0x00000000  A11     : 0x3ff42000  A12     : 0x00000040  A13     : 0x3ff420c0  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8a952ec2  LBEG    : 0x400929a4  LEND    : 0x400929af  LCOUNT  : 0x00000000  
0x400929a4: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:133

0x400929af: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143

ELF file SHA256: 8a1b751dd9af4656d58a0662cd602d60b042277517b48e3fd26ae259dc898452

Backtrace: 0x40082de0:0x3ffe3b60 0x40084756:0x3ffe3b90 0x400d20f4:0x3ffe3bc0 0x40081773:0x3ffe3bf0 0x400795a7:0x3ffe3c30 |<-CORRUPTED
0x40082de0: psram_gpio_config at /home/jeremy/esp/esp-idf/components/esp32/spiram_psram.c:549 (discriminator 6)

0x40084756: psram_enable at /home/jeremy/esp/esp-idf/components/esp32/spiram_psram.c:717

0x400d20f4: esp_spiram_init at /home/jeremy/esp/esp-idf/components/esp32/spiram.c:148

0x40081773: call_start_cpu0 at /home/jeremy/esp/esp-idf/components/esp32/cpu_start.c:162

```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3459119/sdkconfig.txt)

jeremyherbert, 2019-08-02:
Using the sdkconfig file with the `hello_world` application causes the same crash. It also seems to not be a problem if I disable flash encryption and secure boot.

projectgus, 2019-08-02:
@jeremyherbert Is it possible something is physically connected to an ESP32 GPIO that is shared with the PSRAM or the ESP32 flash chip?

jeremyherbert, 2019-08-02:
I don't think so, because I tried this on three separate brand new unflashed PCBs, and this same design has been shipped before. I will continue to investigate. I also tried updating to the latest git rev but same result.

stuarthatchbaby, 2019-08-14:
We're hitting a similar issue with v3.3rc - specifically OTA updates cause the same type of error.

In our case I think it's because our bootloader is built under 3.3beta-1. I'm guessing there's something wrong with the startup code that's dealing with backward compatibility. There are so many permutations and combinations it's difficult to follow.

jeremyherbert, 2019-08-15:
FYI when I revert back to 1f9f39fe4 issue is no longer present

costaud, 2019-08-15:
Yes, there is a bug in psram_gpio_config.
Just to make sure that are you using DIO mode for SPI flash with psram and running an older version of bootloader ?
For a quick fix, 

```c
static void IRAM_ATTR psram_gpio_config(psram_io_t *psram_io, psram_cache_mode_t mode)
{
    int spi_cache_dummy = 0;
    uint32_t rd_mode_reg = READ_PERI_REG(SPI_CTRL_REG(0));
    if (rd_mode_reg & SPI_FREAD_QIO_M) {
        spi_cache_dummy = SPI0_R_QIO_DUMMY_CYCLELEN;
    } else if (rd_mode_reg & SPI_FREAD_DIO_M) {
        spi_cache_dummy = SPI0_R_DIO_DUMMY_CYCLELEN;
        SET_PERI_REG_BITS(SPI_USER1_REG(0), SPI_USR_ADDR_BITLEN_V, SPI0_R_DIO_ADDR_BITSLEN, SPI_USR_ADDR_BITLEN_S);
    }  
    .........
```

Will merge it and sync soon.

jeremyherbert, 2019-08-15:
For me it was a clean flash, no old bootloader. But yes DIO for flash.

stuarthatchbaby, 2019-08-15:
The fix above works for me too, at least at first blush.

projectgus, 2019-08-15:
@jeremyherbert Does the above fix also fix the issue for you, despite not using an older bootloader?

vonnieda, 2019-08-19:
I seem to be experiencing a related and similar issue, but from the opposite direction. For me, if I flash a board with 3.2 and then try to OTA a firmware built with 3.0 I get the above failure. I've filed #3932 for this.

jeremyherbert, 2019-09-22:
I can't reproduce this bug anymore using the latest version, so I am assuming it is fixed.

closed 2019-09-22 by jeremyherbert