espressif/esp-idf #3866: Compile Error dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t'

kvbulusu, 2019-08-02:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-1381-g6fe853a2c
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Linux
Power Supply: USB

Problem Description:
Error while compiling.
esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c: In function 'copy_ble_config':
/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:57:28: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t' {aka 'struct protocomm_ble_config'}
     memcpy(dest_config->ble->device_name,
                            ^~
/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:58:27: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t' {aka 'struct protocomm_ble_config'}
            src_config->ble->device_name,
                           ^~
make[1]: *** [/esp-idf/make/component_wrapper.mk:292: src/esp_local_ctrl_transport_ble.o] Error 1
make: *** [/esp-idf/make/project.mk:584: component-esp_local_ctrl-build] Error 2

mahavirj, 2019-08-06:
@kvbulusu Can you please share your `sdkconfig` file?

kvbulusu, 2019-08-06:
> @kvbulusu Can you please share your `sdkconfig` file?
Attached..
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3473774/sdkconfig.txt)

mahavirj, 2019-08-07:
@kvbulusu 

- Found that your `sdkconfig` has `CONFIG_ESP_HTTPS_SERVER_ENABLE` option disabled, any specific reason? (Example will not compile without this, refer https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_local_ctrl/sdkconfig.defaults). Once I enabled this, I did not encounter any build issues.
- Interim I noticed few compilation issue with `gdbstub` enabled and toolchain `esp32-2019r1`, we will ensure that this gets fixed soon

kvbulusu, 2019-08-08:
@mahavirj  
Thanks.  I'm not using the example and without that config my application used to compile.  But ofcourse I can enable it . will try.

For gdbstub, just FYI. I opened another issue  here .. hope it will be resolved soon https://github.com/espressif/esp-idf/issues/3834#issuecomment-515301149

kvbulusu, 2019-08-11:
@mahavirj   I still cant compile with the config enabled.....Its slightly different error now..
In file included from dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:20:
dev/esp-va-sdk/components/protocomm/include/transports/protocomm_ble.h:75:78: note: expected 'const protocomm_ble_config_t *' {aka 'const struct <anonymous> *'} but argument is of type 'esp_local_ctrl_transport_config_ble_t * const' {aka 'struct protocomm_ble_config * const'}
 esp_err_t protocomm_ble_start(protocomm_t *pc, const protocomm_ble_config_t *config);
                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c: In function 'copy_ble_config':
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:57:28: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t' {aka 'struct protocomm_ble_config'}
     memcpy(dest_config->ble->device_name,
                            ^~
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:58:27: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t' {aka 'struct protocomm_ble_config'}
            src_config->ble->device_name,
                           ^~
dev/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/esp_local_ctrl_transport_ble.o' failed
make[1]: *** [src/esp_local_ctrl_transport_ble.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC build/expat/expat/expat/lib/xmlparse.o
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c: In function 'start_httpd_transport':
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:75:10: error: 'protocomm_httpd_config_t' {aka 'struct <anonymous>'} has no member named 'ext_handle_provided'
         .ext_handle_provided = true,
          ^~~~~~~~~~~~~~~~~~~
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:76:10: error: 'protocomm_httpd_config_t' {aka 'struct <anonymous>'} has no member named 'data'
         .data = {
          ^~~~
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:76:9: warning: braces around scalar initializer
         .data = {
         ^
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:76:9: note: (near initialization for 'pc_conf.stack_size')
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:77:13: error: field name not in record or union initializer
             .handle = &server_handle
             ^
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:77:13: note: (near initialization for 'pc_conf.stack_size')
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:77:23: warning: initialization of 'size_t' {aka 'unsigned int'} from 'void **' makes integer from pointer without a cast [-Wint-conversion]
             .handle = &server_handle
                       ^
dev/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_httpd.c:77:23: note: (near initialization for 'pc_conf.stack_size')
CC build/esp_wifi/src/smartconfig.o
dev/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/esp_local_ctrl_transport_httpd.o' failed
make[1]: *** [src/esp_local_ctrl_transport_httpd.o] Error 1

anurag-kar, 2019-08-12:
@kvbulusu Seems like there is some issue with the local clone of IDF. These new errors could only happen if you are using a very old (older than v3.2 release) master branch. Are use sure you don't have multiple IDF local clones, and if you do, then is your `IDF_PATH` environment variable set correctly to the one with the latest master?

kvbulusu, 2019-08-13:
@anurag-kar  I have 3.2.2 and the latest dev branch (v4.0-dev-1443-g39f090a4f
) .. ...IDF_PATH is set correctly...Tried setting up from scratch multiple times and still see the issue...Not sure what else to check for...

mahavirj, 2019-08-19:
@kvbulusu Can you please help to build example `https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_local_ctrl` from your IDF?  This example uses `esp_local_ctrl` component, so should help us to identify if there is anything missed out in setup/clone etc.

mahavirj, 2019-08-21:
Reopening this issue, above commit only addresses part of the fix, still final confirmation required from @kvbulusu

mahavirj, 2019-09-09:
@kvbulusu Did you get chance to try above suggestion?

kvbulusu, 2019-09-09:
@mahavirj  Thanks for the patience. We can close the issue. I was using esp-va-sdk and its based on v3.2.2 . It needs to be updated to be used with v4 (dev branch).

closed 2019-09-09 by kvbulusu

IgorJeri, 2021-09-04:
I ran into this issue today, what is the status of esp-va-sdk ?
Is it deprecated or will it be updated to support the latest version of espIDF?