espressif/esp-idf #3867: Error in function esp_wifi_init after merged from v3.3

zavovi, 2019-08-02:
Hello,
please, could you help me? I was merged the v3.3 branch into my branch (v3.2) and now I am getting this error:

I (1979) net80211: OS adapter function version error! Version 1 is expected, but it is 2

And function `esp_wifi_init `returning code 0x102

This is my initial code which worked in v3.2:

	```
esp_err_t esp_err;
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

	esp_err = esp_event_loop_init(low_wifi_event_handler, NULL);
	if(esp_err != ESP_OK)
		return TCS_ERR_WIFI;

	tcpip_adapter_init();

	/* Initialize ESP WiFi */
	esp_err = esp_wifi_init(&cfg);
	if(esp_err != ESP_OK)
	{
		TCS_ERROR_PRINTF(LOG_CATEGORY, "Error in esp_wifi_init: %d", esp_err);
		return TCS_ERR_WIFI;
	}

```

Please, could you help me, where is difference between v3.2 and v3.3?
Thank you very much.

liuzfesp, 2019-08-02:
Hi @zavovi, do you update your submodule after upgrade your release?

zavovi, 2019-08-02:
Yes, I updated all and recompiled all.

liuzfesp, 2019-08-29:
Hi @zavovi, If this is a real problem, I think everyone should have problem. So I suspect this issue is related to your update process. Could you tell me the detailed steps to merge v3.3 to v3.2? Or you can clone a new v3.3 and try it again?

zavovi, 2019-08-29:
Hi @liuzfesp, It seems, that there is some issue with merging. I have my branch from v3.2 and I make some minor changes in one file. After that, I merged the v3.3 onto my brach. And there is the issue...

Alvin1Zhang, 2019-08-30:
@zavovi Thanks for reporting this. Seemingly the issue might be in your updating process/merging, right? Would you please help provide more details? Thanks.

zavovi, 2019-09-02:
Hi @Alvin1Zhang ,
Yes, it seems, that was in my merging process, because I cannot merge v3.3 into v3.2 - this is the right issue. Now, I made new branch from v3.3 and updated my changes via patch. Now, it is working.

Thank you.

closed 2019-09-02 by zavovi