espressif/esp-idf #3868: MQTT limit buffer to 1 message

jkockx, 2019-08-02:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
For my application, I'm receiving a continuous stream of MQTT data and I can't find a way to limit the buffering in the mqtt library to prevend lag of my application (level sets of a light bulb). If I simply decrease the size of my buffer in the configuration of the mqtt in my code or with the menuconfig, I end up with an unhandled exception when the buffer becomes to small where I would like the messages to be dropped. The log output on the exception is given below.

## Debug Logs

`D (36286) MQTT_CLIENT: Get data len= 1, topic len=0, total_data: 43 offset: 42
I (36286) MQTT: MQTT_EVENT_DATA
TOPIC=
DATA=)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401041c2  PS      : 0x00060830  A0      : 0x80104a11  A1      : 0x3ffdc000  
0x401041c2: generic_model_onoff_set at /home/programmer/esp/gateway_mqtt/main/app_main.c:1164

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x0000001f  
A6      : 0x00000001  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffdbcc0  
A10     : 0x00000008  A11     : 0xffffffff  A12     : 0x00000001  A13     : 0x3ffdbed0  
A14     : 0x00000008  A15     : 0x3ffdbed0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 4f1fc6d3268ba0beb7589ad30c23ae1588ab54c17378e77d88d9a9e3d8bc15c1

Backtrace: 0x401041bf:0x3ffdc000 0x40104a0e:0x3ffdc020 0x40104b31:0x3ffdc050 0x401db249:0x3ffdc080 0x401db75f:0x3ffdc0c0 0x40105389:0x3ffdc110 0x401054f8:0x3ffdc130 0x401056de:0x3ffdc170 0x40105aa8:0x3ffdc1a0 0x400947dd:0x3ffdc1e0
0x401041bf: generic_model_onoff_set at /home/programmer/esp/gateway_mqtt/main/app_main.c:1163

0x40104a0e: mqtt_event_handler_cb at /home/programmer/esp/gateway_mqtt/main/app_main.c:289

0x40104b31: mqtt_event_handler at /home/programmer/esp/gateway_mqtt/main/app_main.c:323

0x401db249: handler_execute at /home/programmer/esp/esp-idf/components/esp_event/esp_event.c:145

0x401db75f: esp_event_loop_run at /home/programmer/esp/esp-idf/components/esp_event/esp_event.c:545 (discriminator 3)

0x40105389: esp_mqtt_dispatch_event at /home/programmer/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:575

0x401054f8: deliver_publish at /home/programmer/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:619 (discriminator 4)

0x401056de: mqtt_process_receive at /home/programmer/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:876 (discriminator 6)

0x40105aa8: esp_mqtt_task at /home/programmer/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1008

0x400947dd: vPortTaskWrapper at /home/programmer/esp/esp-idf/components/freertos/port.c:143

Rebooting...
`

jkockx, 2019-08-02:
Besides, is it normal that my esp32 is only capable of processing something like 1 mqtt message per second? That was the reason I wanted to make these changes to the buffer in the first place.

david-cermak, 2019-08-02:
Hi @jkockx 

I'm afraid I don't quite understand your application needs. Did you mean a mode in which would library drop a message longer than x bytes? 

Also looking at this exception it seems to receive correctly 1 remaining byte of a message (event prints it `')'` and then crashes in `generic_model_onoff_set` which looks like an application call. Would the crash go away if you removed this call in event callback?

About 1 message per second: This of course depend on connection, but should usually work much faster. Would you mind trying to use the old style event callback instead of event loop and see if that is any different? for example:
```
static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)
{ ... }

    const esp_mqtt_client_config_t mqtt_cfg = {
        .event_handle = mqtt_event_handler,
    }
...
// REMOVE:    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, client);
```

jkockx, 2019-08-02:
OK, thanks for your reponse. your were right that the crash did originate somewhere else. Besides I have found another reason for the slow operation was the extremely slow eclipse mqtt server. Though, I have some issues left for my application.  If I pull up the rate of values send to my device with mqtt my buffer start filling up and causes a delay.  For this reason,  I'm wondering if I could minimize the buffer size and drop incoming messages when the buffer is full.

david-cermak, 2019-08-03:
It's not possible to drop a message after reading certain size (or theoretically yes, you can simply trigger disconnect on first data callback and then force reconnect...) or possibly ignore the rest of the message when `topic_len == 0`

Alvin1Zhang, 2019-08-23:
@jkockx Thanks for reporting the issue. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-03-03:
@jkockx Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details provided. Thanks.

closed 2020-03-03 by Alvin1Zhang