espressif/esp-idf #3869: https OTA fails

korstiaanS, 2019-08-02:
Version 3.2.2:

I created a working OTA from the examples.
It all worked fine for the first in the beginning but now I can't get it work again.
I first thought it was my Mac so installed a new ubuntu to act as SSL server.
But I see the same error on the ESP:

```
I (11926) 999_main: server json: https://192.168.56.233:8070/999.json
E (12339) esp-tls: mbedtls_ssl_setup returned -0x7f00

E (12347) esp-tls: Failed to open new connection
E (12350) TRANS_SSL: Failed to open a new connection
E (12353) HTTP_CLIENT: Connection failed, sock < 0
I (12358) 999_main: unable to download the json file, aborting...

```

It is not the CN.
The CURL tests work fine.

Are there some setting for this? (Stack, ...)
What causes this?

Please help, I'm on a time schedule but I'm stuck now.

Korstiaan

mahavirj, 2019-08-02:
@korstiaanS Seems like short on memory, `-0x7f00` refers to allocation failure. You may want to select config option https://github.com/espressif/esp-idf/blob/release/v3.2/components/mbedtls/Kconfig#L58 in case not done, it will help to reduce dynamic memory by 12K per TLS connection. Also please look out at other places to optimize on memory aspect in your application.

korstiaanS, 2019-08-02:
If I disable BLE it works.
If I enable BLE (and task) I get:
```
E (16677) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (16684) esp-tls: Certificate verified.
E (16692) esp-tls: Failed to open new connection
E (16695) TRANS_SSL: Failed to open a new connection
E (16697) HTTP_CLIENT: Connection failed, sock < 0

```

chegewara, 2019-08-02:
Like @mahavirj said, its most likely not enough heap to create secure connection.
What you can do is to move your initialize BLE code and do it when you are successfully connected with server (but, since its OTA server connection, this may be not very convenient solution).

korstiaanS, 2019-08-03:
It works, but sometimes it won't work.
These are my free stack sizes and heap:

```
I (1601407) 307_main: Stack IGN: 996
I (1601864) 307_gsm: Stack: 1696
I (1602693) 307_main: Stack: 692 - Heap: 55716
I (1604908) 307_netronix: Stack: 936
I (1606055) 307_ble: Stack: 912
I (1606486) 307_gps: Stack: 1396

```

There is enough, isn't?

korstiaanS, 2019-08-03:
Used SPIRAM now

closed 2019-08-03 by korstiaanS