espressif/esp-idf #3871: esp_partition_erase_range

Geka000, 2019-08-03:
#3401  Environment

- Development Kit:     none
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): v3.3-beta3-170-g91f29bef1
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0

- Operating System:     macOS
- Power Supply:         USB

## Problem Description

flash corrupts after esp_partition_erase_range with data partition large then 3Mb

### Expected Behavior

data partition erased

### Actual Behavior

factory partition and maybe bootloader corrupts

### Code to reproduce this issue

partition_table.csv
# Name,Type,SubType,Offset,Size,Flags
# Note: if you change the phy_init or app partition offset,make sure to change the offset in Kconfig.projbuild
nvs,data,nvs,,0x006000,
phy_init,data,phy,,0x001000,
factory,app,factory,,2M,
data,data,0x90,0x0500000,8M,

app_main.c
#include <stdio.h>
#include <esp_partition.h>
#include <string.h>

#define ERASE_SIZE 1024*1024

void app_main(){
	const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, 0x90, NULL);
	printf("partition %p, %d\n", (void*)(partition->address), partition->size);
        // same effect
	//ESP_ERROR_CHECK(esp_partition_erase_range(partition, 0, partition->size));

	size_t i=0;
	do {
		printf("erasing %d\n",i);
		ESP_ERROR_CHECK(esp_partition_erase_range(partition, i, ERASE_SIZE));
		printf("done %d\n",i);
		i += ERASE_SIZE;
	} while( i < partition->size);

	printf("done\n");
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

I (30) boot: ESP-IDF v3.3-beta1-522-g0f927791b 2nd stage bootloader
I (30) boot: compile time 00:48:23
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 16MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00200000
I (82) boot:  3 data             Unknown data     01 90 00500000 00800000
I (89) boot: End of partition table
...
I (176) boot: Loaded app from partition at offset 0x10000
...
I (209) cpu_start: ESP-IDF:          v3.3-beta1-522-g0f927791b
I (215) cpu_start: Starting app cpu, entry point is 0x40080ec0
...
I (0) cpu_start: Starting scheduler on APP CPU.
partition 0x500000, 8388608
erasing 0
done 0
erasing 1048576
done 1048576
erasing 2097152
E (10255) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10255) task_wdt:  - IDLE0 (CPU 0)
E (10255) task_wdt: Tasks currently running:
E (10255) task_wdt: CPU 0: main
E (10255) task_wdt: CPU 1: IDLE1
done 2097152
erasing 3145728
done 3145728
erasing 4194304
E (15145) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15145) task_wdt:  - IDLE0 (CPU 0)
E (15145) task_wdt: Tasks currently running:
E (15145) task_wdt: CPU 0: main
E (15145) task_wdt: CPU 1: IDLE1
done 4194304
erasing 5242880
done 5242880
erasing 6291456
E (20035) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20035) task_wdt:  - IDLE0 (CPU 0)
E (20035) task_wdt: Tasks currently running:
E (20035) task_wdt: CPU 0: main
E (20035) task_wdt: CPU 1: IDLE1
done 6291456
erasing 7340032
done 7340032
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d28a0: ffffffff ffffffff ffffffff
0x400d28a0: _puts_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/puts.c:101 (discriminator 2)

Core 0 register dump:
PC      : 0x400d28a4  PS      : 0x00060030  A0      : 0x800d0985  A1      : 0x3ffb4c70  
0x400d28a4: puts at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/puts.c:137

A2      : 0x00800000  A3      : 0x3ffb69b0  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x800d1cac  A9      : 0x3ffb4c20  
A10     : 0x3f40212c  A11     : 0x3ffb4da0  A12     : 0x00100000  A13     : 0x06ff9ff8  
A14     : 0x3ffaffd4  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff

Backtrace: 0x400d28a4:0x3ffb4c70 0x400d0982:0x3ffb4c90 0x4008464d:0x3ffb4cb0
0x400d28a4: puts at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/puts.c:137

0x400d0982: main_task at /Users/geka/SoftwareDevelop/Projects/esp32/esp-idf/components/esp32/cpu_start.c:530

0x4008464d: vPortTaskWrapper at /Users/geka/SoftwareDevelop/Projects/esp32/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets J]
]Q}�TC_RE�QU�����0x17 (*�}MQ}FLASH_�ꊕj
f+͡�rea*�ɱ ��0
et�Յ���c��1 
etsRչ� 8�016 ��22:5SH�

negativekelvin, 2019-08-03:
If it fails at >3MB and it starts at 0x300000 are you sure you don't have a 8MB flash chip? What is the output of `esptool.py flash_id`?  If you place your data partition at 0x300000 and make it 5MB long does it work?

Geka000, 2019-08-08:
it's my fault. Chip has only 4MB flash. But I was confused with debug output 
```
I (36) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 16MB
```

projectgus, 2019-08-13:
Hi @Geka000,

Thanks for updating. What do you have configured for ["Flash Size" and "Detect Flash Size when flashing"](https://docs.espressif.com/projects/esp-idf/en/v3.3-rc/api-reference/kconfig.html#config-esptoolpy-flashsize) under Flasher Settings in the project config?

If detection is off, the flash size will be shown as whatever is configured in the project regardless of the actual flash chip.

If detection is on, esptool.py should detect the flash size when flashing the bootloader. It's possible you have a flash chip where the detection algorithm doesn't apply. If you think this is happening, could you check the output when flashing and post it here, please? And also the output of `esptool.py -p PORT flash_id`?

Alvin1Zhang, 2019-08-23:
@Geka000 Thanks for reporting the issue, would you please help provide any updates or outputs suggested by @projectgus ? Thanks.

Alvin1Zhang, 2020-01-15:
Thanks for reporting the issue, would you please help provide any updates or outputs suggested by @projectgus ? Thanks.

closed 2020-04-30 by projectgus