espressif/esp-idf #3875: BLE project compile fails with cmake

lachlanleslie1995, 2019-08-05:
Currently swapping a project from original make system to the new cmake build system for esp-idf.      
 I have no issue when compiling and flashing the project with make but when I try to use idf.py I get a LoadProhibited guru error.

[dump.txt](https://github.com/espressif/esp-idf/files/3465440/dump.txt)

From my understanding the error seems to come from whenever I call any of bluetooth library functions like esp_bt_controller_mem_release or esp_bt_controller_init

gengyuchao, 2019-08-08:
Can you provide 'elf' file with the log to help me identify the problem? thank you!

lachlanleslie1995, 2019-08-11:
Hey @gengyuchao sorry for late reply here is the BIN and log:
[Log.txt](https://github.com/espressif/esp-idf/files/3490565/Log.txt)

[SourceCode.zip](https://github.com/espressif/esp-idf/files/3490561/SourceCode.zip)

lachlanleslie1995, 2019-08-19:
@igrr @projectgus Any thoughts?

renzbagaporo, 2019-08-19:
Just to clarify: Both Make and CMake compiles the project successfully. The problem is that the binaries compiled with Make works while the binaries compiled with CMake does not (with the LoadProhibited panic)?

Is is possible for you to set the debug level to debug, rebuild and upload the logs and the new `.elf` again? Perhaps even the `.map` file.

lachlanleslie1995, 2019-08-19:
Correct @renzbagaporo both compile without issue, but on running them the cmake binaries gives a LoadProhibited panic. I'll attach the .map and .elf below.
[build.zip](https://github.com/espressif/esp-idf/files/3518073/build.zip)

renzbagaporo, 2019-08-20:
Hi @lachlanleslie1995, in your sdkconfig what is the value of CONFIG_BT_RESERVE_DRAM?

lachlanleslie1995, 2019-08-20:
Here is my sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3518777/sdkconfig.zip)

renzbagaporo, 2019-08-20:
The config seems to be correct. That makes this issue really weird to me.

Some more clarification: You mention that the issue comes up whenever you call function from BLE library, yet the logs indicate the issue occurs on startup, particularly failing on this assertion:

```
assert(reserved[i+1].start > reserved[i].start);
```

I just want to verify that we are referring to the same issue.

Investigating the failed assertion mentioned, the issue seems to be the contents of reserved regions (from the map file it is from 0x3f41b288 to 0x3f41b2d0)

Taking a peek at the `elf` file at these addresses:

```
 3f41b280 bcf21240 e0f21240 e0e6fa3f 10fffa3f  ...@...@...?...?
 3f41b290 28dbfb3f 5cdbfb3f 0080fb3f 209afb3f  (..?\..?...? ..?
 3f41b2a0 0000fb3f 8863fb3f 00e0fa3f e0e6fa3f  ...?.c.?...?...?
 3f41b2b0 203ffe3f 5043fe3f 0000fe3f 4004fe3f   ?.?PC.?...?@..?
 3f41b2c0 00800740 00000840 00000740 00800740  ...@...@...@...@
```

This region contains data of type `soc_reserved_region_t`, which has a start and end address members. The contents of this region, plus `{&_data_start, &_static_data_end}` and `{&_iram_start, &_iram_end}` is what's checked during startup (see function `s_prepare_reserved_regions`)

The issue seems to be that `&_data_start` and the start address at `0x3f41b2a0` in the memory region is the same, 0x3ffb0000. From the `map` file:

```
.dram0.data     0x3ffb0000     0x2958
                0x3ffb0000                _data_start = ABSOLUTE (.)
                0x3ffb0000                _bt_data_start = ABSOLUTE (.)
```

Ideally, `_data_start` should be `0x3ffb0000 + CONFIG_BT_RESERVE_DRAM` so that the checks would pass. The case where these values would be the same is if CONFIG_BT_RESERVE_DRAM = 0. But your sdkconfig says otherwise.

Just to be sure, can you also upload a file called `esp32_out.ld`, which should be in the project's build folder. Also, may I get project_description.json?

lachlanleslie1995, 2019-08-20:
Hey @renzbagaporo,
I have changed my code slightly since I originally created this issue as I've been continuing to develop with the old make system while this issue gets resolved but originally if I didn't call any BLE functions inside my btEnable func the system would run fine. This no longer seems to be the issue instead it's whenever I call my basic init function even with every other call inside it commented out it still seems to fail.
I have two esp32_out.ld's in my build folder one from esp-idf/esp32 folder inside:
[esp32_out.zip](https://github.com/espressif/esp-idf/files/3522963/esp32_out.zip)|
and another from just esp32 folder:
[esp32_out.zip](https://github.com/espressif/esp-idf/files/3522965/esp32_out.zip)

Here is the project_description
[project_description.zip](https://github.com/espressif/esp-idf/files/3522969/project_description.zip)

renzbagaporo, 2019-08-21:
Ah, the esp-idf/esp32 esp32_out.ld (used by CMake build system) exhibits the problem that I mentioned:

```
  dram0_0_seg (RW) : org = 0x3FFB0000 + 0x0 ,
                                     len = 0x2c200 - 0x0
```

`0x0` is supposed to be CONFIG_BT_RESERVE_DRAM (0xdb5c based on you sdkconfig), but somehow the value is not being used here. 

The presence of two esp32_out.ld suggest you are using the two build systems at the same time, which could account for the issue. Have you tried clearing the build folder and sdkconfig before building with CMake?

lachlanleslie1995, 2019-08-21:
Ahh! That's done it, thanks a bunch @renzbagaporo . Must've not converted from the make to cmake system properly.

closed 2019-08-21 by lachlanleslie1995