espressif/esp-idf #3879: idf.py not recognized after esp-idf toolchain reinstall

viviennejk, 2019-08-05:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v3.2.2
- Build System:  Could be Make or CMake, I don't know
- Compiler version: UNKNOWN (received error " 'xtensa-esp32-elf-gcc' is not recognized as an internal or external command, operable program or batch file.")
- Operating System: Windows
- Power Supply: USB

## Problem Description

I successfully installed the esp-idf tool from the [Getting Started page](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-step-by-step). I then kept running into issues flashing my board, and received the message: "Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7." I tried to change the Python interpreter to Python 2.7 but had no luck (there was limited documentation on it on the ESP-IDF docs). I finally decided to uninstall and delele the esp-idf toolchain and start from scratch but select Python 2.7 instead of 3.7 during the install. I regret it now because I can't even get idf.py to work.

After installing everything again and getting to Step 7 Configure, I run "idf.py menuconfig" and get the following error while in the hello-world project path:

_'idf.py' is not recognized as an internal or external command,
operable program or batch file._

I check the \tools directory and idf.py is indeed there. Tried to copy it over to the \hello-world repository and then got a new error: 

_"Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
'cmake' must be available on the PATH to use idf.py"_

(by this point I'd uninstalled and reinstalled it so many times that I'm back to Python 3.7)

### Expected Behavior

I expected the menuconfig window to open just as described in the Getting Started guide, and as I had been able to do successfully before doing the uninstall-reinstall.

### Actual Behavior

See error messages above. 

## Debug Logs

C:\Users\VivienneJK\Desktop\esp-idf>idf.py menuconfig
'idf.py' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\VivienneJK\Desktop\esp-idf>cd %userprofile%\esp
The system cannot find the path specified.

C:\Users\VivienneJK\Desktop\esp-idf>xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\CMakeLists.txt
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\Makefile
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\README.md
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\main\CMakeLists.txt
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\main\component.mk
C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world\main\hello_world_main.c
6 File(s) copied

C:\Users\VivienneJK\Desktop\esp-idf>idf.py menuconfig
'idf.py' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\VivienneJK\Desktop\esp-idf>cd C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world

C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world>cd C:\Users\VivienneJK\Desktop\esp-idf\hello_world

C:\Users\VivienneJK\Desktop\esp-idf\hello_world>idf.py menuconfig
'idf.py' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\VivienneJK\Desktop\esp-idf\hello_world>cd C:\Users\VivienneJK\Desktop\esp-idf\tools

C:\Users\VivienneJK\Desktop\esp-idf\tools>idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
'cmake' must be available on the PATH to use idf.py

C:\Users\VivienneJK\Desktop\esp-idf\tools>idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
'cmake' must be available on the PATH to use idf.py

C:\Users\VivienneJK\Desktop\esp-idf\tools>C:\Python27\python idf.py
The system cannot find the path specified.

C:\Users\VivienneJK\Desktop\esp-idf\tools>git describe --tags
v3.2.2

C:\Users\VivienneJK\Desktop\esp-idf\tools>xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\VivienneJK\Desktop\esp-idf\tools>cd C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world

C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world>idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
'cmake' must be available on the PATH to use idf.py

C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world>

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

viviennejk, 2019-08-05:
Update: I followed [these instructions](https://docs.espressif.com/projects/esp-idf/en/v3.2.2/get-started-cmake/add-idf_path-to-profile.html) to try to update my PATH to include idf.py. I followed the steps in "Edit Environment Variables".

Now getting this error: C:\Users\VivienneJK\Desktop\esp-idf\examples\get-started\hello_world>idf.py menuconfig
'cmake' must be available on the PATH to use idf.py

negativekelvin, 2019-08-05:
You may have noticed the guides and installers and toolchains are different for latest=master=v4.0 and stable=v3.x

If you run the latest installer it creates a shortcut to open the terminal which sets up the paths

If you run the stable installer it should set up the cmake/toolchain paths but you have to set up the tools and idf paths manually

If you aren't sticking to one version of the guide your results may vary

viviennejk, 2019-08-05:
Yes I realized that belatedly. I wanted to stick to the latest installer because I had that working, but now I don't know what I did differently during the reinstall. I thought I'd deleted everything and ran the exe from scratch.

I've since tried using the stable version (MSYS2) and am working through the instructions for that.

Still would like to know how to resolve this issue from the latest version.

negativekelvin, 2019-08-05:
Well msys2 is a third version so ... fun x3

You still get the cmake path error when you use the start menu shortcut?

Also FYI the latest will install gcc8 toolchain whereas stable will install gcc5 and stable branches will expect to see gcc5

dobairoland, 2019-08-06:
Hi @viviennejk! I'm sorry for the inconvenience. Here are my recommendations:
- Python 3 support was added in v3.2. That is the reason why you are receiving those warnings. v3.2 should work without issues with Python 3. Please use that one because that will be the only supported one in the future. Python 2 will be deprecated. Note: none of the failures you've received are related to Python incompatibility. Those are usually some kind of multi-line exceptions.
- `IDF_PATH\tools` needs to be in the path because Windows needs to know where to find `idf.py`. I see that you've resolved this issue. A temporary and inconvenient fix could be also to run `python c:\full\path\to\idf.py`.
- The MSYS2 environment was the supported one in the v3.x line and running tools from Windows command line an experimental environment. The setup of Win command line is much more user friendly in v4.0. But you should be able to setup for v3.2 as well. I recommend you to do so because this will be supported in the future, so you will not have to migrate.
- You need to setup CMake to be accessible from command line. If you run `cmake --version` then it should print the version and no error message. My guess is that you didn't install it. You can download a Windows installer from here: https://cmake.org/download/ 

Please let us know if you resolved the issue or need further assistance.

dobairoland, 2019-08-13:
Hi @viviennejk. Do you have any update regarding your issue?

viviennejk, 2019-08-22:
Hi @dobairoland thank you for your recommendations. I haven't attempted to use the esp-idf toolchain since posting this because I just found it easier to port over code from Arduino ESP32 libraries and use that platform (I'm not a firmware developer). Will update if I attempt to resolve this issue again. For now, it is unresolved/opened.

dobairoland, 2019-08-23:
Hi @viviennejk. Thanks for the update. I'm closing this issue since this is related to your personal configuration of development environment and I don't see what else could or should be done regarding this for now.

Please fell free to reopen this issue if you will experience these problems again and we will help you.

closed 2019-08-23 by dobairoland