espressif/esp-idf #3880: BT_GATT: No resources

kianwee123, 2019-08-06:
when i configure the blufi,set OPMODE,STA_SSID,STA_PASSWD,in normal circumstances it will run to " example_event_callback()  case ESP_BLUFI_EVENT_SET_WIFI_OPMODE:",but sometime screen print
"E (47963082) BT_GATT: No resources
E (47963772) BT_BTC: btc_blufi_recv_handler checksum error 0e28, pkt b9d2
E (47963772) BLE: BLUFI report error, error code 1"
or
"
E (3455462) BT_GATT: No resources
E (3455522) BLE: blufi_dh_negotiate_data_handler make public failed -12944
E (3455522) BLE: BLUFI report error, error code 8
"
i can't see the underlying code
waiting for your answer ,thanks

Alvin1Zhang, 2019-08-06:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

kianwee123, 2019-08-07:
> Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

return error state is ESP_BLUFI_CHECKSUM_ERROR  or ESP_BLUFI_MAKE_PUBLIC_ERROR,

GYC-Espressif, 2019-12-07:
There is an error in the packet sent to bluefi. Which app do you use, provided by us or written by yourself? Blufi is open source, you can see the whole content.

Alvin1Zhang, 2020-02-27:
@kianwee123 Thanks for reporting. Will close the issue due to short of feedback, feel free to reopen if you could help provide more details suggested by @GYC-Espressif . Thanks.

closed 2020-02-27 by Alvin1Zhang