espressif/esp-idf #3883: Can't read additional input registers using modbus_master example

bonastreyair, 2019-08-06:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.0-dev-1381-g6fe853a2c
- Build System: Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: macOS
- Power Supply: USB

## Problem Description
I've adapted the modbus_master example and I'm having trouble with the addresses. Maybe it is because I don't know much how it really works, but the only way I could find to read any address I want is to set the "Reg Start" in the `const mb_parameter_descriptor_t device_parameters[]` in the `device_params.c` file to the position of the real Modbus address.

Let's say I want to read coil address 00001:
```
{CID_DATA_COIL0, STR("Coil1.0-RW"), STR("Bool"), MB_DEVICE_ADDR1, MB_PARAM_COIL, 0, 1, COIL_OFFSET(coil_byte0), PARAM_TYPE_U16, 2, OPTS(BIT0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER}
```
Or if I want to read coil address 00009:
``` 
{CID_DATA_COIL0, STR("Coil1.0-RW"), STR("Bool"), MB_DEVICE_ADDR1, MB_PARAM_COIL, 9, 1, COIL_OFFSET(coil_byte0) , PARAM_TYPE_U16, 2, OPTS(BIT1, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
``` 
Firstly, I don't understand the `COIL_OFFSET(coil_byte0)` since even if I change the variable to `coil_byte1`, the result is exactly the same...
At least this is for now, the way I can access any desired modbus address.

The problem I have is that this way works OK for all the modbus addresses types except for the input registers type. If I set the address to "10" (for example):
```
{CID_DATA_INPUT3, STR("Input3-R"), STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 10, 2, INPUT_OFFSET(input_register0), PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
```
Then I get the reading of the address of the first holding register in the memory. How is that even possible?

Apart from that, I still don't understand the usage of `OPTS(0,0,0)`, or the `XXXX_OFFSET()`, it seems even if I change any of its values, there is no effect on the final reading or writing.

### Expected Behavior
Be able to access any address by its number without trouble.
- Coils: 1-9999 
  --> in code using `MB_PARAM_COIL` and Reg start  (0-9999)
- Discrete Input: 10001-19999 
  --> in code using `MB_PARAM_DISCRETE` and Reg start (0-9999)
- Input Register: 30001-39999 
  --> in code using `MB_PARAM_INPUT` and  Reg start (0-9998)
- Holding Register: 40001-49999 
  -->in code using `MB_PARAM_HOLDING` and Reg start (0-9998)

### Actual Behaviour
Input Register only let me to access addresses 0-8 but not 10 or further. 

### Steps to reproduce
1. Use the modbus_master + modbus_slave examples simultaneously with two connected ESP32 with MAX485 via MODBUS
2. Set more than 5 input_registers in the slave
3. Set some holding registers
4. Access the 6th input_register from the modbus_master to the modbus_slave
5. See that you are reading the information from the 1st holding register instead of the 6th input_register.

### Code to reproduce this issue
Configuration on the modbus_master:
```
// Input Register Configuration
{CID_DATA_INPUT0  , STR("Input0-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   , 0, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT1  , STR("Input1-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   , 2, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT2  , STR("Input2-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   , 4, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT3  , STR("Input3-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   , 6, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT4  , STR("Input4-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   , 8, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT4  , STR("Input5-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   ,10, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_INPUT4  , STR("Input6-R")   , STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT   ,12, 2, INPUT_OFFSET(input_register0)     , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},

// Holding Register Configuration
{CID_DATA_HOLDING0, STR("Holding0-RW"), STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING , 0, 2, HOLD_OFFSET(holding_register0)    , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_HOLDING1, STR("Holding1-RW"), STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING , 2, 2, HOLD_OFFSET(holding_register0)    , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},
{CID_DATA_HOLDING2, STR("Holding2-RW"), STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_HOLDING , 4, 2, HOLD_OFFSET(holding_register0)    , PARAM_TYPE_FLOAT, 4, OPTS(   0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER},

Result:
I (1560) Sense-Modbus: cid: 16, Input0-R(Float) = 1.31
I (1570) Sense-Modbus: cid: 17, Input1-R(Float) = 2.51
I (1590) Sense-Modbus: cid: 18, Input2-R(Float) = 3.94
I (1590) Sense-Modbus: cid: 18, Input3-R(Float) = 4.95
I (1590) Sense-Modbus: cid: 18, Input4-R(Float) = 5.96
I (1610) Sense-Modbus: cid: 19, Input5-R(Float) = 213.00  // Same as Holding0!!
I (1620) Sense-Modbus: cid: 20, Input6-R(Float) = 126.51  // Same as Holding1!!
I (1640) Sense-Modbus: cid: 21, Holding0-RW(Float) = 213.00
I (1650) Sense-Modbus: cid: 22, Holding1-RW(Float) = 126.51
I (1670) Sense-Modbus: cid: 23, Holding2-RW(Float) = 127.72
```

Configuration on the modbus_slave:
```
// Input Register Configuration
input_reg_params.input_register0 = 1.31;   // Info on addresses 30001-30002
input_reg_params.input_register1 = 2.51;   // Info on addresses 30003-30004
input_reg_params.input_register2 = 3.94;   // Info on addresses 30005-30006
input_reg_params.input_register3 = 4.95;   // Info on addresses 30007-30008
input_reg_params.input_register4 = 5.96;   // Info on addresses 30009-30010
input_reg_params.input_register5 = 6.97;   // Info on addresses 30011-30012 
input_reg_params.input_register6 = 7.98;   // Info on addresses 30013-30014

// Holding Register Values
holding_reg_params.holding_register0 = 213.00;   // Info on addresses 400001-400002
holding_reg_params.holding_register1 = 126.51;   // Info on addresses 400003-400004
holding_reg_params.holding_register2 = 127.72;   // Info on addresses 400005-400006
```

## Debug Logs
No debug logs other than the result shown.

alisitsyn, 2019-08-08:
Hi @bonastreyair,

The ESP-IDF modbus master implementation contains abstraction layer above modbus to work with the characteristics instead of modbus parameters. Each characteristic corresponds to physical value. 
This is required to integrate modbus protocol for example with MESH networks. Part of this layer is data dictionary which is located in device_parameters.h/c. Characteristic ID is the first member of the structure.
In your adaptation of example you made several errors and this is why it does not work in your case.

Please follow the example documentation file readme.md in the example folder. 
1. Your CID is incorrect for input register 5:
```{CID_DATA_INPUT4, STR("Input5-R"), STR("Float"), MB_DEVICE_ADDR1, MB_PARAM_INPUT, 10, 2, INPUT_OFFSET(input_register0), PARAM_TYPE_FLOAT, 4, OPTS(0, 0, 0), PAR_PERMS_READ_WRITE_TRIGGER}, ```
It causes incorrect reading of the parameter.

2. ```The XXXX_OFFSET()``` macro is required if you want to read your caracteristic into specific instance. 
It calculates the offset to correspond field in the appropriate parameter strucure and it will be used to store actual value of the characteristic by modbus stack.
In your case all parameters are stored in the same field of the structure. If you don't want to keep parameter in your specific structure put offset =0 for CID then the instance to store parameter will be created automatically in ```sense_modbus_get_param_data()```.

The ```OPTS(opt1, opt2, opt3)``` are used as optional parameters during processing of field characteristic. For example these can be High/low limits, masks for bit fields and other options which correspond to actual physical characteristic.
This values will be applied to field value to trigger some event send to the network (if value is out of limit it will send alarm event to the network).

The trigger_operation_task() is the main processing task. It uses the characteristic table created by sense_modbus::sense_modbus_get_characteristics() and perform actual user control process. 
This is the place where you can place your processing functionality and use your options.

--
Alex

bonastreyair, 2019-08-15:
Wow so stupid of me on repeting CID_DATA_INPUT4.... sorry.
Thanks for your explanations they really helped, I couldn't find that info on the readme or any documentation. 
Are you saying that there is modbus master with mesh integrated? I can only use this example in IDFv4.0 apart from MDF, since MDF has IDFv3.2... the modbus_master does not work yet... right?

I'm closing the issue, but I would really wan't to know about IDFv4.0 on MDF...
Thanks!

closed 2019-08-15 by bonastreyair

alisitsyn, 2019-08-26:
There is preliminary Modbus-MASH project which worked well with old version of MDF. It is obsolete now.

--

Alex