espressif/esp-idf #3884: idf_monitor.py huge CPU load

no1seman, 2019-08-06:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-dev-1441-g6517c5033
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery
## Problem Description
After last esp-idf repo updates idf_monitor.py began to 100% load of one CPU core under linux (uname -a: Linux ubuntudev 4.15.0-55-generic #60-Ubuntu SMP Tue Jul 2 18:22:20 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux, ubuntu 18.04)  
### Expected Behavior
idf_monitor.py should not use so much CPU resources
### Actual Behavior
Near 100% CPU load
### Steps to repropduce

### Code to reproduce this issue

## Debug Logs

user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.1**  0.1 498104 14528 pts/0    Rl+  15:01  15:20 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8598  0.0  0.0  21536  1044 pts/1    S+   15:16   0:00 grep --color=auto python
user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.1**  0.1 498104 14528 pts/0    Rl+  15:01  15:23 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8600  0.0  0.0  21536  1044 pts/1    R+   15:16   0:00 grep --color=auto python
user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.1**  0.1 498104 14528 pts/0    Rl+  15:01  15:24 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8602  0.0  0.0  21536  1040 pts/1    S+   15:16   0:00 grep --color=auto python
user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.2**  0.1 498104 14528 pts/0    Rl+  15:01  15:25 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8604  0.0  0.0  21536  1008 pts/1    S+   15:16   0:00 grep --color=auto python
user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.2**  0.1 498104 14528 pts/0    Rl+  15:01  15:26 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8606  0.0  0.0  21536  1088 pts/1    R+   15:16   0:00 grep --color=auto python
user@ubuntudev:~$ ps -aux | grep python
root       963  0.0  0.2 177688 17740 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/bin/networkd-dispatcher --run-startup-triggers
root      1416  0.0  0.2 194396 20448 ?        Ssl  aug05   0:00 /usr/bin/python3 /usr/share/unattended-upgrades/unattended-upgrade-shutdown --wait-for-signal
user      8308 **99.1**  0.1 498104 14528 pts/0    Rl+  15:01  15:27 python /home/user/esp/esp-mdf/esp-idf/tools/idf_monitor.py --port /dev/ttyUSB0 --baud 115200 ./build/beacon.elf
user      8608  0.0  0.0  21536  1000 pts/1    S+   15:16   0:00 grep --color=auto python

negativekelvin, 2019-08-06:
https://github.com/espressif/esp-idf/commit/ac5508efd54c1f70d62af7e7a9f77ef2ba750f19

pglen, 2019-08-14:
I have the same issue on Linux Ubuntu. Appeared after update to IDF 4

ginkgm, 2019-08-15:
@no1seman @negativekelvin @pglen 

Sorry about this. It's due to the delay time too short here: https://github.com/espressif/esp-idf/blob/master/tools/idf_monitor.py#L493

Is there anyone can help me to test which value is proper on your OS? (Too large will cause the screen print delay/keyboarad respond delay, too small will use too much CPU resource) The unit is in second.

no1seman, 2019-08-15:
@ginkgm,
thank you for watching on the issue.
seems changing delay in: 'item = self.event_queue.get(False, 0.001)' to  'item = self.event_queue.get(False, 1.0)' doesn't help much:
[image]

negativekelvin, 2019-08-15:
It should be `item = self.event_queue.get(True, 0.001)`

no1seman, 2019-08-15:
with item = self.event_queue.get(True, 0.01) logs printing looks good with #define SOCKETS_DEBUG                   LWIP_DBG_ON and intensive network operations: 4-5% CPU core load.

pglen, 2019-08-15:
FIXED!

 The processor was not breathing in the condition of no data. I inserted a
delay into the flow, so now the processor has a chance to relax.
 See the line I added, marked in red in the code snippet below. I elected
30 milliseconds, as it is an optimum  between as much delay as we can add,
and perception
threshold of the human observer.

Snippet from idf_monitor.py:

def main_loop(self):
        self.console_reader.start()
        self.serial_reader.start()
        try:
            while self.console_reader.alive and self.serial_reader.alive:
                try:
                    item = self.cmd_queue.get_nowait()

                except queue.Empty:
                    try:
                        item = self.event_queue.get(False, 0.001)
                    except queue.Empty:
                        # Let the processor breathe # Peter Glen 15/aug/2019
                        time.sleep(0.03)   # Wait 30 millisec on no data
                        continue

                (event_tag, data) = item
                if event_tag == TAG_CMD:
                    self.handle_commands(data)
                elif event_tag == TAG_KEY:

 An even simpler change is to to remove the second .get, and add just the
time.sleep line.

*Please mention my name in the same spirit that prompted me to help.*

Peter Glen

dobairoland, 2019-08-15:
@pglen We usually don't put names into comments but you can open a Pull request for this and then your contribution will be logged in the form of a commit (including your name and email address).

pglen, 2019-08-15:
No worries. Glad I could help.

negativekelvin, 2019-08-15:
@pglen there are two different queues so you don't want to remove the "second get".  It is trying to block but uses the incorrect False parameter so the correct fix is changing it to True as shown above.  Blocking is better than forcing a delay with sleep.

pglen, 2019-08-15:
The blocking is better, yes, but the serial implementation does not support
what you propose.  (provides no event/semaphore) to block on / wait for.

The reason why the processor is clipping, is that the no data path repeats
the call too soon.

Waiting for 30 ms on no data is an option; killing the processor is not.

Peter

pglen, 2019-08-15:
Tried the blocking option, you are right.

When will this make it into the github subsys?

Thank you.

ginkgm, 2019-08-16:
@pglen it's under internal review now, I think you can see it soon~

Thanks for all your contribution

closed 2019-08-21 by igrr