espressif/esp-idf #3885: uart_get_baudrate return 115201 when is set at 115200

caveman1996, 2019-08-06:
## Environnement: MSYS32 on a Windows VM machine

- Module or chip used:  ESP32-WROOM-32D
- Build System:  Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-75-gbaf03c2c) 5.2.0

- Operating System:     Windows
- Power Supply:         Custom PCB

## Problem Description

I have an application that need to know exactly on which baudrate it is to return a predifined ENUM. The baudrate needs to be exact because it gets converted in a switch case. 

Right now, uart_get_baudrate() return a non-exact baudrate of 115201 when it is set to 115200 (hard coded!). The result is printed with ESP_LOGD() function, thought it ain't the problem for the simple fact that the code crash on the switch case which doesn't find the equivalent baudrate (no [case] for 115201).

### Expected Behavior
uart_get_baudrate() should return 115200, not 115201.

### Actual Behavior
Returns 115201

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    uart_get_baudrate(uart_num, &baudrate);
   ESP_LOGD(TAG_INITOBD, "Baudrate changed to: %d", baudrate);
}

```

## Debug Logs
D (1965) initLogiqueOBD: Baudrate changed to: 115201[1B][0m

negativekelvin, 2019-08-06:
Baudrate is not 100% accurate.  It is going to find the closest match it can, in this case 15201.152hz.  It's a bad idea to arbitrarily remove precision, so your switch statement should turn into an if statement

```
if(baudrate > 114048 && baudrate < 116352) //within 1% of 115200
    return B115200;
else if ....
```

tim-nordell-nimbelink, 2019-08-06:
@negativekelvin - I agree that it's not 100% accurate.  I could see someone wanting to find out the requested baud rate, and separately finding out the actual baud rate.  (The current code returns the actual baud rate, and @caveman1996 is looking for the requested baud rate.)

caveman1996, 2019-08-06:
> Baudrate is not 100% accurate. It is going to find the closest match it can, in this case 15201.152hz. It's a bad idea to arbitrarily remove precision, so your switch statement should turn into an [...]

That's unfortunate... I would have hope to get the requested baudrate passed in the configuration structure (as pointed out by @tim-nordell-nimbelink). Then I'll close this thread. Thanks for the answers!

closed 2019-08-06 by caveman1996