espressif/esp-idf #3886: How should I know the flash size have been fully used or not?

huazhenjiang, 2019-08-07:
## Environment

Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): v2
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-1287-gd7e659d
Build System: CMake
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Windows 10
Power Supply: USB

## Problem Description
I composed of several examples and check the imaga size with idf.py size as the following:

Total sizes:
 DRAM .data size:   13040 bytes
 DRAM .bss  size:   25000 bytes
Used static DRAM:   38040 bytes ( 142696 available, 21.0% used)
Used static IRAM:   69953 bytes (  61119 available, 53.4% used)
      Flash code:  693997 bytes
    Flash rodata:  160224 bytes
Total image size:~ 937214 bytes (.bin may be padded larger)

The message show that the flash size seem have been fully used  if I download this image into the factory partition. 
Am I right? 
here is my partition table.

# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        1M,
ota_0,    app,  ota_0,   ,        1M,
ota_1,    app,  ota_1,   ,        1M,
storage,  data, spiffs,  ,        0xD0000,

I am worry about the flash usage because I have not done my all work.
I do not know the meaning about  "Flash code", "Flash rodata", "Total image size" such like where did them be put in flash or how to calculate the summation of the "Total image size".

which document should I survey for solving the this?

costaud, 2019-08-07:
```
Total image size:~ 937214 bytes
```

This is the image size that you should pay attention to. 
Since all the sections will be bundled into a binary image, and you will have to burn the whole .bin file into the target partition in the flash.

huazhenjiang, 2019-08-07:
Hi @costaud ,

Thank you so much for the speedy reply.
Do you mean the "total size" was the final size that be programmed into 'factory partition'? (maybe larger because (.bin may be padded larger) )

Is there a method to reduce the code size ?
currently, i composed of the following components,
1. softAP provision
2. simple webserver
3. uart echo
4. https ota
5. aws iot with mqtt sub/pub
6. spiffs

negativekelvin, 2019-08-07:
You have release build compiler option enabled?  Do you really need factory partition? You can just have 2 ota partition and make them bigger.

huazhenjiang, 2019-08-07:
Hi @negativekelvin ,

Thank you so much for the speedy reply.
Actually, I do not know how to set the "release build compiler option". Do it invoke at "menuconfig"?
Is the factory partition not necessary ?  hmm... I think I have another question.
Does the source code which be programmed into flash  run directly on flash? or it would be loaded to such like SRAM/DRAM ? 
This is related that if I extend the partition over 1 MB,  the over 1MB code will run successful or not.

closed 2019-09-19 by huazhenjiang