espressif/esp-idf #3887: API or EFUSE to discover if module is WROOM or WROVER?

0neblock, 2019-08-07:
### Brief Description

Is there a recommended API that I can call from code to determine if the code is currently executing on a WROOM module or a WROVER module.

I am looking for a way to do this **without**having to attempt to initialise external PSRAM, as my custom board that uses WROOM has pins 16 and 17 in use, which are not compatible with PSRAM. Are there any EFuses burnt from the factory to determine what module the ESP32 IC is within?

My use case is I have a custom board, which can fit either a WROOM module or a WROVER module. I have many of these that are in the field with an application that can perform an OTA on itself itself automatically via the internet. 
It will send it's currently running firmware version to my server, which responds with a message indicating if there is an update available, with a link to download the firmware, which i then copy to esp_https_ota();

On boards that use a WROVER module, i have re-routed the traces that normally go to pins 16 and 17, to two new pins, so that I can initialise and use PSRAM, and these boards perform the same procedure as above, but download a different 'branch' of the firmware, that contains code specific to the WROVER module. The WROVER version of the firmware has the added functionality of using BLE and storing some Static data in BSS section.

However, I would like all boards to go out with a standard firmware, and the ones with the WROVER module have an option to download the 'Enhanced' firmware version.

Is there a way to do this using only the module ? Or should i be modifying the board using the WROVER modules in another way, such as with a short on a spare IO pin, so I can read that value from code?

Thanks, please let me know if this information is not clear enough.

negativekelvin, 2019-08-07:
If it is wrover and not wrover-b it will be bootstrapped for 1.8v so check GPIO_STRAP_REG.  You can set your own values in efuse during factory programming.

0neblock, 2019-09-03:
I set my own values for programming, thank you.

closed 2019-09-03 by 0neblock