espressif/esp-idf #3888: Update build from scratch docs to use esp32-2019r1-rc2

volodymyrlut, 2019-08-07:
I'm trying to build a MacOS toolchain from scratch. Documentation says that after clonning crosstool-NG I will need to checkout to git `esp32-2019r1`; however, this tag *does not exist*.
Instead `esp32-2019r1-rc2` should be listed in documentation.

I would be glad to update the doc, but I don't understand how to do it, so I'm leaving an issue here.

Thank you.

krzychb, 2019-08-08:
Hi @volodymyrlut, good point, thank you for reporting this issue. I will update documentation.

krzychb, 2019-08-08:
Done! We have ended up fixing the tag :) 
[esp32-2019r1](https://github.com/espressif/crosstool-NG/releases/tag/esp32-2019r1)

closed 2019-08-08 by krzychb

volodymyrlut, 2019-08-08:
@krzychb thank you!