espressif/esp-idf #3894: AP mode, got error wifi: ampdu: ignore deleting tx BA0

playGit888, 2019-08-07:
## Environment

- Development Kit:      [ESP32-PICO-D4]
- Kit version: self made
- Module or chip used:  [ESP32-PICO-D4|ESP32]
- IDF version:    v4.0-dev-1346-g1b6010b
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux-ubuntu16.04]
- Power Supply:         [Battery->dc-dc->3.3v]

## Problem Description
I got losts of error like this:
I (614270) wifi: ampdu: ignore deleting tx BA0
I (614270) wifi: ampdu: ignore deleting tx BA0
I (614270) wifi: ampdu: ignore deleting tx BA0
I (615270) wifi: ampdu: ignore deleting tx BA0

Hello, I am using a PICO D4, and trying the example of AP mode.
So far, I don't have the problem in sta mode.
When this error comes up, the data transfering is impossible.
This problem is random. Most of the time, ESP32 works well.

Someone else have the same problem.
https://www.esp32.com/viewtopic.php?f=13&t=10839&p=44403&hilit=wifi%3A+ampdu%3A+ignore+deleting+tx+BA0&sid=4bb8f22bbe5efb2d3911586f0131c3e6#p44403

https://www.esp32.com/viewtopic.php?f=25&t=10873&p=44538&hilit=wifi%3A+ampdu%3A+ignore+deleting+tx+BA0&sid=4bb8f22bbe5efb2d3911586f0131c3e6#p44538

re-produce the error:
1. example: softap
2. combine another example http server
3. make some "post" or "get" data transfer, or just reflesh the home page you set for http server , ESP32 should work well. 
Wait a few mintues, the error may come up.

Thank you, any suggestion will be appreciated

zhangyanjiaoesp, 2019-08-08:
@playGit888 can you provide  capture packages for further analysis?

playGit888, 2019-08-08:
I would love to, but I don't know how to capture packages.  Would you give me a hint?

playGit888, 2019-08-08:
[image]
[http_server-simple-softAP.tar.gz](https://github.com/espressif/esp-idf/files/3482617/http_server-simple-softAP.tar.gz)

Here is the sample code.
type http://192.168.1.1/hello to access hello page.
wait a few mintues....the error will occur.

By the way, I am using a computer to connect to softAP. 
Is it my computer to cause the problem?

mill1000, 2019-09-09:
Also seeing this issue in esp-idf v3.1.5
Client is an iPad Mini 4 running iOS 12.4 (16G77)

Reconnecting the client (iPad) stops the log message from appearing.

daniol, 2019-09-17:
The current implementation of AMPDU in the ESP32 seems to be wrong as it causes problems with some network adapters. Using UDP over AP causes massive packet drops and TCP over AP causes packet looses and socket timeouts.

I solved this issue disabling AMPDU on the sdkconfig.h as a workaround:

```
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 0
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 0
```

ppisljar, 2019-10-25:
i get the same trying to use esp-who recognition example on esp-who development board.

zhangyanjiaoesp, 2019-11-06:
We are debugging the `delete tx BA` issue recently , and we will solve it as soon as possible.

liuzfesp, 2019-12-02:
@gabrielrbeiro the fix now is in reviewing status, it will be merged into IDF this month.

cristiansedano, 2019-12-10:
> 
> 
> @gabrielrbeiro the fix now is in reviewing status, it will be merged into IDF this month.

thx

Alvin1Zhang, 2020-01-15:
@playGit888 Thanks for reporting.  Would you please help try the latest master to see whether the issue still happens? Thanks.

sosaucily, 2020-03-12:
We had been getting this error periodically, but since updating esp-idf to the head of 3.3.1 we no longer see the error message.

Alvin1Zhang, 2020-03-13:
@playGit888 @sosaucily Thanks for reporting and updates, feel free to reopen if the issue still happens. Thanks.

closed 2020-03-13 by Alvin1Zhang

rajneeshsinha, 2020-12-09:
> We had been getting this error periodically, but since updating esp-idf to the head of 3.3.1 we no longer see the error message.

Hello

I am facing this issue "**wifi: ampdu: ignore deleting tx BA0**" with ESP-IDF version-3.3.1 in amazon freeRTOS SDK when connecting ESP32 AP to iPhone safari browser.

Can anyone provide the PR link for this issue? I just wanted to check what exact changes made this issue fixed.

zhangyanjiaoesp, 2020-12-10:
@rajneeshsinha 
this branch c5e37c70737237bf971bde5b532ce49577a316f6 solve the problem.