espressif/esp-idf #3895: esp_partition_erase_range offset documentation error

123swk123, 2019-08-07:
As per the documentation mentioned, offset is mentioned as relative to partition and it should be 4KB aligned, but in reality alignment is not applied on offset, but on absolute address. so the 4KB alignment is checked like this: (partition.address + offset) % 4096 == 0

https://github.com/espressif/esp-idf/blob/39f090a4f1dee4e325f8109d880bf3627034d839/components/spi_flash/include/esp_partition.h#L252

negativekelvin, 2019-08-08:
Is there a difference since the partition is also aligned?

123swk123, 2019-08-08:
i can create a partition which is not 4KB aligned and pass it on to erase function with offset 0x0000 then it fails with invalid args error.

negativekelvin, 2019-08-08:
Ok that is interesting and possibly useful for read-only partitions.  So document that partition erase requires the partition to be sector aligned.  Sounds good.

projectgus, 2019-08-13:
Hi @123swk123 ,

Thanks for reporting this. I think the restriction we probably need is for all partitions to be sector (4KB) aligned in address and length, I didn't realise it was possible to have ones which were not. Otherwise parts of the partition(s) cannot be erased.

Unless you have a particular reason to require non-4KB aligned partitions, @123swk123 ?

Angus

Alvin1Zhang, 2019-08-21:
@123swk123 Hi, thanks for reporting this, would you please help share if any updates for the issue?  Thanks.

123swk123, 2019-09-09:
@Alvin1Zhang update regarding documentation fix or code fix?

123swk123, 2019-10-15:
> Hi @123swk123 ,
>
> Thanks for reporting this. I think the restriction we probably need is for all partitions to be sector (4KB) aligned in address and length, I didn't realise it was possible to have [...]

Hi @projectgus,
I have a requirement where I need to use every bytes of the remaining space after application image + boot loader image. so we are reluctant to loose even 4KB for the reason of erase api requirement.

but I do understand this constraints comes from spi flash memory and hence we need to write some additional logic in spi flash component handle this unaligned erase operation.

projectgus, 2019-10-15:
Hi @123swk123 ,

Yes, this is a hardware limitation. SPI NOR flash chips have 4KB pages and we can only erase a single page at a time (erasing sets all bits to 1s). The only way to change a bit from 0 to 1 is to erase the entire 4KB page.

The only way to work around this would to be read the entire 4KB page into RAM, erase the page, then write back the parts which you don't plan to re-write.

We probably won't ever support this in ESP-IDF because it's too risky if there's a power failure. However if you really need the space then you could implement this in your project using the raw spi_flash API (which takes absolute addresses rather than partition-relative addresses).

123swk123, 2019-10-15:
hi @projectgus 

for the beta release of my project I don't plan to optimize the space utilization probably in near future I plan to update the spi_flash component and using component config to control this erase behavior. I will keep you posted once I finish the update.

Alvin1Zhang, 2020-05-28:
@123swk123 Would you please help share if any updates for the issue? Thanks.

Alvin1Zhang, 2022-01-25:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2022-01-25 by Alvin1Zhang