espressif/esp-idf #3898: Firmware is not working properly after OTA upgrade

GitOfLance, 2019-08-08:
### Environment
- Module or chip used: ESP32-WROVER-B
- IDF version : v4.0-dev-1443-g39f090a4f
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

##Problem Description
I refer to this demo(`https://github.com/espressif/esp-idf/blob/39f090a4f1dee4e325f8109d880bf3627034d839/examples/system/ota/native_ota_example/main/native_ota_example.c`) to implement the OTA upgrade function. I first put the project file in `.bin` format under the `build/` folder in the project into the `server`,like this:
[image]

Then send it to the device through the wifi, and OTA upgrade after successful data verification.
My code:
```
    new_partition = esp_ota_get_next_update_partition(NULL);

    ... //调试信息

    /* 开始OTA更新 */
    if (ESP_OK != esp_ota_begin(new_partition, OTA_SIZE_UNKNOWN, &new_handler))     /* 该分区全部擦除 */
    {
        ESP_LOGE(TAG, "ESP OTA BEGIN Failed");
        return -2;
    }

    /* 新Image写入OTA分区 */
    if (ESP_OK != esp_ota_write(new_handler, new_firmware, firmware_size))      /* 可以分次写入，自动排序 */
    {
        ESP_LOGE(TAG, "ESP OTA WRITE Failed");
        return -3;
    }

    /* 结束OTA分区更新 */
    if (ESP_OK != esp_ota_end(new_handler))
    {
        ESP_LOGE(TAG, "ESP OTA END Failed");
        return -4;
    }

    /* 指定bootloader引导该新APP分区 */
    if (ESP_OK != esp_ota_set_boot_partition(new_partition))
    {
        ESP_LOGE(TAG, "ESP OTA SET BOOT Failed");
        return -5;        
    }

    /* 重启 */
    esp_restart();
```
However, after the upgrade, the program will print long and long spaces, and no corresponding debugging information will appear.
Log informtion:
[image]
[image]

Partition Table：
```
# Name,   Type, SubType, Offset,   Size
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     ,         16K
otadata,  data, ota,     ,         8K
phy_init, data, phy,     ,         4K
factory,  0,    0,       ,         2M
coredump, data, coredump,,         64K
ota_0,    0,    ota_0,   ,         2M
ota_1,    0,    ota_1,   ,         2M
filestorage,  data, spiffs, ,      100K
```

GitOfLance, 2019-08-15:
Hi, @liuzfesp 
Have you received this question?

Alvin1Zhang, 2019-08-21:
@GitOfLance Hi, sorry for late reply and thanks for reporting the issue, we have received your issue and will look into it. Thanks.

GitOfLance, 2019-08-21:
@Alvin1Zhang Ok, I am looking forward to your reply.

27-301, 2019-08-22:
Does native_ota_example work for you? Please share your sample project which can reproduce this issue directly, also include sdkconfig & partition table.

GitOfLance, 2019-08-26:
Hi, @27-301 @Alvin1Zhang 
Very surprised, I want to complete the process, but I have not reproduced it.
Therefore, this problem can only be put on hold for a while.
Have you updated the OTA code? Or let the code fly for a while...

27-301, 2019-08-28:
Which commit do you use by now? Noticed that you have tested by commit 39f090a4f before.

GitOfLance, 2019-09-02:
Hi, @27-301 
I'm so Sorry, although I looked at the commit record, I still didn't think which version I used.

Alvin1Zhang, 2020-05-28:
@GitOfLance Would you please help share if any updates for the issue? Thanks .

GitOfLance, 2020-05-29:
Hi, @Alvin1Zhang 
The OTA upgrade was successful, and there was no problem with the code logic. After many tests, it was found that the server developer accidentally damaged the .bin file. And The ESP-IDF used at the time was the v4.0 development version

Alvin1Zhang, 2020-05-29:
@GitOfLance Thanks for sharing the updates, and glad that you found the cause of the issue. Could this issue be closed now? Thanks.

closed 2020-05-29 by GitOfLance