espressif/esp-idf #3899: std::conditional_variable::wait_for - no timeout

perimeno, 2019-08-08:
## Description
std::Conditional_variable wait_for methods constantly returns cv_status::no_timeout

## Environment
- Development Kit: ESP32-PICO-Kit
- Kit version: v4
- Module or chip used:  ESP32-PICO-D4
- IDF version v3.2.2
- Build System: Make or CMake (identical behavior)
- Compiler version  1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply:  USB

## Steps to reproduce
Compile and upload the following code:
```cpp
#include <iostream>
#include <thread>
#include <condition_variable>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_log.h>

using namespace std;
extern "C" void app_main()
{

    mutex m;
    unique_lock<mutex>lck(m);
    condition_variable cnd;

    thread t1([&](){
        for(int i=0;;++i){
        cout<<"A thread"<<endl<<std::flush;
        this_thread::sleep_for(std::chrono::seconds(2));
        cout<<"A thread notify"<<endl<<std::flush;
        cnd.notify_one();
        }
    });
    thread t2([&](){
        while(1){
        if(cnd.wait_for(lck,std::chrono::milliseconds(1500))==cv_status::timeout){
            cout<<"B thread timeout"<<endl<<std::flush;
        }
        else{
            cout<<"B thread notified"<<endl<<std::flush;
        }
        }
    });

    while(1){
        this_thread::sleep_for(std::chrono::seconds(5));
    }
}

```
### Expected output
```A thread
B thread timeout
A thread notify
A thread
B thread notified
B thread timeout
A thread notify
A thread
B thread notified
B thread timeout
A thread notify
...
```
### Actual output
```A thread
B thread notified
A thread notify
B thread notified
A thread
B thread notified
A thread notify
B thread notified
A thread
B thread notified
A thread notify
B thread notified
...
```
## Other info
sdkconfig is default, except 2MBit/s upload speed to 4 MB flash
I tried to run the following code on the latest master brach compiled with above mentioned compiler and ```tensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0```, but during the last case, the system restarts because of unsuccessful mutex unlocking.

X-Ryl669, 2019-08-12:
Your code is wrong. You have 2 core here and the order of execution is not like you expect. 

To use `condition_variable`, you need to lock a mutex to modify a variable before accessing using the cv. In your thread 1, you don't do that. This is required so the compiler issues a full memory barrier (so that CPU#1 can see what was modified by CPU#0)

Also, a cv can support spurious wakeup and you don't deal with them. Typically, when using a cv, you've to check the condition after being woken up and resume waiting if it's not happening. 
Here you have no condition to check so it's unlikely it'll work as you expect.
See https://en.cppreference.com/w/cpp/thread/condition_variable

Also, remember you are running in realtime OS. In such system, a lock can livelock if a higher priority task is waiting for a lock locked by a low priority task. You have to use priority inheritance whenever possible here, and this means no C++'s `std::mutex` (unless FreeRTOS allows changing the mutex style after construction, I don't know).

perimeno, 2019-08-12:
Thanks for the clarification! To be honest, I haven't heard spurious wakeup before (:/). If someone runs into this problem in the future [here](https://www.modernescpp.com/index.php/c-core-guidelines-be-aware-of-the-traps-of-condition-variables) is a good article about it. 

I see your point in the last paragraph, it's good to know, I'll keep in mind.
Thank you again for your answer @X-Ryl669 , it was really helpful!

closed 2019-08-12 by perimeno