espressif/esp-idf #3900: Send OnOff to group address

jkockx, 2019-08-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v2]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: v4.0-dev-1191-g138c941fa
- Build System:         [Make]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

I achieved to implement a provisioner which is able to:
- provision a node
- add and bind an application keys to both local client models as the server models in the node
I think that I was also able to add a subscription to a group address (100) to the model in the node.  So, I think I'm doing things right uptill here.
Though, if I now try to send a message to the OnOff client using the group address (100) with the generic OnOff api, I get the error as given in the Debug Logs section.
The errors seem to make sense as the api seems to check for the destination address in the list of provisioned devices mesh_nodes based upon its unicast address. The group address is of course not present in this list. Could someone help me out how I can get my OnOff set message to my OnOff Client using the group address?

## Debug Logs
Debug log of the addition of the subscription to the OnOff Client:
`I (846956) BLUETOOTH: esp_ble_mesh_config_client_cb, error_code = 0x00, event = 0x01, addr: 0x0005, opcode: 0x801b
I (846956) BLUETOOTH: ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD
add_group_address:status=1,sub_addr=100,element_addr=5,model_id=4096`

Debug log of the attempt to send OnOff set messages to the client using the group address (100):
`E (872896) BLE_MESH: model_send, Failed to check DST
E (872896) BLE_MESH: gen_set_state, Failed to send Generic Set message (err -22)
I (872916) BLUETOOTH: esp_ble_mesh_generic_client_cb, error_code = 0xffffffea, event = 0x01, addr: 0x0064, opcode: 0x8203
E (872926) BLUETOOTH: Send generic client message failed, opcode 0x8203`

jkockx, 2019-08-09:
I added some funcitonality to my code to get the subscription list from the model I tried to add the subscription to.  My assumption subscription in the node was successfull is probably false because I get an emty list back from this function. The code I used to add the subscription address was the following and the information I got back from this operation is given in the original issue:
`esp_err_t esp_ble_mesh_set_msg_common(esp_ble_mesh_client_common_param_t *common,
        uint16_t unicast, int net_idx, uint16_t app_idx,
        esp_ble_mesh_model_t *model, uint32_t opcode)
{
    if (!common || !model) {
        return ESP_ERR_INVALID_ARG;
    }

    common->opcode = opcode;
    common->model = model;
    /** NetKey Index of the subnet through which to send the message. */
    common->ctx.net_idx = net_idx;
    common->ctx.app_idx = app_idx;
    /** Remote address **/
    common->ctx.addr = unicast;
    common->ctx.send_ttl = MSG_SEND_TTL;
    common->ctx.send_rel = MSG_SEND_REL;
    // MSG_TIMEOUT 0 such that the value from menuconfig will be used (in milliseconds)
    common->msg_timeout = 0;
    common->msg_role = MSG_ROLE;
    return ESP_OK;
}

int cfg_client_add_group_address(uint16_t element_addr, uint16_t sub_addr, uint16_t model_id) {
	int err;
	esp_ble_mesh_client_common_param_t common = {0};
	esp_ble_mesh_cfg_client_set_state_t set_state = {0};
	esp_ble_mesh_set_msg_common(&common, element_addr, 0, 0, config_client.model, ESP_BLE_MESH_MODEL_OP_MODEL_SUB_ADD);
	set_state.model_sub_add.element_addr = element_addr;
	set_state.model_sub_add.sub_addr = sub_addr;
	set_state.model_sub_add.model_id = model_id;
	set_state.model_sub_add.company_id = CID_NVAL;
	err = esp_ble_mesh_config_client_set_state(&common, &set_state);
	if (err) {
		ESP_LOGE(TAGCFG, "%s: Config Model add subscription to group address to model", __func__);
		return err;
	}
	return err;
}`

WCCWCC, 2019-08-12:
Hi jkockx,
     1. Your destination address is wrong.
        `E (872896) BLE_MESH: model_send, Failed to check DST`
     2. You can change the address to 0xc000.
    （Dynamically Assigned） -> 0xC000-0xFEFF

Alvin1Zhang, 2019-08-21:
@jkockx Thanks for reporting the issue. Would you please help share if your issue has been resolved? Thanks.

jkockx, 2019-09-09:
OK, I was away from this project for a while. The problem was indeed that I tried to use an address from the unicast address space as a group address. 
Thank you for the resolving my issue!

Alvin1Zhang, 2019-10-11:
@jkockx Thanks for reporting, feel free to reopen if the issue still exists, thanks.

closed 2019-10-11 by Alvin1Zhang