espressif/esp-idf #3901: The filter for the monitor works poorly and there is no documentation for cmake.

ArTyGill, 2019-08-08:
idf.py monitor --print-filter="*:E GATTS_TABLE_DEMO:I"
Displays a message ( Error: No such command "GATTS_TABLE_DEMO:I". )

ArTyGill, 2019-08-08:
idf.py monitor --print-filter="GATTS_TABLE_DEMO:I"
this works good, but I need the rest of the tags

negativekelvin, 2019-08-08:
Which os and shell?  Did you try escaping the space?

dobairoland, 2019-08-09:
Hi @ArTyGill. Thanks for reporting the issue but unfortunately I cannot reproduce it based on the information you've provided. The `print-filter` used by you is working for me. 

So please provide also the full log message and IDF version beside the previously mentioned OS and shell. Thanks!

dobairoland, 2019-08-13:
@ArTyGill Do you have any update regarding your issue?

Alvin1Zhang, 2019-08-21:
@ArTyGill Thanks for reporting the issue, will close this due to short of feedback. Feel free to reopen if you still have this issue, thanks.

closed 2019-08-21 by Alvin1Zhang