espressif/esp-idf #3902: lwip compile error

chegewara, 2019-08-09:
In menuconfig disabling this option:
`LWIP->TCP->Queue incoming out-of-order segments (CONFIG_TCP_QUEUE_OOSEQ)`
causes this error:
```
components/lwip/lwip/src/core/pbuf.c: In function 'pbuf_free':
components/lwip/lwip/src/core/pbuf.c:792:29: error: dereferencing pointer to incomplete type 'struct netif'
               && p->l2_owner->l2_buffer_free_notify != NULL) {
```
esp-idf version: v3.2

joedeveloper, 2019-10-17:
Similar situation on commit 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df (HEAD, tag: v3.3)

freakyxue, 2019-11-13:
@chegewara 

We will debug the problem as soon as possible

vikramvel, 2019-11-18:
Hi, I am facing the same error.
Is this issue has been fixed ?

freakyxue, 2020-01-16:
@vikramvel @chegewara @joedeveloper    
I'm sorry I didn't reply in time.
I found the  root cause.
In the `pbuf. c` file,add lwip/netif.h.such as:
`#include "lwip/netif.h"`

kangg18, 2020-05-27:
function signature might be miss typed   in  lwip/netif.h

 void (*l2_buffer_free_notify)(void *user_buf) 
===> 
 void (*l2_buffer_free_notify)(struct netif *netif,void *user_buf)

this function might be related  to pppos and ip forward functional in NAPT  function

Alvin1Zhang, 2020-05-28:
Sorry for replying late, the fix has been available at https://github.com/espressif/esp-idf/commit/db38a4e06826d052858e70b5464d0a07f9d902e5, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-28 by Alvin1Zhang