espressif/esp-idf #3903: gatt_server vs gatt_security_server

tuttyfrutyee, 2019-08-09:
I have the latest version of esp-idf. Build and flashed both **gatt_server** and **gatt_security_server** to my esp32 devCkit.

 Tried to connect to these servers with my android phone with both an application downloaded from google store and manually selecting from phone bluetooth settings. 

Gatt_security_server **successfully** connected to my phone in both case, whereas with gatt_server example I had trouble connecting by manually from **android settings**. Gatt_server gave me these errors after few seconds from connection attempt, and then disconnected:

 **W (180914) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB
E (180914) BT_BTM: Device not found**

I could not find any documentation about why this is the case. Could you help me understand why manually selecting the bluetooth server(**gatt_server**) from settings gave me these errors.

chegewara, 2019-08-09:
Did you completely erase esp32 when you flash new firmware? 
Also did you erase bonded device when you are switching from secure server to not secure server?

When one device have stored bond info about peer device then it may cause a lot of issues.

tuttyfrutyee, 2019-08-12:
Yes I tried both erasing the bond from phone and idf.py erase_flash and the problem still remains.

gengyuchao, 2019-08-15:
I just updated the IDF to the latest version and tried using the gatt_server example, and there was no problem connecting to my phone. Have you made any changes to the example? I suggest you check it by git diff, if there still have the same problem, please tell me the more detailed log and phone model, so that I can see the cause of the problem.

tuttyfrutyee, 2019-08-20:
No I haven't changed anything. Tried to built it with two different computers with latest idf. 
Everything at start seems to work nicely, then this happens.

```
W (54147) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

E (54147) BT_BTM: Device not found

I (54147) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13

```
Phone Model : Xiaomi Mi 5, Android Version : 8.0.0, MIUI Version : MIUI 10 Global 8.11.22 Beta

gengyuchao, 2019-08-20:
I (54147) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
This reason means that Connection is terminated by the peer user. Could you try to use another phone to confirm if it is the problem of phone model?Because this situation is not occur on my phone.

gengyuchao, 2019-09-23:
Hey, I tried the latest version of esp-idf and tested it with Xiaomi Mi 5, but it never happened. Do you still have this problem? If it still exists, can you tell me what application you use on your phone? Maybe you can try changing the phone app to connect.

Alvin1Zhang, 2019-12-30:
@tuttyfrutyee Thanks for reporting. Would you please help share if your issue still exists? If it still happens, would you please share more details, like what application you use on your phone and etc., this would help up in reproducing and debugging issues a lot. Thanks.

Alvin1Zhang, 2020-01-03:
@tuttyfrutyee Thanks for reporting. Would you please help share if your issue still exists?  Thanks.

Alvin1Zhang, 2020-03-12:
@tuttyfrutyee Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2020-03-12 by Alvin1Zhang

hassance3, 2020-06-05:
I have the same problem, using gatts_server demo from BLE examples folder, Android 10 phone, Android 7 tablet.

younglee123, 2020-06-10:
This problem is related with the security process of smart phone . Some phone will force to pair with our device which do not have a security program. So you may get the above error message.
There have two solutions for this question: 
1.Use APP (such as nRF) to connect with the board. 
2.Add security program in the board(such as the gatt_security_server demo).

hassance3, 2020-06-12:
Thank you @younglee123 yeah gatt_security_server_demo has a complete secure pairing, I successfully added it into gatts_demo, it's like mixing 'gatts_demo.c' and 'example_ble_sec_gatts_demo.c'.

It's working fine, Thank you.

tanzhi963, 2024-02-17:
This issue is due to using the Bluetooth option in the system settings. If you directly connect to the Bluetooth of ESP32 after searching, an error message will appear, and the error message will be the same as described above. I haven't solved this problem either, I think it's because some services were not completed in the code.